openwrt/asu #12: platforms with generic images

dangowrt, 2017-06-30:
For x86 platforms there are generic image names rather than board-specific. Yet, /tmp/sysinfo/board_name is populated, but a generic image should be requested on those platforms.
Ie. `lede-x86-generic-combined-squashfs.img` for x86/generic and `lede-x86-64-combined-squashfs.img` for x86/64 are used for all boards respective boards.
Anyway, just be aware that some platforms do and some platforms don't have board-specific images.
Maybe use a different BIN_DIR rather than just EXTRA_IMAGE_NAME for each build and fuzzy-match the image name, falling back on generic names if the board-name wasn't found?

aparcar, 2017-07-02:
currently the server ignores the profile/board whenever the target is `x86`
images are not separated based on profile, would that somehow improve the structure?

aparcar, 2017-07-29:
@dangowrt currently [checks](https://github.com/aparcar/gsoc17-attended-sysupgrade/blob/master/update-server/image_request.py#L22) if the profile is valid. Should it just fallback to "Generic" if the profile is unknown?

dangowrt, 2017-07-29:
> Should it just fallback to "Generic" if the profile is unknown?

Sounds reasonable.

aparcar, 2017-07-29:
`ramips` does not have a Generic profile, in that case the error should still fire right?

commit 6debc7a0 (2017-07-29): fix #12
closed 2017-07-29 by aparcar