openwrt/asu #24: hidden dependencies

dangowrt, 2017-07-18:
when generating an image, the `PACKAGES` list is passed to the imagebuilder which parses it and then thoughs it at `opkg install ...`.
opkg now installs the packages listed **including their dependencies**.
hence the list of packages in an image-request does not map straight to the list of packages installed in the image. there are many images-requests which should point to the same image.

There are two possible ways to fix this:

1. re-implement opkg in SQL, ie. import all the dependency, provides, conflicts, ... into the database and evaluate the exact list of packages

2. collect the manifest generated by the IB which comes with the image and store the information about installed packages and versions in the database. mapping (multiple) image-requests to the same image can then be accomplished once the IB has run **once** for each synonymous image-request. That creates an avoidable overhead, but has the side-effect of verifying the reproducibility of the images built (as the hash of the image should be identical if the manifest is identical).

I have a strong tendency for option **__2__**

aparcar, 2017-07-18:
Good point. Will change the database layout accordingly

aparcar, 2017-07-29:
with ee27a736f2b2f9eb764d9e5a6733c6ded80c2f74 the images are now stored based on resulting manifest. 
Requesting an image with `luci` installed and another with `luci` and `luci-base` installed result in the same image/manifest, however the hash of the requested packages is different. From now on it's linked to a table containing all created images with a hash of the manifest.

closed 2017-07-29 by aparcar

dangowrt, 2017-07-29:
I reckon the different hash of the rootfs is related to squashfs not
producing reproducible images (yet). I reckon @lynxis should migrate
tools/squashfs to use that new squashfs upstream and we should
backport that change to lede-17.01.

aparcar, 2017-07-29:
I think this is already in progress.

FreifunkUFO, 2017-08-08:
nice! 
works for me: i tried it with freifunk-meshkit-imagebuilder (with yesterdays lede-trunk) and i got "attendedsysupgraded" the same count of packages as installed before.