openwrt/asu #26: not working for Ubiquity devices

FreifunkUFO, 2017-08-09:
there is an error when trying it with UBNT devices: board not found

FreifunkUFO, 2017-08-09:
f.i. board-name is "loco-m-xw", but imagebuilder wants "ubnt-loco-m-xw".

root@74-57:~# ubus call system board {
"kernel": "4.4.79",
"hostname": "74-57",
"system": "Atheros AR9342 rev 2",
"model": "Ubiquiti Loco M XW",
"board_name": "loco-m-xw",

so please check LEDE (maybe a new ticket there?) and write a workaround.
f.i. with an "if model=Ubiquity"-question and modified subfunction:

function image_request() {                                                           
        console.log("image_request")                                                                                  
        request_dict = {}                                                                                                           
        request_dict.version = latest_version;                                          
        request_dict.board = "ubnt-" + data.board_name                      
        server_request(request_dict, "image-request", image_request_handler)                                     
}  

deeplink: https://github.com/openwrt/luci/blob/master/applications/luci-app-attendedsysupgrade/luasrc/view/attendedsysupgrade.htm#L180

FreifunkUFO, 2017-08-09:
[image]

aparcar, 2017-08-09:
thanks for the testing, seems like the imagebuilder calls the image `ubnt-loco-m-xw` while the `board_info` contains `loco-m-xw`. I'll have to ask the lede project why the `ubnt-` part is missing

FreifunkUFO, 2017-08-10:
is it only happening for ubnt-devices?
what about a workaround? (with an "if model=Ubiquity"-question and modified subfunction (+=ubnt))

dangowrt, 2017-08-10:
On Thu, Aug 10, 2017 at 06:15:35AM -0700, FreifunkUFO wrote:
> is it only happening for ubnt-devices?
> what about a workaround? (with an "if model=Ubiquity"-question and modified subfunction (+=ubnt))

unifying the board_name to be identical with the IB profile name should
be done in OpenWrt/LEDE. As a temporary work-around there could be a DB
table to resolve this in a generic way.
Please don't introduce any vendor-name regex in the Python code...

aparcar, 2017-08-10:
```
 id  | distro | release  | target | subtarget |         profile         | model                
 281 | lede   | snapshot | ar71xx | generic   | ubnt-loco-m-xw          | Ubiquiti Loco XW
```
"model": "Ubiquiti Loco M XW",
"board_name": "loco-m-xw",

Looks like I have to work with (SQL) regex after all.

@dangowrt @FreifunkUFO other ideas to solve this in a more generic way?

closed 2017-08-10 by aparcar

dangowrt, 2017-08-11:
First of all, you should file a bug at http://bugs.lede-project.org/
because the board_name should be changed in
target/linux/ar71xx/base-files/lib/ar71xx.sh and its users to be
consistent with the name used when creating the an image.

However, it might not be feasible to backport this change to
lede-17.01.
Hence you might still need an extra table for alternative board names.
This will be needed anyway as sometimes a boardname changes between
releases...

FreifunkUFO, 2017-11-14:
ticket was closed. but please link the filed bug at lede-project about board_name