openwrt/asu #27: possibility of different updateserver

FreifunkUFO, 2017-08-09:
before clicking the "search update" button the adress of the update-server should appear.

it also should be editable! (in order to use a different updateserver)

[image]

aparcar, 2017-08-09:
currently you can edit the option via uci in `/etc/config/attendedsysupgrade`. A config will be added to the web-view

aparcar, 2017-10-24:
currently we have 2 active server:
* ledeupdate.planetexpress.cc
* betaupdate.libremesh.org

I move this issue to https://github.com/aparcar/gsoc17-attended-sysupgrade/issues/52

closed 2017-10-24 by aparcar

FreifunkUFO, 2017-10-25:
i cant see any mention about this issue at referenced #52

aparcar, 2017-10-26:
@FreifunkUFO checkout the initial post, it has a ToDo list attached