openwrt/asu #28: browserproblems with firefox

FreifunkUFO, 2017-08-09:
with firefox i cant click the "search" button.
with cromium all is set. my OS is ubuntu

aparcar, 2017-08-09:
please press str+shift+k in firefox to see the console output. does it show any error messages?

aparcar, 2017-08-15:
@FreifunkUFO ping

FreifunkUFO, 2017-08-15:
yes, it does. the error is called "Fehler". 
(btw. Firefox is available at http://mozilla.org)

aparcar, 2017-08-15:
I'm using Firefox while developing and did not acknowledge this behaviour yet. What version do you use?

FreifunkUFO, 2017-08-15:
it might be the most current version. i was also trying with outdated version.
btw: i was trying LEDE with freifunk-settings, built with meshkit via imagebuilder

aparcar, 2017-08-16:
I tried the following versions of Firefox and all worked without a problem:
* 52.2.0 (64-bit)
* 55.0 (64-bit)
* 57.0a1 (2017-08-15) (64-bit)
All work as expected.

The MeshKit image x86/64 would boot, stuck on the following message forever:
`Waiting for root device /dev/mtdblock0...`

Is there a more verbose output than just "Fehler" possible?

aparcar, 2017-08-18:
@FreifunkUFO did we resolv this issue via IRC?

aparcar, 2017-08-31:
closing due to no response
worked here with all current Firefox releases (on linux)

closed 2017-08-31 by aparcar