openwrt/asu #29: make the firmware-destination editable

FreifunkUFO, 2017-08-09:
when i see "new update avaiblae", i can see the prefered destination (f.i. "SNAPSHOT")

please make that textfield editable. i need the possibility to switch from LEDE-release to LEDE-trunk (and vice versa) or other firmware variation.

[image]

FreifunkUFO, 2017-08-09:
goals for instance:

- openwrt-trunk
- openwrt-release
- lede-release
- lede-trunk
- lede-trunk from last month
- lede-release with pimped wifi drivers
- lede-release with stripped symbols
- lede-trunk with enabled kernel debug symbols

aparcar, 2017-08-13:
currently only releases and snapshots are planed. Also downgrades are not really planed. 
I could add a field to switch between _latest release_ and _snapshot_ if currently not running a snapshot.

aparcar, 2017-11-03:
I'm not intending to support all kinds of different image builder settings, this would currently create to much overhead in the database and image requests. The server is supposed to supply images for sysupgrades, not focused on developers with custom needs.

However, if you create a patch I'd like to add this feature. If not please close this issue.

aparcar, 2017-11-12:
Update server is now editable in latest luci-app-attendedsysupgrade, depends on the imagebuilder settings of the chosen server all variations of builds are possible

closed 2017-11-12 by aparcar