openwrt/asu #30: make the packages-list editable

FreifunkUFO, 2017-08-09:
when requesting the image there should be a possibility to VIEW AND CHANGE the list of packages

FreifunkUFO, 2017-08-09:
[image]

aparcar, 2017-08-09:
Viewing is already planed, changing is something, if even planed, for after the GSoC

aparcar, 2017-08-11:
currently a simple implementation would be to disable single packages via checkboxes. 
adding packages is more complicated as there won't be a auto completion for package names. 
@FreifunkUFO would the first version help you?

aparcar, 2017-10-16:
this is a request for luci-app-attendedupgrade. I will shortly create a PR for that on openwrt/luci

FreifunkUFO, 2017-10-16:
so please keep this ticket open!?

aparcar, 2017-10-19:
https://github.com/openwrt/luci/pull/1404

FreifunkUFO, 2017-10-20:
nice!

the layout is getting even more worse with every change/wish :-)

[image]

aparcar, 2017-10-20:
It looks a bit better on the official bootstrap version, but feel free to add design suggestions :)

closed 2017-10-20 by aparcar