openwrt/asu #31: improve the settings in /etc/config

FreifunkUFO, 2017-08-09:
on https://github.com/openwrt/luci/blob/master/applications/luci-app-attendedsysupgrade/luasrc/view/attendedsysupgrade.htm#L90 there is a 

settings[0].

please change it to a better openwrt standard

aparcar, 2017-08-09:
Do you have a better idea for the names? `config setttings "update"` looks a bit odd to me.

FreifunkUFO, 2017-08-10:
you could use:

config settings 'settings'

dangowrt, 2017-08-10:
On Thu, Aug 10, 2017 at 08:59:54AM +0000, FreifunkUFO wrote:
> you could use:
> 
> config settings 'settings'

imho that'd make it even worse.

i suggest:
```
config updateserver 'beta'
	option name 'beta'
	option url 'http://betaupdate.libremesh.org/'
	option selected '1'

config updateserver 'official'
	option name 'official'
	option url 'http://update.openwrt.org/'

config updateclient
	option whatever '1'
```

aparcar, 2017-08-10:
looks good to me, I'll set this up

aparcar, 2017-08-15:
merged upstream
https://github.com/openwrt/luci/commit/415d3aeb164754e9e16408f95d6e5ff94b0fdf62
https://github.com/openwrt/packages/commit/bc0a03071e24fddc118186f9aced64840a77d80e

currently only one update server is supported. If you feel the need of a dropdown menu for a large variety of update servers please open a separate issue

closed 2017-08-15 by aparcar