openwrt/asu #35: ipq806x images inconsistent naming

aparcar, 2017-08-13:
trying to build a image for ipq806x/generic produces the following list of files, no sysupgrade is found, only a `tar` file.

```
DEBUG:image:['/tmp/tmpgt6us7pp/lede-ipq806x-EA8500-squashfs-factory.bin', 
'/tmp/tmpgt6us7pp/lede-ipq806x-EA8500-squashfs-sysupgrade.tar', 
'/tmp/tmpgt6us7pp/lede-ipq806x-squashfs-root.img', 
'/tmp/tmpgt6us7pp/lede-ipq806x-ubifs-root.img', 
'/tmp/tmpgt6us7pp/sha256sums', 
'/tmp/tmpgt6us7pp/lede-ipq806x-vmlinux.elf', 
'/tmp/tmpgt6us7pp/lede-ipq806x-device-ea8500.manifest']
```

@dangowrt should I implement auto-extracting if tar files are found? What else could fix this?
Also, why are there `img` _and_ `bin` files?

dangowrt, 2017-08-13:
On Sun, Aug 13, 2017 at 01:50:43AM +0000, Paul Spooren wrote:
> trying to build a image for ipq806x/generic produces the following list of files, no sysupgrade is found, only a `tar` file.
>
> ``` [...]

No. The tar files contain the sysupgrade metadata and images needed
for NAND based devices. The sysupgrade command directly takes such a
tar file, so need to extract anything.

> Also, why are there `img` _and_ `bin` files?

We use `img` for filesystem images and `bin` for opaque binaries.

aparcar, 2017-08-13:
currently all sysupgrade images are renamed to a geneirc name `distro-release-target-subtarget-sysupgrade.bin`. Should the server store additionally the extension for all created images?

commit 256c54f9 (2017-08-14): fix #35
closed 2017-08-14 by aparcar