openwrt/asu #38: button doenst work atm: Search for update -> Access denied

FreifunkUFO, 2017-08-16:
today i was trying last lede-release on x86, and was installing official lede trunk packages for attended sysupgrade.

atm all ubus-http-request (without pressing any button) got "Access denied"

FreifunkUFO, 2017-08-16:
[image]

FreifunkUFO, 2017-08-16:
only happening with lede-release! (some luci stuff was changed)

but using this tool here to upgrade lede-release to lede-trunk is still very important. i think we have to opkg-update that luci-package too, when installing luci-attended-sysupgrade from snapshop to avoid this problem.

how is the name of that luci-package?

aparcar, 2017-08-16:
@jow- the special luci1 was recently merged into luci snapshots and removed from my script. Do you see a way to make luci-app-attendedsyupgrade run on older releases?

aparcar, 2017-08-19:
fixed with https://github.com/openwrt/luci/pull/1314

closed 2017-08-31 by aparcar

FreifunkUFO, 2017-10-20:
how is the name of that luci-package?

aparcar, 2017-10-21:
luci-app-attendedsysupgrade

FreifunkUFO, 2017-10-21:
LOL no.

there were some further changes in luci needed. so my question is about the package according to former question:

> Do you see a way to make luci-app-attendedsyupgrade run on older releases?

aparcar, 2017-10-21:
I recently created some pull requests asking to add the project to lede-17.01 stable branch. I'll keep you updated