openwrt/asu #47: support for NAND and block-based devices

dangowrt, 2017-10-12:
currently only NOR-based devices with squashfs-sysupgrade images are supported.
Support also devices which are block-device or NAND-flash based by picking up the required files generated by the ImageBuilder.

aparcar, 2017-10-12:
@dangowrt so how will the client (e.g. luci-app-attendedsysupgrade) know what file is the actual upgrade?
possible approach is to implement update-request/ and build-request/, while update only return url of sysupgrade, filesize and checksum the build-request contains a array of all created files.
the update-server (maybe from now on image-server?) has to determine itself what the sysupgrade file is

dangowrt, 2017-10-12:
All files (with their sha256sums signed) should be served to allow use-cases other than attended-sysupgrade.
And yes, the build server should pick the `squashfs` variant matching the sysupgrade file scheme of that target. The knowledge _which_ image file that is could be a list compared top-down for the first matching glob by the image server to pick which image to sign and suggest to the client:
```
*-squashfs-sysupgrade.bin
*-squashfs-sysupgrade.tar
*-squashfs.trx
*-squashfs.chk
*-squashfs.bin
*-squashfs-sdcard.img.gz
*-combined-squashfs*
```

x86 is going to be re-structured at some point hopefully, so the more logical thing would be to have different boards/devices like generic-uefi, generic-mbr, generic-iso, vmware-mbr (which could have e.g. different kernel parameters or bootloader options and returning a vmdk image), ... 
Until then, I suggest to simply only offer the `*-combined-squashfs*` image as that's what works for most people.

aparcar, 2017-10-13:
Great, the sysupgrade file is now detected based in the list above.
currently the server unifies the names of sysupgrade files. however, having various files per build needs a fast way to return them. Storing them in database or using a (python) glob could be an approach

aparcar, 2017-10-17:
All builds store now all files, an example here:
https://betaupdate.libremesh.org/static/lede/17.01.3/ar71xx/generic/tl-wdr4300-v1/
I'll create another API call (and pretty up nginx) to make these files accessible

aparcar, 2017-10-17:
@dangowrt could you please provide the same list as above for factory images?

dangowrt, 2017-10-17:
hm, it's not that trivial, because it depends on the factory flashing
method (web-ui, tftp-server, tftp-client, ...) as well as the flash
layout of the stock firmware (hence, sometimes multiple or intermediate
images are needed, especially for NAND-based devices).
the safest thing is to server *all* generated files as the
device-specific documentation may refer to a subset of them.

aparcar, 2017-10-21:
instead of storing the sysupgrade filename in the database a file link from `sysupgrade.bin` to `<complicated_sysupgrade_name_here>.<something>` could be created. Clients receive a url to all created files. The sysupgrade client simply requests `url/sysupgrade.bin`. 
@dangowrt

aparcar, 2017-10-24:
https://github.com/aparcar/gsoc17-attended-sysupgrade/issues/56

aparcar, 2017-10-25:
fixed with 767c192188db307c20843b0156ada6806160b44e 56f3daf183a41a6aba678923cd2ad0e627822db7 and cfff9735b4cc7131b8f7d0b0f5cf6bdb2cd1e424

images are no longer renamed. This still can lead to problem when the imagesbuilder names images in an odd way, like leaving the subtarget away, however mho this should be unified instead of creating an endless number of cases with different naming styles. 

@dangowrt please close the issue when it fits you!

aparcar, 2017-11-02:
should be all done

closed 2017-11-02 by aparcar