openwrt/asu #52: make layout great again

FreifunkUFO, 2017-10-20:
the layout is getting even more worse with every change/wish.
some changes should be done for a consistent style of this project.

FreifunkUFO, 2017-10-20:
@aparcar wrote:

It looks a bit better on the official bootstrap version, but feel free to add design suggestions :)

aparcar, 2017-10-20:
@egon0 can you help me out?

FreifunkUFO, 2017-10-20:
to start with the first page (without clicking the search button):

* there should be a small description about that package. (see other luci-subpages). of course with translations (at least for german language).
* button: "search for updates" sounds inconsistent, because the headline and name of the project is "... upgrade"! so the name "updates" isnt nice.. the result of the search is not "many updates", but one "upgrade". you might change the name to "online search" to clarify the online-character of this project

* the checkbox "Search for package updates": i really do not understand the purpose of that checkbox. also my problem with "updates", while the project does an "upgrade".
a translation also would be nice. (i8n)
if that checkbox does something during the "search" it should be ABOVE the search-button, because its an option for that search!? (see other luci-pages)

aparcar, 2017-10-20:
PRs are welcome 🙄 No seriously, I'll look into that

FreifunkUFO, 2017-10-20:
when clicking the first button:

* The message "imagebuilder not ready, please wait" should be in the same style as other luci-pages. please have a look at the layout of "Änderungen werden angewandt".

FreifunkUFO, 2017-10-20:
@aparcar first we have to collect the problems by ticketcomments. Press Release (PR) would not be nice for that moment.

aparcar, 2017-10-20:
Pull Requests?

FreifunkUFO, 2017-10-20:
cite:
"Search for package updates"

so, what does it mean? 
and, what would be a nice german translation?

aparcar, 2017-10-20:
Anwendungsaktualisierungen. however, I'm not sure how to implement translations with JavaScript.

FreifunkUFO, 2017-10-20:
so when disabling this checkbox there are  older packages but new kernel? i dont get it.

aparcar, 2017-10-20:
If 17.01.4 receives a package update but no new major 17.01.x release, with the option checked you simply get a new image

egon0, 2017-10-24:
Ping 😎

aparcar, 2017-10-26:
I'm not sure why I can not assign this directly to @egon0 , but he is the person going to solve this 👍

FreifunkUFO, 2017-10-29:
see https://github.com/aparcar/luci/commit/9e1728cd625b841cb35de5191d82639f7bd52286

FreifunkUFO, 2017-10-30:
having a description is nice. please change that tagging style from <p> to

<div class="cbi-map-descr">

aparcar, 2017-10-31:
Please see https://github.com/aparcar/luci/tree/v2 for recent changes.

FreifunkUFO, 2017-11-02:
please see https://github.com/openwrt/luci/pull/1408 for recent changes.
and @aparcar told me, new layout can be found here https://github.com/aparcar/luci/tree/v2

aparcar, 2017-11-02:
@FreifunkUFO please excuse the "chaos", v2 was merged in acls as there already existed an PR

FreifunkUFO, 2017-11-02:
and see https://github.com/openwrt/packages/pull/4979/files#diff-066cc010efe5ae62f0ca12bfec1e872d but its only for lede-relase (not for lede-trunk)

aparcar, 2017-11-02:
https://github.com/openwrt/packages/pull/5053

aparcar, 2017-11-12:
All done in latest luci version, PR is currently pending
https://github.com/openwrt/luci/pull/1437

@FreifunkUFO if translations are necessary please propose a solution, I have no experience with that

aparcar, 2017-11-25:
@FreifunkUFO ping

aparcar, 2018-08-03:
Closing this issue due to no response, please reopen if still relevant.

closed 2018-08-03 by aparcar