openwrt/asu #54: install problems with freifunk-meshkit firmware

FreifunkUFO, 2017-10-21:
with 2 days old trunk i am on installing attended sysupgrade via opkg
there are some error messages

opkg update
Downloading http://gadow.freifunk.net:8004/meshkit/buildroots/ar71xx-lede-201710                                                                  19/packages//Packages.gz
Updated list of available packages in /var/opkg-lists/reboot_core
root@120-9:~# opkg list | grep atten
luci-app-attendedsysupgrade - git-17.291.51172-a98548e-1 - LuCI support for atte                                                                  nded sysupgrades
rpcd-mod-attendedsysupgrade - 1-1 - implements sysupgrade function with ubus add                                                                   uci settings and needed acls
root@120-9:~# opkg install luci-app-attendedsysupgrade
Installing luci-app-attendedsysupgrade (git-17.291.51172-a98548e-1) to root...
Downloading http://gadow.freifunk.net:8004/meshkit/buildroots/ar71xx-lede-201710                                                                  19/packages//luci-app-attendedsysupgrade_git-17.291.51172-a98548e-1_all.ipk
Installing uhttpd-mod-ubus (2017-08-19-3fd58e9b-1) to root...
Downloading http://gadow.freifunk.net:8004/meshkit/buildroots/ar71xx-lede-201710                                                                  19/packages//uhttpd-mod-ubus_2017-08-19-3fd58e9b-1_mips_24kc.ipk
Installing cgi-io (4) to root...
Downloading http://gadow.freifunk.net:8004/meshkit/buildroots/ar71xx-lede-201710                                                                  19/packages//cgi-io_4_mips_24kc.ipk
Installing rpcd-mod-packagelist (0.1-1) to root...
Downloading http://gadow.freifunk.net:8004/meshkit/buildroots/ar71xx-lede-201710                                                                  19/packages//rpcd-mod-packagelist_0.1-1_mips_24kc.ipk
Installing rpcd-mod-attendedsysupgrade (1-1) to root...
Downloading http://gadow.freifunk.net:8004/meshkit/buildroots/ar71xx-lede-201710                                                                  19/packages//rpcd-mod-attendedsysupgrade_1-1_mips_24kc.ipk
Configuring cgi-io.
Configuring rpcd-mod-packagelist.
Configuring rpcd-mod-attendedsysupgrade.
//usr/lib/opkg/info/rpcd-mod-attendedsysupgrade.postinst: .: line 130: attendeds                                                                  ysupgrade: not found
Configuring uhttpd-mod-ubus.
//usr/lib/opkg/info/uhttpd-mod-ubus.postinst: .: line 130: 00_uhttpd_ubus: not f                                                                  ound
Configuring luci-app-attendedsysupgrade.

FreifunkUFO, 2017-10-21:
and next try:

` opkg remove  luci-app-attendedsysupgrade   rpcd-mod-attendedsysupgrade
No packages removed.
root@120-9:~# opkg remove  luci-app-attendedsysupgrade rpcd-mod-packagelist  rpcd-mod-attendedsysupgrade
Removing package rpcd-mod-packagelist from root...
root@120-9:~# vi /etc/opkg/distfeeds.conf
root@120-9:~#
root@120-9:~# opkg install luci-app-attendedsysupgrade
Installing luci-app-attendedsysupgrade (git-17.291.51172-a98548e-1) to root...
Downloading http://gadow.freifunk.net:8004/meshkit/buildroots/ar71xx-lede-20171019/packages//luci-app-attendedsysupgrade_git-17.291.51172-a98548e-1_all.ipk
Installing rpcd-mod-packagelist (0.1-1) to root...
Downloading http://gadow.freifunk.net:8004/meshkit/buildroots/ar71xx-lede-20171019/packages//rpcd-mod-packagelist_0.1-1_mips_24kc.ipk
Installing rpcd-mod-attendedsysupgrade (1-1) to root...
Downloading http://gadow.freifunk.net:8004/meshkit/buildroots/ar71xx-lede-20171019/packages//rpcd-mod-attendedsysupgrade_1-1_mips_24kc.ipk
Configuring rpcd-mod-packagelist.
Configuring rpcd-mod-attendedsysupgrade.
//usr/lib/opkg/info/rpcd-mod-attendedsysupgrade.postinst: .: line 130: attendedsysupgrade: not found
Configuring luci-app-attendedsysupgrade.
`

FreifunkUFO, 2017-10-21:
perhaps that "exit 01"?

https://github.com/openwrt/packages/blob/master/utils/rpcd-mod-attendedsysupgrade/files/attendedsysupgrade.defaults

aparcar, 2017-10-23:
I fixed the mentioned issue in this PR https://github.com/openwrt/packages/pull/4979

aparcar, 2017-10-28:
> //usr/lib/opkg/info/uhttpd-mod-ubus.postinst: .: line 130: 00_uhttpd_ubus: not f ound
Is it possible that this is currently an error in the snapshots repo? @dangowrt ?

aparcar, 2017-11-03:
It *is* a problem of the snapshot build and not the scripts of `rpcd-mod-attendedsysupgrade`

@FreifunkUFO please close this issue

closed 2017-11-12 by aparcar