openwrt/asu #55: build problems: Profile does not exist!

FreifunkUFO, 2017-10-23:
atm sysupgrade wont work for me:

[
    "lede",
    "snapshot",
    "ar71xx",
    "generic",
    "tl-wr902ac-v1",
    "f5b148faad84",
    ""
]

Profile "tl-wr902ac-v1" does not exist!
Use "make info" to get a list of available profile names.
Makefile:197: recipe for target 'image' failed
make: *** [image] Error 1

FreifunkUFO, 2017-10-23:
also jefe.tk is affected:
when building image:
internal server error. please try again later

aparcar, 2017-10-23:
Is it possible that the daily snapshot does not support the device?

FreifunkUFO, 2017-10-23:
the device was supported by LEDE the last weeks and running nice.
but you are true: @hauke killed the device recently :-(

https://github.com/lede-project/source/commit/f7a6fd31539be54d14d7c52b491b40b26bf8f740

aparcar, 2017-10-23:
using snapshots means living on the edge - sorry can't do anything about it currently

closed 2017-10-23 by aparcar