openwrt/asu #57: problems with LEDE SNAPSHOTS on cpe510

FreifunkUFO, 2017-10-25:
the possibility for upgrading LEDE snapshots is missing at the moment.

error: unknown device, please check model and board params

FreifunkUFO, 2017-10-25:
parameters are:

{"version":"SNAPSHOT","board":"cpe510","packages":["base-files","busybox","dropbear","mtd","uci","opkg","netifd","fstools","uclient-fetch","logd","kmod-gpio-button-hotplug","swconfig","dnsmasq","iptables","ip6tables","firewall","odhcpd","odhcp6c","uboot-envtools","batctl","collectd-mod-conntrack","collectd-mod-exec","collectd-mod-interface","collectd-mod-load","collectd-mod-olsrd","collectd-mod-ping","freifunk-common","freifunk-policyrouting","hostapd-mini","ip6tables-mod-nat","kmod-ath9k","kmod-batman-adv","libiwinfo-lua","luci-app-firewall","luci-app-freifunk-policyrouting","luci-app-freifunk-widgets","luci-app-statistics","luci-i18n-base-de","luci-i18n-firewall-de","luci-i18n-freifunk-de","luci-i18n-freifunk-policyrouting-de","luci-i18n-meshwizard-de","luci-i18n-olsr-de","luci-i18n-splash-de","luci-mod-admin-full","luci-mod-freifunk","luci-mod-freifunk-community","luci-theme-freifunk-generic","meshwizard","olsrd-mod-jsoninfo","olsrd-mod-txtinfo","rssileds","screen","tcpdump-mini","uhttpd","horst","luci-app-attendedsysupgrade","luci-mod-status","kernel"],"model":"TP-Link CPE510 v1.0","distro":"LEDE","target":"ar71xx","subtarget":"generic"}

FreifunkUFO, 2017-10-25:
maybe its a similiar mistake, see https://github.com/aparcar/jefe/issues/10

aparcar, 2017-10-25:
I was told that the snapshot option tend to break user devices so now there are two server:
* https://ledeupdate.planetexpress.cc - fast
* https://betaupdate.libremesh.org - with snapshots

https://jefe.tk uses the snapshot server, may should change in the future

closed 2017-10-25 by aparcar

FreifunkUFO, 2017-10-25:
oh..

so there is a need for switching/changing the updateserver via webui?
 https://github.com/aparcar/gsoc17-attended-sysupgrade/issues/27

FreifunkUFO, 2017-10-25:
we may have to blame LEDE for this issue. i cant see any cpe510 atm on http://downloads.lede-project.org/snapshots/targets/ar71xx/generic/

so i have to change my testing platform again. (recently, my 902ac devices are now "not supported" atm by "normal" LEDE, and so attended-sysupgrade

p.s. how is it tending to break devices? i cant see any issue, link or hint. or it the "normal" LEDE behaviour sometimes with some devices?

FreifunkUFO, 2017-10-25:
about CPE510:
the profile-name seems to be changed to "cpe510-520-v1" AND error-message about too big kernel (a well-known problem on LEDE with tplink-safeloader flashlayout)
related to https://github.com/lede-project/source/commit/f7a6fd31539be54d14d7c52b491b40b26bf8f740

aparcar, 2017-10-26:
I'm afraid that normal in snapshots - stuff moves. 

> p.s. how is it tending to break devices? i cant see any issue, link or hint. or it the "normal" LEDE behaviour sometimes with some devices?

Maybe it's comparable  to the old cyanogenmod nightlies, upgrading every day destroyed my system every other month. Better stick with the stable releases, at least for common users :)?

FreifunkUFO, 2017-10-26:
you are only mentioning rumors. i dont use cyanogenmod. :-/

so please tell us details or the names of relavant packages or devices told to you. 
LEDE stable releases dont support newer hardware.