openwrt/asu #58: problem: reboot LEDE before using attended-sysupgrade

FreifunkUFO, 2017-10-25:
the postinstall scripts seems to have problems:
when installing atrtended-sysupgrade on openwrt/lede (via opkg) this project cant be used until rebooting the device.

also, manually restart the rpcd- and uhttpd service doenst help.  there might be problems on the postinstall-scripts.

aparcar, 2017-10-25:
@dangowrt could you please help me out here?

jow-, 2017-10-25:
Try issueing an "rm -r /tmp/luci-indexcache /tmp/luci-modulecache/" in postinstall.

jow-, 2017-10-25:
Ah, also I think master got switched to uhttpd-mod-lua by default, you need to restart uhttpd to reload the LuCI framework.

aparcar, 2017-10-25:
@jow- I reused the "postinst" of a different luci-app and added a second commit to the existing PR
https://github.com/openwrt/luci/pull/1408/commits/76ea5878d724bcbcb72dff38ad6b066cde03e4e5

FreifunkUFO, 2017-10-25:
hm, i forgot to mention: i did restart uhttpd and rpcd, AND was doing "rm-r /tmp/lu*"
may i forgot the -f? but i cant test at the moment..

FreifunkUFO, 2017-10-25:
so, now i found the problem: 

the uci-settings are done via /etc/uci-defaults, but that needs a reboot. 
@jow-  wouldnt it be nicer to ship the /etc/config/attendedsysupgrade with the package instead of setting the values via uci-commands in /etc/uni-defaults? or is there a command to execute these files without reboot (for the post-install hook)?

(tested with one week old trunk firmware and these 3 packages from lede-snapshot from today)

> root@211-49:/etc/config# cd /etc/uci-defaults/
root@211-49:/etc/uci-defaults# ls
00_uhttpd_ubus      attendedsysupgrade
root@211-49:/etc/uci-defaults# cat *
#!/bin/sh

if [ -z "$(uci -q get uhttpd.main.ubus_prefix)" ]; then
        uci set uhttpd.main.ubus_prefix=/ubus
        uci commit uhttpd
fi

exit 0
#!/bin/sh

[ -e /etc/config/attendedsysupgrade ] && exit 0

touch /etc/config/attendedsysupgrade

uci -q batch <<EOF
set attendedsysupgrade.updateserver=updateserver
set attendedsysupgrade.updateserver.url='https://betaupdate.libremesh.org'

set attendedsysupgrade.updateclient=updateclient
set attendedsysupgrade.updateclient.update_packages='1'

commit attendedsysupgrade
EOF

exit 01
root@211-49:/etc/uci-defaults# ls /etc/config/a*
ls: /etc/config/a*: No such file or directory
root@211-49:/etc/uci-defaults# uname -a
Linux 211-49 4.4.93 #0 Thu Oct 19 10:27:44 2017 mips GNU/Linux
root@211-49:/etc/uci-defaults#

dangowrt, 2017-10-26:
Running uci-defaults is already triggered in default_postinst. See:

See https://github.com/lede-project/source/blob/master/package/base-files/files/lib/functions.sh#L238

FreifunkUFO, 2017-10-26:
ok, but that is not working atm. 

maybe its related to https://github.com/aparcar/gsoc17-attended-sysupgrade/issues/54 - the not-working postinstall-script..

FreifunkUFO, 2017-11-01:
the postinst (mentioned by @dangowrt) seems different. that is using file with suffix "postinst-pkg" instead of "postinst"
also there are some, different formats for postinstall-scripts:

root@211-49:~# cat /usr/lib/opkg/info/luci-app-olsr.postinst-pkg
[ -n "${IPKG_INSTROOT}" ] || {
        (. /etc/uci-defaults/40_luci-olsr) && rm -f /etc/uci-defaults/40_luci-olsr
        exit 0
}
root@211-49:~# cat /usr/lib/opkg/info/uhttpd-mod-ubus.postinst
#!/bin/sh
[ "${IPKG_NO_SCRIPT}" = "1" ] && exit 0
[ -x ${IPKG_INSTROOT}/lib/functions.sh ] || exit 0
. ${IPKG_INSTROOT}/lib/functions.sh
default_postinst $0 $@
root@211-49:~# reboot -f

aparcar, 2017-11-02:
This problem exists only in snapshot and works fine on 17.01-snapshot (aka branch lede-17.1).

closed 2017-11-02 by aparcar