openwrt/asu #60: License notice on web ui

dangowrt, 2017-10-25:
A note about the license of the distributed software should be included on the web frontend.
In the future:
To predict the licenses of a certain image we need
 - teach ImageBuilder `make package_list` to output the license as well as dependencies for each package
 - package dependency tracking and reference [SPDX](https://spdx.org/licenses/) in database

We might also allow users of the IB webui to force a specific set of allowed licenses... But that's far future :)

FreifunkUFO, 2017-10-25:
its asoociäted to https://github.com/aparcar/gsoc17-attended-sysupgrade/issues/52

aparcar, 2017-10-26:
first part looks like a job for you @dangowrt . Once the IB has this new function I'll integrate it into the database

aparcar, 2018-08-07:
related https://forum.openwrt.org/t/add-spdx-license-to-opkg-output/18415/2

aparcar, 2020-02-25:
This really depends on an opkg patch https://patchwork.ozlabs.org/patch/1115478/