openwrt/asu #62: problems with LEDE SNAPSHOTS on x86-vm

FreifunkUFO, 2017-10-29:
the possibility for upgrading LEDE snapshots is missing at the moment.

error: unknown device, please check model and board params

10.61.143.13, x86-vm, previously att-sysupgrade was already running a few times (some days ago)

FreifunkUFO, 2017-10-29:
Attended Sysupgrade

new update available

SNAPSHOT to SNAPSHOT

-> request image..

unknown device, please check model and board params

```

version | SNAPSHOT
-- | --
board | qemu-standard-pc-i440fx-piix-1996
packages | […]
0 | base-files
1 | busybox
2 | dropbear
3 | mtd
4 | uci
5 | opkg
6 | netifd
7 | fstools
8 | uclient-fetch
9 | logd
10 | dnsmasq
11 | iptables
12 | ip6tables
13 | firewall
14 | odhcpd
15 | odhcp6c
16 | kmod-via-velocity
17 | cfdisk
18 | collectd-mod-conntrack
19 | collectd-mod-exec
20 | collectd-mod-interface
21 | collectd-mod-load
22 | collectd-mod-olsrd
23 | collectd-mod-ping
24 | e2fsprogs
25 | fdisk
26 | freifunk-common
27 | freifunk-policyrouting
28 | hostapd-mini
29 | ip6tables-mod-nat
30 | kmod-3c59x
31 | kmod-8139too
32 | kmod-button-hotplug
33 | kmod-e100
34 | kmod-e1000
35 | kmod-natsemi
36 | kmod-ne2k-pci
37 | kmod-pcnet32
38 | kmod-r8169
39 | kmod-sis900
40 | kmod-tg3
41 | kmod-via-rhine
42 | libiwinfo-lua
43 | luci
44 | luci-app-firewall
45 | luci-app-freifunk-policyrouting
46 | luci-app-freifunk-widgets
47 | luci-app-statistics
48 | luci-i18n-base-de
49 | luci-i18n-firewall-de
50 | luci-i18n-freifunk-de
51 | luci-i18n-freifunk-policyrouting-de
52 | luci-i18n-olsr-de
53 | luci-i18n-splash-de
54 | luci-mod-admin-full
55 | luci-mod-freifunk
56 | luci-mod-freifunk-community
57 | luci-theme-freifunk-generic
58 | mc
59 | mkf2fs
60 | olsrd-mod-jsoninfo
61 | partx-utils
62 | uhttpd
63 | kernel
64 | rpcd-mod-packagelist
65 | luci-app-attendedsysupgrade
66 | rpcd-mod-attendedsysupgrade
model | QEMU Standard PC (i440FX + PIIX, 1996)
distro | LEDE
target | x86
subtarget | generic
```

FreifunkUFO, 2017-10-30:
`
root@143-13-test:~# cat /etc/openwrt_*
DISTRIB_ID='LEDE'
DISTRIB_RELEASE='SNAPSHOT'
DISTRIB_REVISION='r5117-fbde9ac'
DISTRIB_CODENAME='reboot'
DISTRIB_TARGET='x86/generic'
DISTRIB_ARCH='i386_pentium4'
DISTRIB_DESCRIPTION='LEDE Reboot SNAPSHOT r5117-fbde9ac'
DISTRIB_TAINTS=''
r5117-fbde9ac
`

aparcar, 2017-10-31:
This was may caused be an outdated worker. Please try again and close on success.

aparcar, 2017-11-03:
Fixed here 114aad9ff9e4795c8a08ec17ec81962ef1627423

Pleas close issue

closed 2017-11-12 by aparcar