openwrt/asu #63: change the config file and settings

FreifunkUFO, 2017-10-29:
the naming of the settings is confusing (upgrade vs. update).

FreifunkUFO, 2017-10-29:
you already did some changes: https://github.com/aparcar/luci/commit/9e1728cd625b841cb35de5191d82639f7bd52286

FreifunkUFO, 2017-10-29:
i think that "rename url to server_url" is not needed, because "url" is coupled to value "updateserver" already.

`
root@211-41:~# uci get attendedsysupgrade.updateserver.url
https://betaupdate.libremesh.org

`

FreifunkUFO, 2017-10-29:
would be nice to change the filename 🥇 
because project/packagename is with "-", so the filename should also have a "-". other configs already have "-" or "_" 

`root@211-41:~# cat /etc/config/
attendedsysupgrade      freifunk-widgets        olsrd
batman-adv              horst                   olsrd6
dhcp                    luci                    profile_leipzig
dropbear                luci_olsr               rpcd
firewall                luci_splash             system
freifunk                luci_splash_leases      ubootenv
freifunk-gwcheck        luci_statistics         ucitrack
freifunk-mapupdate      meshkit                 uhttpd
freifunk-policyrouting  meshwizard              wireless
freifunk-watchdog       network
`

FreifunkUFO, 2017-10-30:
please honour ticket https://github.com/aparcar/gsoc17-attended-sysupgrade/issues/64 . would be nice to make it now, when config is even changing.

FreifunkUFO, 2017-10-30:
the new feature "search on opening the view without clicking" is mad. nowhere else on menu of luci-components there is an internet-access without explicitly asking.
would be nicer to see the url/description of the choosed server near the search-button ( or -in the future -  a list of possible servers #64 )

FreifunkUFO, 2017-10-30:
next detail: if you "rename updateserver to upgradeserver"
you should even more simpily the config, and name it just "server" :-o

aparcar, 2017-10-30:
> the new feature "search on opening the view without clicking" is mad
@jow- is that mad? I remember we discussed once an upgrade check adding a banner to the login view in case there is an upgrade. That would require "search on opening" as well.

aparcar, 2017-10-30:
> would be nice to change the filename 🥇

at some situations a dash does not work, we decided to unify it to attendedsysupgrade instead of a variation of attended_sysupgrade, attended-sysupgrade and finally attendedsysupgrade

aparcar, 2017-10-30:
> i think that "rename url to server_url" is not needed, because "url" is coupled to value "updateserver" already.

this should be possible

generally, please open single issues per proposed change as these may result in individual discussions :) thanks

FreifunkUFO, 2017-11-02:
its done here https://github.com/openwrt/luci/commit/e8259af9ce7cb442865d583d2c5b73acff1691ce#diff-b5fba9ff24d0045d1377a05a46b32f68

FreifunkUFO, 2017-11-02:
a wrong server was set: https://github.com/aparcar/packages/blob/1864a76e9cd34020bf10b59e3b794beda02a6186/utils/rpcd-mod-attendedsysupgrade/files/attendedsysupgrade.defaults

aparcar, 2017-11-02:
that was requested by @jow-

FreifunkUFO, 2017-11-03:
new uci options: https://github.com/openwrt/packages/pull/5053

aparcar, 2017-11-12:
New config options are upstream

closed 2017-11-12 by aparcar