openwrt/asu #64: settings: change the server-URL to a list of servers

FreifunkUFO, 2017-10-30:
change the url-value to an uci-list. i saw there are now at least two different servers online..
also a Description for the different URLs would be nice. and there is a need for a nice default list.

FreifunkUFO, 2017-10-30:
would be nice to make it now (when introducing other changes to the config, and BEFORE promoting this tool) instead of implementing it later.

a list is needed because openwrt/lede is a open source project with many servers available. also fake-for adresses  (f.i. olsr-hna)  would be nice for offline-usage of this tool.

FreifunkUFO, 2017-10-30:
see f.i. example

`
root@211-41:/etc/config# uci get freifunk-widgets.example_search.engine
Google|http://www.google.de/search?q= Freifunk&#160;Wiki|http://wiki.freifunk.ne                          t/index.php?search=

config widget 'example_search'
        option template 'search'
        option enabled '0'
        option title 'Search'
        list engine 'Google|http://www.google.de/search?q='
        list engine 'Freifunk&#160;Wiki|http://wiki.freifunk.net/index.php?searc                          h='
        option width '50%'
        option paddingright '8%'

`

FreifunkUFO, 2017-11-03:
im now "assigned"? what does it mean and what do you think about this ticket and urgency?

aparcar, 2017-11-03:
missclicked, my turn. But I'd highly appreciate help

It's somewhat the last issue keeping the app from being really functional. So it's the next thing to do

aparcar, 2017-11-25:
This has been fixed in luci-app-attendedsysupgrade "v3". It's not a list but an easy to set domain. 

@FreifunkUFO I don't see the need for a complex list, eventually a default like `update.lede-project.org` should be used as the the server makes the more sense the more people use it. 
For experimental changes the textfield should work fine. Objections?

closed 2017-11-25 by aparcar