openwrt/asu #65: problems with LEDE snapshot on UBNT bullet

FreifunkUFO, 2017-10-30:
some problem when requesting the firmware-image

FreifunkUFO, 2017-10-30:
```
[
    "lede",
    "snapshot",
    "ar71xx",
    "generic",
    "ubnt-bullet-m",
    "ba9832af6360",
    ""
]

Checking 'working-make'... ok.
Checking 'case-sensitive-fs'... ok.
Checking 'proper-umask'... ok.
Checking 'gcc'... ok.
Checking 'working-gcc'... ok.
Checking 'g++'... ok.
Checking 'working-g++'... ok.
Checking 'ncurses'... ok.
Checking 'zlib'... ok.
Checking 'perl-thread-queue'... ok.
Checking 'tar'... ok.
Checking 'find'... ok.
Checking 'bash'... ok.
Checking 'patch'... ok.
Checking 'diff'... ok.
Checking 'cp'... ok.
Checking 'seq'... ok.
Checking 'awk'... ok.
Checking 'grep'... ok.
Checking 'getopt'... ok.
Checking 'stat'... ok.
Checking 'unzip'... ok.
Checking 'bzip2'... ok.
Checking 'wget'... ok.
Checking 'perl'... ok.
Checking 'python'... ok.
Checking 'git'... ok.
Checking 'file'... ok.
Checking 'ldconfig-stub'... ok.
Building images for ar71xx - Ubiquiti Bullet-M
Packages: base-files batctl busybox dnsmasq dropbear firewall freifunk-common freifunk-policyrouting fstools horst hostapd-mini ip6tables ip6tables-mod-nat iptables kernel kmod-ath9k kmod-batman-adv kmod-gpio-button-hotplug kmod-usb-core kmod-usb-ohci kmod-usb2 libiwinfo-lua logd luci-app-attendedsysupgrade luci-app-firewall luci-app-freifunk-policyrouting luci-app-freifunk-widgets luci-i18n-base-de luci-i18n-firewall-de luci-i18n-freifunk-de luci-i18n-freifunk-policyrouting-de luci-i18n-meshwizard-de luci-i18n-olsr-de luci-i18n-splash-de luci-mod-admin-full luci-mod-freifunk luci-mod-freifunk-community luci-theme-freifunk-generic meshwizard mtd netifd odhcp6c odhcpd olsrd-mod-jsoninfo opkg rpcd-mod-attendedsysupgrade rpcd-mod-packagelist swconfig tcpdump-mini uboot-envtools uci uclient-fetch uhttpd base-files busybox dnsmasq dropbear firewall fstools ip6tables iptables kernel kmod-ath9k kmod-gpio-button-hotplug kmod-usb-core kmod-usb-ohci kmod-usb2 logd mtd netifd odhcp6c odhcpd opkg swconfig uboot-envtools uci uclient-fetch

Package list missing or not up-to-date, generating it.

Building package index...
Collected errors:
 * opkg_conf_load: Could not lock /srv/imagebuilder/lede/snapshot/ar71xx/generic/build_dir/target-mips_24kc_musl/root-ar71xx//tmp/opkg.lock: Resource temporarily unavailable.

Installing packages...
Collected errors:
 * opkg_conf_load: Could not lock /srv/imagebuilder/lede/snapshot/ar71xx/generic/build_dir/target-mips_24kc_musl/root-ar71xx//tmp/opkg.lock: Resource temporarily unavailable.
Makefile:146: recipe for target 'package_install' failed
make[2]: *** [package_install] Error 255
Makefile:109: recipe for target '_call_image' failed
make[1]: *** [_call_image] Error 2
Makefile:202: recipe for target 'image' failed
make: *** [image] Error 2
```

FreifunkUFO, 2017-10-30:
the next try (with different packagelist!!) was successfully :-)

Attended Sysupgrade
image created

Keep settings

FreifunkUFO, 2017-10-30:
for that file i was deleting "horst" on the packagelist.

 now, with latest firmware i can make another att-sysupgrade.. when i add "horst" to it (see below) i still get that old error message!

aparcar, 2017-10-31:
This was may caused be an outdated worker. Please try again and close on success.

FreifunkUFO, 2017-11-01:
same problem today:

[
    "lede",
    "snapshot",
    "ar71xx",
    "generic",
    "ubnt-bullet-m",
    "35e8331ea080",
    ""
]

Building images for ar71xx - Ubiquiti Bullet-M
Packages: base-files batctl busybox dnsmasq dropbear firewall freifunk-common freifunk-policyrouting fstools hostapd-mini ip6tables ip6tables-mod-nat iptables kernel kmod-ath9k kmod-batman-adv kmod-gpio-button-hotplug kmod-usb-core kmod-usb-ohci kmod-usb2 libiwinfo-lua logd luci-app-attendedsysupgrade luci-app-firewall luci-app-freifunk-policyrouting luci-app-freifunk-widgets luci-i18n-base-de luci-i18n-firewall-de luci-i18n-freifunk-de luci-i18n-freifunk-policyrouting-de luci-i18n-meshwizard-de luci-i18n-olsr-de luci-i18n-splash-de luci-mod-admin-full luci-mod-freifunk luci-mod-freifunk-community luci-theme-freifunk-generic meshwizard mtd netifd odhcp6c odhcpd olsrd-mod-jsoninfo opkg rpcd-mod-attendedsysupgrade rpcd-mod-packagelist swconfig uboot-envtools uci uclient-fetch uhttpd base-files busybox dnsmasq dropbear firewall fstools ip6tables iptables kernel kmod-ath9k kmod-gpio-button-hotplug kmod-usb-core kmod-usb-ohci kmod-usb2 logd mtd netifd odhcp6c odhcpd opkg swconfig uboot-envtools uci uclient-fetch

Collected errors:
 * opkg_conf_load: Could not lock /srv/imagebuilder/lede/snapshot/ar71xx/generic/build_dir/target-mips_24kc_musl/root-ar71xx//tmp/opkg.lock: Resource temporarily unavailable.

FreifunkUFO, 2017-11-01:
now i was adding "horst" to the package list and now there is success in creating an image.

[
    "lede",
    "snapshot",
    "ar71xx",
    "generic",
    "ubnt-bullet-m",
    "5366ece999e2",
    ""
]

Building images for ar71xx - Ubiquiti Bullet-M
Packages: base-files batctl busybox dnsmasq dropbear firewall freifunk-common freifunk-policyrouting fstools horst hostapd-mini ip6tables ip6tables-mod-nat iptables kernel kmod-ath9k kmod-batman-adv kmod-gpio-button-hotplug kmod-usb-core kmod-usb-ohci kmod-usb2 libiwinfo-lua logd luci-app-attendedsysupgrade luci-app-firewall luci-app-freifunk-policyrouting luci-app-freifunk-widgets luci-i18n-base-de luci-i18n-firewall-de luci-i18n-freifunk-de luci-i18n-freifunk-policyrouting-de luci-i18n-meshwizard-de luci-i18n-olsr-de luci-i18n-splash-de luci-mod-admin-full luci-mod-freifunk luci-mod-freifunk-community luci-theme-freifunk-generic meshwizard mtd netifd odhcp6c odhcpd olsrd-mod-jsoninfo opkg rpcd-mod-attendedsysupgrade rpcd-mod-packagelist swconfig uboot-envtools uci uclient-fetch uhttpd base-files busybox dnsmasq dropbear firewall fstools ip6tables iptables kernel kmod-ath9k kmod-gpio-button-hotplug kmod-usb-core kmod-usb-ohci kmod-usb2 logd mtd netifd odhcp6c odhcpd opkg swconfig uboot-envtools uci uclient-fetch

aparcar, 2017-11-03:
The restart mechanism was broken. I tried the requests and it builds fine.

closed 2017-11-11 by aparcar