openwrt/asu #66: problems with LEDE snapshot and the packagelist

FreifunkUFO, 2017-11-01:
i tried attended-sysupgrade today several times on a meshkit-node (with 3 days old trunk).

on the third upgrade i lost network-contact. the att-sysupgrade did not take packelist / all my routing-protocols to the next image.

FreifunkUFO, 2017-11-01:
[
    "lede",
    "snapshot",
    "ar71xx",
    "generic",
    "ubnt-bullet-m",
    "5366ece999e2",
    ""
]

Building images for ar71xx - Ubiquiti Bullet-M
Packages: base-files batctl busybox dnsmasq dropbear firewall freifunk-common freifunk-policyrouting fstools horst hostapd-mini ip6tables ip6tables-mod-nat iptables kernel kmod-ath9k kmod-batman-adv kmod-gpio-button-hotplug kmod-usb-core kmod-usb-ohci kmod-usb2 libiwinfo-lua logd luci-app-attendedsysupgrade luci-app-firewall luci-app-freifunk-policyrouting luci-app-freifunk-widgets luci-i18n-base-de luci-i18n-firewall-de luci-i18n-freifunk-de luci-i18n-freifunk-policyrouting-de luci-i18n-meshwizard-de luci-i18n-olsr-de luci-i18n-splash-de luci-mod-admin-full luci-mod-freifunk luci-mod-freifunk-community luci-theme-freifunk-generic meshwizard mtd netifd odhcp6c odhcpd olsrd-mod-jsoninfo opkg rpcd-mod-attendedsysupgrade rpcd-mod-packagelist swconfig uboot-envtools uci uclient-fetch uhttpd base-files busybox dnsmasq dropbear firewall fstools ip6tables iptables kernel kmod-ath9k kmod-gpio-button-hotplug kmod-usb-core kmod-usb-ohci kmod-usb2 logd mtd netifd odhcp6c odhcpd opkg swconfig uboot-envtools uci uclient-fetch

p.s. same device (and normal packagelist for that device) as https://github.com/aparcar/gsoc17-attended-sysupgrade/issues/65

FreifunkUFO, 2017-11-01:
[
    "lede",
    "snapshot",
    "ar71xx",
    "generic",
    "ubnt-bullet-m",
    "5588e593e7f9",
    ""
]

Building images for ar71xx - Ubiquiti Bullet-M
Packages: base-files busybox dnsmasq dropbear firewall fstools ip6tables iptables kmod-ath9k kmod-gpio-button-hotplug kmod-usb-core kmod-usb-ohci kmod-usb2 libc libgcc logd mtd netifd odhcp6c odhcpd opkg ppp ppp-mod-pppoe swconfig uboot-envtools uci uclient-fetch wpad-mini base-files busybox dnsmasq dropbear firewall fstools ip6tables iptables kernel kmod-ath9k kmod-gpio-button-hotplug kmod-usb-core kmod-usb-ohci kmod-usb2 libc libgcc logd mtd netifd odhcp6c odhcpd opkg ppp ppp-mod-pppoe swconfig uboot-envtools uci uclient-fetch wpad-mini

p.s. same device (and normal packagelist for that device) as #65

FreifunkUFO, 2017-11-01:
sooo, next try.. with meshkit firmware:

first att-sysupgrade:

[
    "lede",
    "snapshot",
    "ar71xx",
    "generic",
    "ubnt-bullet-m",
    "5366ece999e2",
    ""
]

Building images for ar71xx - Ubiquiti Bullet-M
Packages: base-files batctl busybox dnsmasq dropbear firewall freifunk-common freifunk-policyrouting fstools horst hostapd-mini ip6tables ip6tables-mod-nat iptables kernel kmod-ath9k kmod-batman-adv kmod-gpio-button-hotplug kmod-usb-core kmod-usb-ohci kmod-usb2 libiwinfo-lua logd luci-app-attendedsysupgrade luci-app-firewall luci-app-freifunk-policyrouting luci-app-freifunk-widgets luci-i18n-base-de luci-i18n-firewall-de luci-i18n-freifunk-de luci-i18n-freifunk-policyrouting-de luci-i18n-meshwizard-de luci-i18n-olsr-de luci-i18n-splash-de luci-mod-admin-full luci-mod-freifunk luci-mod-freifunk-community luci-theme-freifunk-generic meshwizard mtd netifd odhcp6c odhcpd olsrd-mod-jsoninfo opkg rpcd-mod-attendedsysupgrade rpcd-mod-packagelist swconfig uboot-envtools uci uclient-fetch uhttpd base-files busybox dnsmasq dropbear firewall fstools ip6tables iptables kernel kmod-ath9k kmod-gpio-button-hotplug kmod-usb-core kmod-usb-ohci kmod-usb2 logd mtd netifd odhcp6c odhcpd opkg swconfig uboot-envtools uci uclient-fetch

---

next att-sysupgrade... was not rendered because server said its stil existing:

[
    "lede",
    "snapshot",
    "ar71xx",
    "generic",
    "ubnt-bullet-m",
    "5366ece999e2",
    ""
]

Building images for ar71xx - Ubiquiti Bullet-M
Packages: base-files batctl busybox dnsmasq dropbear firewall freifunk-common freifunk-policyrouting fstools horst hostapd-mini ip6tables ip6tables-mod-nat iptables kernel kmod-ath9k kmod-batman-adv kmod-gpio-button-hotplug kmod-usb-core kmod-usb-ohci kmod-usb2 libiwinfo-lua logd luci-app-attendedsysupgrade luci-app-firewall luci-app-freifunk-policyrouting luci-app-freifunk-widgets luci-i18n-base-de luci-i18n-firewall-de luci-i18n-freifunk-de luci-i18n-freifunk-policyrouting-de luci-i18n-meshwizard-de luci-i18n-olsr-de luci-i18n-splash-de luci-mod-admin-full luci-mod-freifunk luci-mod-freifunk-community luci-theme-freifunk-generic meshwizard mtd netifd odhcp6c odhcpd olsrd-mod-jsoninfo opkg rpcd-mod-attendedsysupgrade rpcd-mod-packagelist swconfig uboot-envtools uci uclient-fetch uhttpd base-files busybox dnsmasq dropbear firewall fstools ip6tables iptables kernel kmod-ath9k kmod-gpio-button-hotplug kmod-usb-core kmod-usb-ohci kmod-usb2 logd mtd netifd odhcp6c odhcpd opkg swconfig uboot-envtools uci uclient-fetch

---

jetzt ein sys-att-upgrade mit zusötzlichem paket (damit der server neu baut)

[
    "lede",
    "snapshot",
    "ar71xx",
    "generic",
    "ubnt-bullet-m",
    "ba9832af6360",
    ""
]

Building images for ar71xx - Ubiquiti Bullet-M
Packages: base-files batctl busybox dnsmasq dropbear firewall freifunk-common freifunk-policyrouting fstools horst hostapd-mini ip6tables ip6tables-mod-nat iptables kernel kmod-ath9k kmod-batman-adv kmod-gpio-button-hotplug kmod-usb-core kmod-usb-ohci kmod-usb2 libiwinfo-lua logd luci-app-attendedsysupgrade luci-app-firewall luci-app-freifunk-policyrouting luci-app-freifunk-widgets luci-i18n-base-de luci-i18n-firewall-de luci-i18n-freifunk-de luci-i18n-freifunk-policyrouting-de luci-i18n-meshwizard-de luci-i18n-olsr-de luci-i18n-splash-de luci-mod-admin-full luci-mod-freifunk luci-mod-freifunk-community luci-theme-freifunk-generic meshwizard mtd netifd odhcp6c odhcpd olsrd-mod-jsoninfo opkg rpcd-mod-attendedsysupgrade rpcd-mod-packagelist swconfig tcpdump-mini uboot-envtools uci uclient-fetch uhttpd base-files busybox dnsmasq dropbear firewall fstools ip6tables iptables kernel kmod-ath9k kmod-gpio-button-hotplug kmod-usb-core kmod-usb-ohci kmod-usb2 logd mtd netifd odhcp6c odhcpd opkg swconfig uboot-envtools uci uclient-fetch

--

aparcar, 2017-11-03:
I tried all combinations and they build fine. Likely the snapshot had a problem and caused the error. Please try daily snapshots first on you local machine before reporting them as an error of the server.

closed 2017-11-03 by aparcar