openwrt/asu #67: problem: LEDE-snapshot: reboot needed before using attended-sysupgrade

FreifunkUFO, 2017-11-03:
the postinstall scripts seems to have problems:
when installing atrtended-sysupgrade on openwrt/lede

see https://github.com/aparcar/gsoc17-attended-sysupgrade/issues/58 
and postinstall script for /etc/uci-defaults..

aparcar, 2017-11-03:
That's an issue of the current snapshot and not of the `uci-defaults` script used by [rpcd-mod-attendedsysupgrade](https://github.com/openwrt/packages/tree/master/utils/rpcd-mod-attendedsysupgrade). 

I tried (as written before) the script with the current *stable* release 17.01.4 and it works as expected.

aparcar, 2017-11-03:
I'll kindly as you to close this issue once okay for you. As well, please report this bug to upstream as it seems to affect multiple packages. I can't find it right now but I think `uhttpd-mod-something` had a similar problem, printing "line 130 uhttpd-mod-something: not found"

aparcar, 2017-11-03:
Found the line [here](https://github.com/aparcar/gsoc17-attended-sysupgrade/issues/54)

>Configuring uhttpd-mod-ubus.
//usr/lib/opkg/info/uhttpd-mod-ubus.postinst: .: line 130: 00_uhttpd_ubus: not f ound

dangowrt, 2017-11-09:
Fixed upstream by
https://github.com/lede-project/source/commit/73c745f64f17a0326c53e614aa0dd125b47053ec

closed 2017-11-09 by dangowrt