openwrt/asu #72: deprecate snapshot support

aparcar, 2017-11-05:
Supporting snapshots brings a whole lot of problems and always leading to debugging if it's a problem of the server or the daily snapshot imagebuilder. As snapshots are likely only for development purpose I'd like to remove all snapshot support from the server.

The server is not meant for snapshot testing as official buildbots already do this.

aparcar, 2017-11-05:
@FreifunkUFO @dangowrt

dangowrt, 2017-11-05:
ACK

aparcar, 2017-11-09:
@FreifunkUFO comments?

FreifunkUFO, 2017-11-14:
NO ACK.
please dont delete snapshot/trunk support. of course, trunk sometimes have some problems. 
 
i saw that in my testings, when LEDE was just disabling my test-hardware (cpe510, 902ac) ( see https://github.com/aparcar/attendedsysupgrade-server/issues/55 and https://github.com/aparcar/attendedsysupgrade-server/issues/57 ) or when post-install-start wasnt working ( https://github.com/aparcar/attendedsysupgrade-server/issues/67 ) 

but, supporting snapshots is very nice for development purposes. ("official buildbots" cant "do this"). it much more better then compiling own lede-snapshots over night, because it uses the "normal" snapshot files and packages

aparcar, 2017-11-14:
as @dangowrt changed his mind too I'll keep it for now

closed 2017-11-14 by aparcar

dangowrt, 2017-11-14:
Just make sure that:
 * only ugrades from SNAPSHOT -> SNAPSHOT are possible, updates from RELEASE -> SNAPSHOT may be cool for pre-release testing but should by no means be available for normal users
 * chef should not offer SNAPSHOT builds
 * clearly exclude snapshot support from bug tracker. You use snapshots -> bug invalid.

egon0, 2017-11-14:
+1 for everything dangowrt just wrote. so we playing kids could have some time-consuming fun with snapshots but the "normal user" sticks to release builds. i think this is a great compromise.

FreifunkUFO, 2017-11-20:
-1 for everything dangowrt just wrote.

aparcar, 2017-11-20:
@FreifunkUFO 👎 or 👍 ?

* release -> snapshot is always manually possible by editing `/etc/openwrt_release`
* [chef](https://chef.libremsh.org) could support snapshots with a warning in the advanced view
* snapshot supports creates a lot of confusion (for me) when snapshot bugs are reported as asu bugs 😒

dangowrt, 2017-11-20:
@FreifunkUFO no offense, but someone has to do the work and skim through the tickets created due to snapshots failing despite that most of the time the cause is a problem with the snapshot codebase rather than the sysupgrade mechanics (we just wasted a week trying to fix uci-defaults, then noticed that the it's a problem with the LEDE snapshot caused by busybox bash recently becoming a bit more strict). Any constructive crtiticism is still welcome, ie. name what exactly bothers you suggest what we could do it make it better.