openwrt/asu #82: How to use this with LEDE?

claell, 2017-11-20:
I am interested in using this for my LEDE installation, but don't know exactly how to integrate it. Do I simply need to install `luci-app-attendedsysupgrade` in LEDE?

Also since this seems pretty helpful, are there plans to integrate it directly into all LEDE installations?

aparcar, 2017-11-20:
Thanks for your interest. Yes, currently you can install `luci-app-attendedsysupgrade` or `auc` (for cli) from a snapshot build.

There have been some recent affords to bring this into the stable repository of LEDE, but there is still some work to do. This step would make it easy installable via opkg. 

>Also since this seems pretty helpful, are there plans to integrate it directly into all LEDE installations?

I'd really like the idea but as this is still a young project the server would just die if used on high scale. Currently I'm trying to improve the performance to be eventually suitable for an official service.

claell, 2017-11-21:
Ah ok. Currently I am using the stable version, so I will wait until it becomes available there.

So you are running an own server for this? I thought it could simply use the LEDE servers.

aparcar, 2018-04-01:
@claell yes this runs on a private VM. However if it's ever stable and scales, I'll ask to run this on official servers.

I'll close the issue as it should be resolved?

closed 2018-04-01 by aparcar

claell, 2018-04-08:
Yes, you can close this. Is this in the official repositories already, or do I have to wait a bit?

claell, 2018-04-10:
Alright, will wait for it then