openwrt/asu #86: add server's usign and pgp keys to attendedsysupgrade-common

dangowrt, 2017-12-25:
Store server pubkey (both usign and pgp) on target (ie. in `/etc`), so future upgrades can be verified.

aparcar, 2017-12-25:
what are good suffixes for usign and pgp?

dangowrt, 2017-12-25:
when stored on the target FS that doesn't really matter (file suffixes do matter for auto-mime-detecting web-servers and windows-based systems only)

dangowrt, 2018-02-23:
In the current implementation, only usign pubkeys are needed on the target.
The files should be named according to the hash of the pubkey and placed in `/etc/opkg/keys`, see 
https://git.openwrt.org/?p=keyring.git;a=tree;f=usign;hb=HEAD for what ends up to be the lede-keyring package.

aparcar, 2019-04-06:
Fixed via https://github.com/openwrt/packages/commit/54a2c8d087e65ce5389586b62b2ec41f00c0c8c0

closed 2019-04-06 by aparcar