openwrt/asu #87: trunk: Failed to execute template dispatcher target for entry '/admin/system/attended_sysupgrade'

FreifunkUFO, 2018-01-12:
on a c59 with meshkit (lede-trunk from yesterday) i got an error:

http://10.61.93.111/cgi-bin/luci//admin/system/attended_sysupgrade/

/usr/lib/lua/luci/dispatcher.lua:478: Failed to execute template dispatcher target for entry '/admin/system/attended_sysupgrade'.
The called action terminated with an exception:
/usr/lib/lua/luci/template.lua:97: Failed to execute template 'attendedsysupgrade'.
A runtime error occured: [string "/usr/lib/lua/luci/view/attendedsysupgrade.h..."]:12: bad argument #1 to 'parse' (string expected, got nil)
stack traceback:
	[C]: in function 'assert'
	/usr/lib/lua/luci/dispatcher.lua:478: in function 'dispatch'
	/usr/lib/lua/luci/dispatcher.lua:121: in function </usr/lib/lua/luci/dispatcher.lua:120>

aparcar, 2018-01-16:
Can you reproduce the error with an locally installed ImageBuilder? Does this error appear in stable?

FreifunkUFO, 2018-01-16:
i dont know how to reproduce. maybe someone should test it with "normal" lede-trunk?

as said before: my test was on "locally installed" (on freifunk-leipzig meshkit server) imagebuilder (aka meshkit), where the imagebuilder also was "locally built" (on 11.1.2018).

what does "stable" means, is it lede-17.01.4?
i never tested that version.

aparcar, 2018-04-01:
@FreifunkUFO is this still relevant?

aparcar, 2018-08-03:
Closed due to no response

closed 2018-08-03 by aparcar