openwrt/asu #88: Any reason why RPI3 is not supported?

tlaurion, 2018-01-12:
Setting luci's Attended Sysupgrade to https://ledeupdate.planetexpress.cc/ results in an error.

Actual behavior:
Unsupported device, release, target, subtraget or board

tlaurion, 2018-01-12:
Sorry. https://ledeupdate.planetexpress.cc/ doesn't support RPI3, not the project

dangowrt, 2018-01-12:
The true cause for this is that the `brcm2708` target hasn't been converted to use the new image generation code yet. Take a look `x86` or `sunxi` for examples of other block-device (ie. sdcard/emmc and such) platforms which already do that. And maybe file a bug on bugs.lede-project.org so the brcm2708 folks will take notice.

aparcar, 2018-01-16:
@tlaurion could you create a bug on upstream and close this issue? It's not really related to the server which only uses existing image builders

tlaurion, 2018-01-26:
Posted [here](https://bugs.lede-project.org/index.php?do=details&task_id=1303)

tlaurion, 2018-01-31:
[Christian Lamparter commented on 26.01.2018 21:34](https://bugs.lede-project.org/index.php?do=details&task_id=1303):

> Hm, too bad you closed the issue report.
> 
> "The true cause for this is that the brcm2708 target hasn't been converted to use the new image generation code yet. Take a look x86 or sunxi for examples of other block-device (ie. sdcard/emmc and such) platforms which already do that."
> 
> The brcm2708 target in master/trunk has been converted to use the new image generation code in trunk/master! This was done in patch:
> brcm2708: convert to metadata
> 
> That said, I tested the luci-app-attendedsysupgrade app on my Netgear WNDR3700v2 and got the same error message: "Unsupported device, release, target, subtraget or board".

tlaurion, 2018-01-31:
@aparcar ?

aparcar, 2018-01-31:
@tlaurion I responded, the problem is a missing squasfs image. Maybe it can be easily added - let's see

tlaurion, 2018-02-07:
@aparcar : Any update?

aparcar, 2018-02-07:
@tlaurion no, as I'm not the one that can push anything. The squasfs filesystem is upgradeable as it's stores files in an overlay filesystem while the current images uses ext4. Reflashing ext4 results in a total overwrite of all custom modified files and essentially *resets* your device - so not supported.
As far as I can tell it's possible to modify the build system that there are ext4 **and** squasfs files, see x86/64. However until that's not the case, it's unsupported :(

chunkeey, 2018-02-09:
@aparcar 

All brcm2708 targets in snapshot already use the "same sysupgrade code as x86/64".

openwrt/openwrt@246916ddf4a1a95fd063e6d7c00d73bb0b5f36a6

But again, this "sysupgrade magic" does not have much to do if this at all. For example the overlay-fs gets completely wiped and formatted when a sysupgrade is performed on the ar71xx-generic/nand ramips too... The way it works for brcm2708 and apm821xx is that the sysupgrade.tgz (CONF_TAR) file that stores just the configuration is generated by "stage2" of sysupgrade. This file gets placed into the new /boot where its safe on the reboot with the new image, that CONF_TAR is move to / and then gets extracted by /lib/preinit/80_mount_root and deleted by /etc/init.d/done. ar71xx/ramips/... do pretty much the same thing with the CONF_TAR but in a slightly different way. it's appended to the 

So, would you please be so kind to just do the two requested changes in
<https://bugs.openwrt.org/index.php?do=details&task_id=1303&order=id&sort=desc>
1. add *.img.gz to sysupgrade-files filter in worker.py
2. add board_name renames for raspberrypi,3-model-b to rpi-3 in the snapshot.yml

And we can work from there and see if anything else is missing. And if it works, the rpi-2 and rpi-1 / cm can be added as well.

Thanks.

aparcar, 2018-02-11:
https://chef.libremesh.org/#c0646f3e5e7b51c is now found, please check if it works for you

aparcar, 2018-02-11:
@chunkeey and thank you very much for the clarification of the sysupgrade process!

dangowrt, 2018-02-11:
To me the reason why it doesn't make sense to support read-write rootfs (ie. ext4, jffs2, ubifs, ...) but only squashfs still is that on systems with r/w rootfs you can simply update via opkg. Even kernel updates via opkg seem more feasible to me than shipping pre-generated ext4-images.

chunkeey, 2018-02-12:
@aparcar 
great. @tlaurion now is your time to shine ;) Can you please test the images Paul created? I gave my RPI-3 away to my brother for a project. But I can test the rpi-b / raspberrypi,model-b. 

@dangowrt 
Well, this is bordering on being "off-topic", right? Have you tried using opkg to update those read-write rootfs installations? You could take x86_64's [combined-ext4.img.gz](http://downloads.lede-project.org/releases/17.01.4/targets/x86/64/lede-17.01.4-x86-64-combined-ext4.img.gz) for a quick test in a kvm or qemu and try to update the base installation to the current snapshot base. What I found was that the naive "opkg upgrade <packagelist>" will fail half-way through since it can't download the packages due to a missing symbols in /bin/wget (uloop_run_timeout). I don't think that opkg alone is capable of performing such a big upgrade, it would need a tool like "apt-get/aptitude/apt".... But then every package would need to take the version of the libraries it depends on in account, so the apt-like tool can sort the packages in the right order. Of course, if you have a solution let me know.

dangowrt, 2018-02-12:
Package upgrades using opkg are rarely usable, that's true. But should that problem be fixed by shipping (and caching and wasting additional CDN resources on) ext4 images for upgrades?
I honestly never understood the need for read-write rootfs images during production use (it _is_ a useful tool during development, work on features which happen in preinit, failsafe and such).
Using read-write images ruins the ability to factory-reset or use snapshots, both features rely on overlayfs and a read-only rootfs (ie. squashfs) underneath.
As the lack of those essential OpenWrt features can well be considered as bug on those target platforms which do not offer them I also do not see the need to work-around other side-effects if the obvious answer to the whole issue is simply to add squashfs images for the sdcard targets which currently lack them.

aparcar, 2018-02-12:
@chunkeey is it complicated to support squashfs on RPIs?

tlaurion, 2018-02-15:
What is the url I  should enter in "Attended Sysupgrade" Luci app?
Putting https://ledeupdate.planetexpress.cc/ still results in

New firmware release availableSNAPSHOT to snapshotPackage upgrades available
*kmod-hid-generic*: 4.9.76-1 to 4.9.77-1
*kmod-hid*: 4.9.76-1 to 4.9.77-1
*kmod-usb-core*: 4.9.76-1 to 4.9.77-1
*kmod-nls-base*: 4.9.76-1 to 4.9.77-1
*kmod-nf-conntrack6*: 4.9.76-1 to 4.9.77-1
*kmod-pppox*: 4.9.76-1 to 4.9.77-1
*kmod-usb-hid*: 4.9.76-1 to 4.9.77-1
*kmod-sound-core*: 4.9.76-1 to 4.9.77-1
*kmod-brcmutil*: 4.9.76+2017-11-01-1 to 4.9.77+2017-11-01-3
*kmod-input-evdev*: 4.9.76-1 to 4.9.77-1
*luci*: git-18.012.27029-5248587-1 to git-18.030.59690-47d1eaa-1
*base-files*: 184-r5952-cfdfa6d to 184-r5985-33e8f7e
*iptables*: 1.6.1-1 to 1.6.1-2
*kmod-input-core*: 4.9.76-1 to 4.9.77-1
*fstools*: 2018-01-02-11efbf3b-1 to 2018-01-13-18090d97-1
*libip6tc*: 1.6.1-1 to 1.6.1-2
*procd*: 2018-12-02-a5954cf3-2 to 2018-01-23-653629f1-2
*kmod-nf-nat*: 4.9.76-1 to 4.9.77-1
*libip4tc*: 1.6.1-1 to 1.6.1-2
*logd*: 2017-11-06-2c0d9cfe-1 to 2017-11-13-e7a63fba-1
*libxtables*: 1.6.1-1 to 1.6.1-2
*kmod-ipt-nat*: 4.9.76-1 to 4.9.77-1
*ip6tables*: 1.6.1-1 to 1.6.1-2
*kmod-pppoe*: 4.9.76-1 to 4.9.77-1
*kmod-nf-ipt*: 4.9.76-1 to 4.9.77-1
*kmod-ip6tables*: 4.9.76-1 to 4.9.77-1
*odhcp6c*: 2017-09-05-51733a6d-4 to 2017-09-05-1f93bd4c-5
*ubus*: 2017-11-13-5f87f548-1 to 2018-01-16-5bae22eb-1
*kmod-nf-ipt6*: 4.9.76-1 to 4.9.77-1
*libubus*: 2017-11-13-5f87f548-1 to 2018-01-16-5bae22eb-1
*kmod-cfg80211*: 4.9.76+2017-11-01-1 to 4.9.77+2017-11-01-3
*kmod-nf-conntrack*: 4.9.76-1 to 4.9.77-1
*kernel*: 4.9.76-1-f1c690668b21139ca18b54e4238aeef6 to
4.9.77-1-c6ed93e30954a5ac73ce66e0d8932dd1
*kmod-nls-cp437*: 4.9.76-1 to 4.9.77-1
*kmod-fs-vfat*: 4.9.76-1 to 4.9.77-1
*kmod-rt2800-usb*: 4.9.76+2017-11-01-1 to 4.9.77+2017-11-01-3
*kmod-nls-utf8*: 4.9.76-1 to 4.9.77-1
*kmod-slhc*: 4.9.76-1 to 4.9.77-1
*luci-app-attendedsysupgrade*: git-18.012.27029-5248587-1 to
git-18.030.59690-47d1eaa-1
*ubox*: 2017-11-06-2c0d9cfe-1 to 2017-11-13-e7a63fba-1
*luci-theme-bootstrap*: git-18.023.74248-ee409b6-1 to
git-18.030.59690-47d1eaa-1
*kmod-ipt-conntrack*: 4.9.76-1 to 4.9.77-1
*kmod-ppp*: 4.9.76-1 to 4.9.77-1
*kmod-brcmfmac*: 4.9.76+2017-11-01-1 to 4.9.77+2017-11-01-3
*kmod-mmc*: 4.9.76-1 to 4.9.77-1
*kmod-lib-crc-ccitt*: 4.9.76-1 to 4.9.77-1
*kmod-nls-iso8859-1*: 4.9.76-1 to 4.9.77-1
*kmod-ipt-core*: 4.9.76-1 to 4.9.77-1
*iwinfo*: 2017-08-23-c1a03e82-1 to 2018-01-16-5a5e21b1-1
*kmod-sound-arm-bcm2835*: 4.9.76-1 to 4.9.77-1

And selecting "Request Firmware" results in:
Unsupported device, release, target, subtraget or board

tlaurion, 2018-02-15:
Just did the following and it screwed up the RPI3 testing device:
UPGRADES=$(opkg list-upgradable | awk -F " " {'print $1'})
opkg install $UPGRADES

Definitely in need of a working attended sysupgrade.

aparcar, 2018-02-15:
@tlaurion I'll look into this tonight. If you can please provide the json send by your browser to the server. Depending on your browser there is a "dev console" that logs all requests and shows the json.

aparcar, 2018-02-15:
```
{
    "version": "snapshot",
    "distro": "openwrt",
    "target": "brcm2708",
    "subtarget": "bcm2710",
    "packages": [
        "kmod-hid-generic", "kmod-hid", "kmod-usb-core", "kmod-nls-base", "kmod-nf-conntrack6", "kmod-pppox", "kmod-usb-hid", "kmod-sound-core", "kmod-brcmutil", "kmod-input-evdev", "luci", "base-files", "iptables", "kmod-input-core", "fstools", "libip6tc", "procd", "kmod-nf-nat", "libip4tc", "logd", "libxtables", "kmod-ipt-nat", "ip6tables", "kmod-pppoe", "kmod-nf-ipt", "kmod-ip6tables", "odhcp6c", "ubus", "kmod-nf-ipt6", "libubus", "kmod-cfg80211", "kmod-nf-conntrack", "kernel", "kmod-nls-cp437", "kmod-fs-vfat", "kmod-rt2800-usb", "kmod-nls-utf8", "kmod-slhc", "luci-app-attendedsysupgrade", "ubox", "luci-theme-bootstrap", "kmod-ipt-conntrack", "kmod-ppp", "kmod-brcmfmac", "kmod-mmc", "kmod-lib-crc-ccitt", "kmod-nls-iso8859-1", "kmod-ipt-core", "iwinfo", "kmod-sound-arm-bcm2835"
    ],
    "board": "rpi-3"
}
```

results in:

```
{
"files": "https://ledeupdate.planetexpress.cc/json/openwrt/snapshot/brcm2708/bcm2710/rpi-3/ec7788893258b50/", 
"log": "https://ledeupdate.planetexpress.cc/static/openwrt/snapshot/brcm2708/bcm2710/rpi-3/ec7788893258b50/build-a5d82415c21a071.log", 
"request_hash": "1ed5ac372319", 
"image_hash": "a5d82415c21a071", 
"sysupgrade": "https://ledeupdate.planetexpress.cc/static/openwrt/snapshot/brcm2708/bcm2710/rpi-3/ec7788893258b50/openwrt-ec7788893258b50-brcm2708-bcm2710-rpi-3-ext4-sdcard.img.gz"
}
```

so I can't reproduce the problem, sorry

aparcar, 2018-02-15:
@tlaurion please post the output of the following

`ubus call system board`
`cat /proc/cpuinfo`

tlaurion, 2018-02-18:
root@OpenWrt:~# ubus call system board
```
{
	"kernel": "4.9.76",
	"hostname": "OpenWrt",
	"model": "Raspberry Pi 3 Model B Rev 1.2",
	"board_name": "raspberrypi,3-model-b",
	"release": {
		"distribution": "OpenWrt",
		"version": "SNAPSHOT",
		"revision": "r5952-cfdfa6d",
		"target": "brcm2708\/bcm2710",
		"description": "OpenWrt SNAPSHOT r5952-cfdfa6d"
	}
}
```
root@OpenWrt:~# cat /proc/cpuinfo
```
processor	: 0
BogoMIPS	: 38.40
Features	: fp asimd evtstrm crc32
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
BogoMIPS	: 38.40
Features	: fp asimd evtstrm crc32
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
BogoMIPS	: 38.40
Features	: fp asimd evtstrm crc32
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
BogoMIPS	: 38.40
Features	: fp asimd evtstrm crc32
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4
```

tlaurion, 2018-02-18:
Request header:
```
POST //api/upgrade-check HTTP/1.1
Host: ledeupdate.planetexpress.cc
User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:52.0) Gecko/20100101 Firefox/52.0
Accept: */*
Accept-Language: en-US,en;q=0.5
Accept-Encoding: gzip, deflate, br
Referer: http://192.168.0.1:8080/cgi-bin/luci/admin/system/attended_sysupgrade
Content-Type: application/json
Content-Length: 2856
Origin: http://192.168.0.1:8080
DNT: 1
Connection: keep-alive

```POST data:`
{"version":"SNAPSHOT","packages":{"iwinfo":"2017-08-23-c1a03e82-1","opkg":"2017-12-07-3b417b9f-2","kmod-usb-core":"4.9.76-1","ubus":"2017-11-13-5f87f548-1","iw":"4.9-1","busybox":"1.27.2-3","kmod-sound-arm-bcm2835":"4.9.76-1","libubus-lua":"2018-01-16-5bae22eb-1","libiwinfo-lua":"2018-01-16-5a5e21b1-1","kmod-rt2800-usb":"4.9.76+2017-11-01-1","libiwinfo":"2018-01-16-5a5e21b1-1","kmod-lib-crc-ccitt":"4.9.76-1","luci-theme-bootstrap":"git-18.023.74248-ee409b6-1","kmod-pppoe":"4.9.76-1","kmod-pppox":"4.9.76-1","kmod-ipt-conntrack":"4.9.76-1","base-files":"184-r5952-cfdfa6d","kmod-nf-nat":"4.9.76-1","kmod-input-core":"4.9.76-1","netifd":"2018-01-24-af3cadb6-2","dnsmasq":"2.78-10","procd":"2018-12-02-a5954cf3-2","libblkid":"2.30.2-2","ubusd":"2018-01-16-5bae22eb-1","kmod-usb-hid":"4.9.76-1","kmod-fs-vfat":"4.9.76-1","firewall":"2017-11-07-c4309372-1","lighttpd":"1.4.48-2","kmod-nf-ipt":"4.9.76-1","kmod-mmc":"4.9.76-1","libuci":"2018-01-01-5beb95da-1","libip4tc":"1.6.1-1","kmod-ip6tables":"4.9.76-1","odhcp6c":"2017-09-05-51733a6d-4","fstools":"2018-01-02-11efbf3b-1","uclient-fetch":"2017-11-02-4b87d831-1","libuclient":"2017-11-02-4b87d831-1","uci":"2018-01-01-5beb95da-1","wpad-mini":"2017-08-24-c2d4f2eb-6","kmod-input-evdev":"4.9.76-1","dropbear":"2017.75-5","brcm2708-gpu-fw":"2017-08-08-e7ba7ab135f5a68b2c00a919ea9ac8d5528a5d5b","kmod-nls-utf8":"4.9.76-1","kmod-nls-cp437":"4.9.76-1","mtd":"21","odhcpd-ipv6only":"1.3-1","libjson-c":"0.12.1-1","kmod-brcmfmac":"4.9.76+2017-11-01-1","libip6tc":"1.6.1-1","ppp":"2.4.7-12","libubox":"2018-01-07-1c08e803-1","kmod-sound-core":"4.9.76-1","logd":"2017-11-06-2c0d9cfe-1","libjson-script":"2018-01-07-1c08e803-1","libblobmsg-json":"2018-01-07-1c08e803-1","iptables":"1.6.1-1","luci-app-attendedsysupgrade":"git-18.012.27029-5248587-1","kmod-brcmutil":"4.9.76+2017-11-01-1","jshn":"2018-01-07-1c08e803-1","kmod-ipt-core":"4.9.76-1","e2fsprogs":"1.43.7-1","kmod-ppp":"4.9.76-1","brcmfmac-firmware-usb":"2017-09-06-a61ac5cf-1","libubus":"2017-11-13-5f87f548-1","kmod-nf-conntrack":"4.9.76-1","usign":"2015-07-04-ef641914-1","libxtables":"1.6.1-1","ip6tables":"1.6.1-1","kmod-hid-generic":"4.9.76-1","kmod-nf-ipt6":"4.9.76-1","kmod-nls-iso8859-1":"4.9.76-1","luci":"git-18.012.27029-5248587-1","kmod-nf-conntrack6":"4.9.76-1","ubox":"2017-11-06-2c0d9cfe-1","lede-keyring":"2017-01-20-a50b7529-1","kernel":"4.9.76-1-f1c690668b21139ca18b54e4238aeef6","libnl-tiny":"0.1-5","kmod-nls-base":"4.9.76-1","fwtool":"1","jsonfilter":"2016-07-02-dea067ad-1","hostapd-common":"2017-08-24-c2d4f2eb-6","kmod-hid":"4.9.76-1","wireless-regdb":"2017-10-20-4343d359","kmod-slhc":"4.9.76-1","kmod-cfg80211":"4.9.76+2017-11-01-1","brcmfmac-firmware-43430-sdio":"2017-09-06-a61ac5cf-1","ppp-mod-pppoe":"2.4.7-12","kmod-ipt-nat":"4.9.76-1"},"upgrade_packages":"1","distro":"OpenWrt","target":"brcm2708","subtarget":"bcm2710"}`

Firmware download request header:```

POST //api/upgrade-request HTTP/1.1
Host: ledeupdate.planetexpress.cc
User-Agent: Mozilla/5.0 (X11; Linux x86_64; rv:52.0) Gecko/20100101 Firefox/52.0
Accept: */*
Accept-Language: en-US,en;q=0.5
Accept-Encoding: gzip, deflate, br
Referer: http://192.168.0.1:8080/cgi-bin/luci/admin/system/attended_sysupgrade
Content-Type: application/json
Content-Length: 2906
Origin: http://192.168.0.1:8080
DNT: 1
Connection: keep-alive

```
Firmware download POST data:`
{"version":"snapshot","board":"raspberrypi,3-model-b","model":"Raspberry Pi 3 Model B Rev 1.2","packages":{"base-files":"184-r5985-33e8f7e","brcm2708-gpu-fw":"2017-08-08-e7ba7ab135f5a68b2c00a919ea9ac8d5528a5d5b","brcmfmac-firmware-43430-sdio":"2017-09-06-a61ac5cf-1","brcmfmac-firmware-usb":"2017-09-06-a61ac5cf-1","busybox":"1.27.2-3","dnsmasq":"2.78-10","dropbear":"2017.75-5","e2fsprogs":"1.43.7-1","firewall":"2017-11-07-c4309372-1","fstools":"2018-01-13-18090d97-1","fwtool":"1","hostapd-common":"2017-08-24-c2d4f2eb-6","ip6tables":"1.6.1-2","iptables":"1.6.1-2","iw":"4.9-1","iwinfo":"2018-01-16-5a5e21b1-1","jshn":"2018-01-07-1c08e803-1","jsonfilter":"2016-07-02-dea067ad-1","kernel":"4.9.77-1-c6ed93e30954a5ac73ce66e0d8932dd1","kmod-brcmfmac":"4.9.77+2017-11-01-3","kmod-brcmutil":"4.9.77+2017-11-01-3","kmod-cfg80211":"4.9.77+2017-11-01-3","kmod-fs-vfat":"4.9.77-1","kmod-hid":"4.9.77-1","kmod-hid-generic":"4.9.77-1","kmod-input-core":"4.9.77-1","kmod-input-evdev":"4.9.77-1","kmod-ip6tables":"4.9.77-1","kmod-ipt-conntrack":"4.9.77-1","kmod-ipt-core":"4.9.77-1","kmod-ipt-nat":"4.9.77-1","kmod-lib-crc-ccitt":"4.9.77-1","kmod-mmc":"4.9.77-1","kmod-nf-conntrack":"4.9.77-1","kmod-nf-conntrack6":"4.9.77-1","kmod-nf-ipt":"4.9.77-1","kmod-nf-ipt6":"4.9.77-1","kmod-nf-nat":"4.9.77-1","kmod-nls-base":"4.9.77-1","kmod-nls-cp437":"4.9.77-1","kmod-nls-iso8859-1":"4.9.77-1","kmod-nls-utf8":"4.9.77-1","kmod-ppp":"4.9.77-1","kmod-pppoe":"4.9.77-1","kmod-pppox":"4.9.77-1","kmod-rt2800-usb":"4.9.77+2017-11-01-3","kmod-slhc":"4.9.77-1","kmod-sound-arm-bcm2835":"4.9.77-1","kmod-sound-core":"4.9.77-1","kmod-usb-core":"4.9.77-1","kmod-usb-hid":"4.9.77-1","lede-keyring":"2017-01-20-a50b [... cut ...] 017-09-05-1f93bd4c-5","odhcpd-ipv6only":"1.3-1","opkg":"2017-12-07-3b417b9f-2","ppp":"2.4.7-12","ppp-mod-pppoe":"2.4.7-12","procd":"2018-01-23-653629f1-2","ubox":"2017-11-13-e7a63fba-1","ubus":"2018-01-16-5bae22eb-1","ubusd":"2018-01-16-5bae22eb-1","uci":"2018-01-01-5beb95da-1","uclient-fetch":"2017-11-02-4b87d831-1","usign":"2015-07-04-ef641914-1","wireless-regdb":"2017-10-20-4343d359","wpad-mini":"2017-08-24-c2d4f2eb-6"},"distro":"OpenWrt","target":"brcm2708","subtarget":"bcm2710"}`

Fails. :|

aparcar, 2018-02-18:
The correct boar_rename is missing. Problem is that the profile to build is different from the board_name the RPI receives via ubus.
https://github.com/aparcar/attendedsysupgrade-server/blob/master/distributions/openwrt/snapshot.yml

aparcar, 2018-02-19:
@tlaurion I added the rename entry manually, can you please try again & close the issue if it works?

tlaurion, 2018-02-21:
Firmware generation works but resulting upgrade tries to mount and boot
from flash and fails instead of attempting to mount the sdcard and boot
from it.

aparcar, 2018-02-21:
did you try a manual sysupgrade before? does it behave the same way? The sysupgrade-app runs the normal sysupgrade command - maybe it's not an issue of the image but on the sysupgrade scripts for RPI? @chunkeey ?

chunkeey, 2018-02-21:
@aparcar
Yes, there was a recently a problem with all the brcm2708's sysupgrade images. Stintel fixed it two days ago: https://github.com/lede-project/source/commit/858ff4f3e1877a4baf882788cfd99deab26fbb5e

I tried it with my rpi-b/raspberrypi,model-b but I didn't had much luck (I don't have a model 3 to test at the moment). So, I faked it (by editing /etc/os_release openwrt_release and the sysinfo' board ) to return be raspberrypi,model-3 brcm2710. This seemed to work at least I got to the stage where the image was built: https://ledeupdate.planetexpress.cc/static/faillogs/request-89d37bc6e997.log

It looks I would need to get hold of an older openwrt release snapshot first. 

by the way, this is what a squashfs-image could look like:
https://github.com/chunkeey/apm82181-lede/commit/469b2d6982b0b06fdca860cdd544a81faf1519a7
I made something similar for the apm821xx a while ago but had no use for it, until now. Sadly I won't be able to post it until next month.

aparcar, 2018-02-21:
> I tried it with my rpi-b/raspberrypi,model-b but I didn't had much luck (I don't have a model 3 to test at the moment). So, I faked it (by editing /etc/os_release openwrt_release and the sysinfo' [...]

You can give me the `board_name` info of whatever RPI you got and I can add it to the `board_rename` table.

> by the way, this is what a squashfs-image could look like:
chunkeey/apm82181-lede@469b2d6

Please keep me updated when you create a PR of this

chunkeey, 2018-02-21:
> You can give me the board_name info of whatever RPI you got and I can add it to the board_rename table.

that would be:
raspberrypi,model-b should map to rpi-b

Is it possible to map one board_name to two different new board_name?
I'm asking because of the rpi-3. Since it has two images (one AARCH64/ARMv8 brcm2710)
and it also shares the 32-bit ARMv7 brcm2709 images with the rpi-2.

> > by the way, this is what a squashfs-image could look like:
> > chunkeey/apm82181-lede@469b2d6
> >
> Please keep me updated when you create a PR of this
K, No problem.

commit d477a87d (2018-02-10): hopefully fix #88
closed 2018-05-22 by aparcar