openwrt/asu #89: new sub-target ar71xx/tiny

dangowrt, 2018-02-23:
All the 4MB-flash devices have been moved from ar71xx/generic to ar71xx/tiny on OpenWrt master and the upcoming 18.03/4 release.
- [ ] support ar71xx/tiny on server side
- [ ] add a replacement table to move devices currently running ar71xx/generic on 17.01.x to ar71xx/tiny when 18.03/4 is released

For details such as the devices affected, see https://github.com/openwrt/openwrt/commit/0cd5e85e7ad621223b0787e66d8ad20fb2694135

aparcar, 2018-03-02:
I guess the following devices?

```
a02-rb-w300n
dir-600-a1
dir-601-a1
dir-601-b1
dir-615-c1
dir-615-i1
dir-615-i3
fr-54rtr
n150r
rnx-n360rt
tew-632brp
tew-652brp
tew-712br
tl-mr10u-v1
tl-mr11u-v1
tl-mr11u-v2
tl-mr12u-v1
tl-mr13u-v1
tl-mr3020-v1
tl-mr3040-v1
tl-mr3040-v2
tl-mr3220-v1
tl-mr3220-v2
tl-mr3420-v1
tl-mr3420-v2
tl-wa701nd-v1
tl-wa701nd-v2
tl-wa7210n-v2
tl-wa730re-v1
tl-wa750re-v1
tl-wa7510n-v1
tl-wa801nd-v1
tl-wa801nd-v2
tl-wa801nd-v3
tl-wa830re-v1
tl-wa830re-v2
tl-wa850re-v1
tl-wa860re-v1
tl-wa901nd-v1
tl-wa901nd-v2
tl-wa901nd-v3
tl-wa901nd-v4
tl-wa901nd-v5
tl-wdr3320-v2
tl-wr1041n-v2
tl-wr703n-v1
tl-wr710n-v2
tl-wr720n-v3
tl-wr720n-v4
tl-wr740n-v1
tl-wr740n-v3
tl-wr740n-v4
tl-wr740n-v5
tl-wr740n-v6
tl-wr741nd-v1
tl-wr741nd-v2
tl-wr741nd-v4
tl-wr741nd-v5
tl-wr743nd-v1
tl-wr743nd-v2
tl-wr802n-v1
tl-wr802n-v2
tl-wr840n-v2
tl-wr840n-v3
tl-wr841-v10
tl-wr841-v11
tl-wr841-v12
tl-wr841-v1.5
tl-wr841-v3
tl-wr841-v5
tl-wr841-v7
tl-wr841-v8
tl-wr841-v9
tl-wr843nd-v1
tl-wr847n-v8
tl-wr940n-v4
tl-wr941nd-v2
tl-wr941nd-v3
tl-wr941nd-v4
tl-wr941nd-v5
tl-wr941nd-v6
tl-wr941nd-v6-cn
uImage-lzma.bin
vmlinux.bin
vmlinux.elf
vmlinux.lzma
vmlinux-lzma.elf
whr-g301n
whr-hp-g300n
whr-hp-gn
wlae-ag300n
wnr1000v2
wnr1000v2-vc
wnr2000
wnr2000v3
wnr2000v4
wnr612v2
wp543
wpe72
wpn824n
```

dangowrt, 2018-03-21:
looks right to me

aparcar, 2018-08-07:
the tiny subtarget seems to be fully supported by the server, I was able to build various images

closed 2018-08-07 by aparcar