openwrt/asu #91: board_name and device-tree compatible don't match

dangowrt, 2018-03-21:
Some recent changes in OpenWrt introduced a new generic board detection method. Apparently this ruined the correlation between the ImageBuilder PROFILE name, image filename and board_name as seen on the target.
```
OpenWrt target sunxi/cortexa7 (friendlyarm,nanopi-neo) not found. Please report this at https://github.com/aparcar/attendedsysupgrade-server/issues
```
Now, it is arguable who's mistake that is. Imho the rule for device-tree files in the kernel is that the most significant `compatible` string should somehow correlate with the DTS filename (and that is identical to the ImageBuilder profile name).
Maybe @mkresin can spread some insight on the intended way to solve that dilemma...

mkresin, 2018-03-21:
> Some recent changes in OpenWrt introduced a new generic board detection method.

Nope, the generic - device tree compatible string based - board detection is around since ages. All I've done (and still do) is switching targets using a target specific board detection to the generic one (if the devicetree is used).

> Apparently this ruined the correlation between the ImageBuilder PROFILE name, image filename and board_name as seen on the target.

IMHO it isn't more broken than before.

In case I switched to the generic board detection, I adjusted the profile names (which will be used as image filenames) to allow a correlation. For example the lantiq BT HomeHub 5a uses `bt,homehub-v5a` as userspace boardname. The image and profile name is `bt_homehub-v5a`, as a comma is problematic in Makefiles and might be a protected char on some filesystems.

The whole boardname, profile and imagename matching was discussed on the lede-dev mailinglist with @aparcar a while ago. The outcome was that this way an automatic matching can be done in most cases.

> OpenWrt target sunxi/cortexa7 (friendlyarm,nanopi-neo) not found. Please report this at https://github.com/aparcar/attendedsysupgrade-server/issues

To my knowledge the sunix target uses the board detection since the beginning. The only thing I changed was a switch from the target specific `sunxi_board_name()` to the generic `board_name()` function. But this change was a noop, as both read the content of `/tmp/sysinfo/board_name`.

What makes an automatching hard/impossible here, is the missing relation between profile/image name and the userspace boardname. Without knowing the inner workings of the attendedsysupgrade-server, I suspect that @aparcar needs to add a manual mapping.

From my point of view, the best solution would be to use a profile/image name that matches the userspace boardname to at least make a automatching possible:

```
define Device/friendlyarm_nanopi-neo
  DEVICE_TITLE := FriendlyARM NanoPi NEO
  SUPPORTED_DEVICES := friendlyarm,nanopi-neo
  SUNXI_DTS := sun8i-h3-nanopi-neo
endef
TARGET_DEVICES += friendlyarm_nanopi-neo
```

The SUPPORTED_DEVICES can be dropped by adding something like the following to the sunix image build code:

```
define Device/Default
  SUPPORTED_DEVICES := $(subst _,$(comma),$(1))
endef
```

> Imho the rule for device-tree files in the kernel is that the most significant compatible string should somehow correlate with the DTS filename (and that is identical to the ImageBuilder profile name).

So far, I wasn't able to find anything about dts filenames and have seen all kind of filenames in the kernel. Another reason why I haven't touch the dts filenames yet, is that I don't see much gain in doing so.

In my opinion it is rather uncomfortable if dts filenames and profiles names (more image filenames) are matched. It is kind of hard to get that `sun8i-h3-nanopi-neo-ext4-sdcard.img.gz` is intended for a FriendlyARM NanoPi NEO in contrast to a filename `friendlyarm_nanopi-neo-ext4-sdcard.img.gz`.

Let me if I missed the point of your question.

dangowrt, 2018-11-14:
@aparcar still not able to use attendedsysupgade on any other platform than legacy (ie. non-DTS) ar71xx and x86. Please implement the `,` to `_` replacement rule in a generic way, so we won't have to create rules for each and every device.

```
auc (0.0.9)
running OpenWrt 18.06.0-rc2 on lantiq/xrx200 (tplink,tdw8970)
checking https://ledeupdate.planetexpress.cc for release upgrade or updated packages
OpenWrt target lantiq/xrx200 (tplink,tdw8970) not found. Please report this at https://github.com/aparcar/attendedsysupgrade-server/issues
```

aparcar, 2018-11-14:
@dangowrt please try the new upgrade server at http://as-test.stephen304.com/ instead, the planetexpress one while be shutdown for good in a while.

ideally https://github.com/aparcar/attendedsysupgrade-server/commit/5ce835fafaa32eaa61c5976611b00ab513837476 solves the problem, I have no dev setup to test this right now, but will as soon as possible

aparcar, 2019-04-06:
This is solved within the ucert branch as it uses the `supported_devices` provided by the ImageBuilder.

aparcar, 2020-02-25:
The new server only uses OpenWrt profiles which is slowly migrating to device tree only, I'd say this issue is fixed within this project.

closed 2020-03-05 by aparcar