openwrt/asu #92: auc: remove https dependence

aparcar, 2018-04-02:
checking the images manually or via usign should be possible

dangowrt, 2018-04-02:
why? I deliberately made HTTPS mandatory as 'checking images' in which way ever is currently not possible as the worker's pubkey is not present on the target system.

aparcar, 2018-04-02:
I could download the image and then check the sha256 which is also available at the server viewable on via https

aparcar, 2018-04-02:
Use case, I install auc on my 841, it checks for an upgrade and requests the image. The image won't be downloaded but a download URL presented to the user. It's then possible to download the image via browser (https) and use scp to upload the image to the router. 

It's may a strange use case but currently I'd like to create a sysupgrade for my router but using all packages from /usr/lib/opkg/status result on chef.libremesh.org in an imagesize fail...

dangowrt, 2018-04-02:
@aparcar: using OpenWrt snapshot...? (you really shouldnt expect that to work/fit on ancient 4m/32m devices)

aparcar, 2018-04-02:
why wouldn't sysupgrade/auc work on a legacy device? I thoughts thats why they created ar71xx/tiny?

dangowrt, 2018-04-02:
yes, but on a release (ie. 18.05 or whatever) everything gets stripped, debugging features get switched off etc. then you got those 150kb extra space for mbedtls. On development snapshots you don't.

aparcar, 2018-08-03:
@dangowrt please update auc that it uses either https or relies on usign certificates, both appear to be a valid verification method. The latter should work out of the box on most devices

dangowrt, 2018-08-14:
ucert verification is supported by `sysupgrade` itself, so all signature verification can be dropped from online-update clients.

aparcar, 2018-08-14:
@dangowrt Perfect, please make HTTPS encryption optionally :)

dangowrt, 2018-08-14:
Well, just because the image is valid and signed, it doesn't mean yet that it's the one I asked for. Bad folks may still use replay attacks to give you a valid and signed image with a different combination of packages, eg. missing NTP support (making TLS MitM much easier) or missing ca-certifficates (which leads to an _accept-all_ situation). Hence I'd still leave HTTPS enabled by default.

aparcar, 2021-03-14:
OpenWrt ships now with HTTPS as the libs are required for WPA3 anyway.

closed 2021-03-14 by aparcar