openwrt/asu #94: how to proceed

aparcar, 2018-04-17:
Given the fact that this project is nearly a year old and still I'm (basically) the only one contributing, I clearly produced a codebase nobody wants to mess with nor did I deliver a documentation about anything but the API which could help new devs to start. 

The code mess is mostly as I rapidly created a (my first) web app in python with changing requirements during the last GSoC, various features weren't fully planned for the initial version but then somewhat "unclean" implemented (multible worker, database optimizations, image signing). Documentation alone doesn't make that really more appealing.

Still, within the last year I'd estimate about 2000, mostly individual, images where created and hopefully run somewhere. Also the project is, at least for my personal developing, usable as I can quickly update snapshots without package re-installation. This motivates me to continue the project, but due to my lack of experience I'm asking how should I proceed with the project? 

* The multi worker feature could be removed, worst case, people wait a few seconds for their build. Snapshots should always have lower priority so stable firmware images are quickly created and due to caching mostly instantly ready
* Signing on a single server wouldn't be as complicated as creating the current idea of a trust chain.
* Moving from ODBC to PostgreSQL should remove a big stumbling stone of new devs as ODBC is somewhat unhandy to setup - at least from my experience various people complained about that.
* The server should become a python package which is easily installed to play with locally
* The server should also become a OpenWrt package, making it easy to deploy various instances.
* more?

I'd be happiest for ideas, please respond

aparcar, 2018-04-17:
@dangowrt @jow- @egon0 @FreifunkUFO @nicopace @ilario

mtippmann, 2018-04-17:
I attempt to provide pull requests for: 
- Remove odbc and use psycog (and maybe sqlalchemy core) instead. 
- Notes for setting up without ansible and a `requirements.txt` for virtualenv

dangowrt, 2018-04-17:
The OpenWrt package for the server side is already usable, I guess it can be merged into openwrt/packages at any time you feel it's ready for that.

If you migrate away from ODBC to any python-specific access layer, please **benchmark** before and after that change. I see this a something which people want to change for rather irrational reasons -- database connections always have to be setup (ie. credentials, server address, etc.) at installation, the OpenWrt package automatizes this process and ODBC is not an obstacle there.
However, if there is no performance impact and things don't become very ugly, I don't mind which fashion of the day people want to follow.

Making the server-side a `pip`-installable package would be great and also make the OpenWrt packaging much simplier -- I'm not into Python at all, but virtually all other python packages use that approach rather than just copying files to weird locations in the filesystem or using ansible for that.

mtippmann, 2018-04-17:
@dangowrt the reason for getting rid of pyodbc was for me that it's not plug'n'play installable via pip - at least on Arch you have to setup some .ini files in /etc and look for library names that are not really documented. So there is probably no technical reason against odbc but for it was quite hassle to get it working - besides that the sql is specific to postgres anyway because of PL/pgSQL so psycog should be fine. 

What would be a reasonable benchmark that stresses the db?

dangowrt, 2018-04-17:
@mtippmann yeah, ODBC is not well integrated in archlinux.
Those ini files can quite easily be auto-generated, see the [OpenWrt ODBC packages](https://github.com/openwrt/packages/blob/master/libs/unixodbc/files/odbc.init)  -- the archlinux-way of dealing with ODBC seems to be more like to only fulfill build-requirements but nobody actually uses it...
The advantage is that ODBC is more than just a wrapper for db-client-libraries and mapping of SQL to C-type datatypes, but also provides a transaction-layer as well as automatically making use of prepared statements if supported by the database backend.
Importing package lists and making some simple queries within the attended-sysupgrade database should be ok as a benchmark -- I mostly worry about large bulk-transactions which happend to work nicely in ODBC because type-conversion and such is done quite efficiently in C code.
Regarding portability, I still believe it could make sense to preserve the modularity needed to switch to other DB layeres -- I personally believe that most of the PostgreSQL-specific PL/pgSQL stuff can quite easily be replaced by 'pure' SQL functions which would make them compatible with mariaDB as well as commercial RDBMs.
Also note that ODBC provides language independence of the DB connection, ie. the same system-wide ODBC connection can be used from different programming environments without having to think about connection encoding (byte-order, unicode and stuff), datatypes, ... all over again. We may at some point replace parts of the sysupgrade server (such as the workers or serving cached images more efficiently) with code written in languages other than python which was the main reason to choose ODBC as an access layer in the beginning.

aparcar, 2018-04-17:
@dangowrt @mtippmann I'm against using sqlalchemy but psycogs2 seems like a good solution. It supports the preparing of statements which where impossible to do manually with ODBC (you remeber @dangowrt ?). Also I started to use so much PostgreSQL json magic that I don't feel like moving to a different DB anytime soon.

aparcar, 2018-04-17:
About the other parts, do we require multi worker or wouldn't it be a better approach to simplify the server code as much as possible and let communities (Freifunk, LibreMesh, whoever) use their own servers? No central point of failure, no hassle which specially patched ImageBuilders...? The average build time was ~11 seconds per image, usually 3 seconds for ar71xx, 300 seconds for x86. Maybe we can optimize that rather then spinning up more workers. Avoiding all the trust chain thing

dangowrt, 2018-04-17:
well, relying in an automatized way to handle PREPARE is kinda better than doing it manually, ain't it?
mariaDB also introduced supports all sorts of [JSON magic](https://mariadb.com/kb/en/library/differences-between-json_query-and-json_value/) recently, so porting the DB schema once we consider it stable is an option...

nicopace, 2018-04-26:
We were planning to test it on our networks, but haven't had time for it yet :(
More pressing matters have been around (like bmx6 issues).
Once we move pass them we will get to this, that is really relevant!

aparcar, 2018-08-03:
I'll just continue with this project as I'm learning quite a lot. More people seem to be interested in the server - that's cool, too.

closed 2018-08-03 by aparcar