openwrt/asu #95: can't remove firewall package

brianjmurrell, 2018-04-24:
When I try to use the _ImageBuilder_ at https://chef.libremesh.org/ with LEDE and select _Advanced_ to alter package selections, my resulting image has the _firewall_ package installed even though I removed it from the package list.

Trying to utilise _ImageBuilder_'s *-firewall* (remove package) nomenclature in the package list just returns an error.

I know you can remove _firewall_ from LEDE using _ImageBuilder_ since I do it for all of my internal AP builds here for my local network.

Why can't I remove it at https://chef.libremesh.org/ just as easily?

jow-, 2018-04-24:
Maybe you have other meta pacakges selected, like `luci`, `luci-ssl` or `luci-app-firewall` ?

aparcar, 2018-04-24:
> Trying to utilise ImageBuilder's -firewall (remove package) nomenclature in the package list just returns an error.

This is implemented in a different way, whatever package is not in the textarea will be automatically removed from the firmware. This is done via a diff with the "default packages". It was meant to allow package removal without knowing the "default packages" per device. However that might a bit confusing for people used to the pure ImageBuilder.

brianjmurrell, 2018-04-24:
Having to remove _luci_ just to remove _firewall_ is a bit of [throwing the baby out with the bathwater](https://en.wikipedia.org/wiki/Don%27t_throw_the_baby_out_with_the_bathwater).  It should be enough to just not include _luci-app-firewall_ to be able to have _luci_ and not have _firewall_.

But I don't have _luci-app-firewall_ in my list of packages for Chef either.

With ImageBuilder I am able to put `-firewall -luci-app-firewall` in my packages list to remove both of those and still retain _luci_.

brianjmurrell, 2018-05-01:
Any resolution to this, given that the ImageBuilder itself is capable of handling this use-case?

aparcar, 2018-07-20:
@dangowrt can you help out here. If I understood you correctly it's not possible to install the `luci` but remove other packages with are automatically included in it. @brianjmurrell seems to experience a different behavior...

In any case, I'm working on a better version of the server which then allow *negative* packages, if that is then correctly handled by the ImageBuilder is another thing...

aparcar, 2018-08-07:
@brianjmurrell I did some testing but I could not reproduce that behavior with an ImageBuilder. I guessed that the negative packages have be be at the end (at least behind luci), but that does not help neither.

    make image PROFILE="tl-wdr3600-v1" PACKAGES="luci -luci-firewall -firewall"

I close the issue as it's not a fault of the server (if it's a fault after all). If you can provide a working PAKCAGES var I'm happy to implement that with the server

brianjmurrell, 2018-09-11:
This works here:

```
make -C ../openwrt-imagebuilder-18.06.1-ar71xx-tiny.Linux-x86_64/ image PROFILE="DIR615C1" PACKAGES="-dnsmasq -firewall -luci-app-firewall ssmtp -kmod-ppp -kmod-pppoe -kmod-pppox -kmod-slhc -ip6tables -iptables -odhcp6c -odhcpd -ppp -ppp-mod-pppoe"
```

None of those packages are added to the image:

```
Building images for ar71xx - D-Link DIR-615 rev. C1
Packages: ssmtp base-files busybox dropbear fstools kernel kmod-ath9k kmod-gpio-button-hotplug libc libgcc logd mtd netifd odhcpd-ipv6only opkg swconfig uboot-envtools uci uclient-fetch wpad-mini

Installing packages...
Installing libc (1.1.19-1) to root...
Installing libgcc (7.3.0-1) to root...
Copying /net/server/src/openwrt-imagebuilder-18.06.1-ar71xx-tiny.Linux-x86_64/dl/libgcc_7.3.0-1_mips_24kc.ipk.
Configuring libgcc.
Configuring libc.
Installing kernel (4.9.120-1-c4141763fd59104e24eeb6bb2f68b426) to root...
Configuring kernel.
Installing ssmtp (2.64-5) to root...
Copying /net/server/src/openwrt-imagebuilder-18.06.1-ar71xx-tiny.Linux-x86_64/dl/ssmtp_2.64-5_mips_24kc.ipk.
Installing zlib (1.2.11-2) to root...
Copying /net/server/src/openwrt-imagebuilder-18.06.1-ar71xx-tiny.Linux-x86_64/dl/zlib_1.2.11-2_mips_24kc.ipk.
Installing libopenssl (1.0.2p-1) to root...
Copying /net/server/src/openwrt-imagebuilder-18.06.1-ar71xx-tiny.Linux-x86_64/dl/libopenssl_1.0.2p-1_mips_24kc.ipk.
Installing base-files (192-r7258-5eb055306f) to root...
Copying /net/server/src/openwrt-imagebuilder-18.06.1-ar71xx-tiny.Linux-x86_64/dl/base-files_192-r7258-5eb055306f_mips_24kc.ipk.
Installing libubox (2018-07-25-c83a84af-2) to root...
Copying /net/server/src/openwrt-imagebuilder-18.06.1-ar71xx-tiny.Linux-x86_64/dl/libubox_2018-07-25-c83a84af-2_mips_24kc.ipk.
Installing libuci (2018-08-11-4c8b4d6e-1) to root...
Copying /net/server/src/openwrt-imagebuilder-18.06.1-ar71xx-tiny.Linux-x86_64/dl/libuci_2018-08-11-4c8b4d6e-1_mips_24kc.ipk.
Installing libnl-tiny (0.1-5) to root...
Copying /net/server/src/openwrt-imagebuilder-18.06.1-ar71xx-tiny.Linux-x86_64/dl/libnl-tiny_0.1-5_mips_24kc.ipk.
Installing libubus (2018-07-26-40e0931e-1) to root...
Copying /net/server/src/openwrt-imagebuilder-18.06.1-ar71xx-tiny.Linux-x86_64/dl/libubus_2018-07-26-40e0931e-1_mips_24kc.ipk.
Installing libjson-c (0.12.1-2) to root...
Copying /net/server/src/openwrt-imagebuilder-18.06.1-ar71xx-tiny.Linux-x86_64/dl/libjson-c_0.12.1-2_mips_24kc.ipk.
Installing libblobmsg-json (2018-07-25-c83a84af-2) to root...
Copying /net/server/src/openwrt-imagebuilder-18.06.1-ar71xx-tiny.Linux-x86_64/dl/libblobmsg-json_2018-07-25-c83a84af-2_mips_24kc.ipk.
Installing ubusd (2018-07-26-40e0931e-1) to root...
Copying /net/server/src/openwrt-imagebuilder-18.06.1-ar71xx-tiny.Linux-x86_64/dl/ubusd_2018-07-26-40e0931e-1_mips_24kc.ipk.
Installing ubus (2018-07-26-40e0931e-1) to root...
Copying /net/server/src/openwrt-imagebuilder-18.06.1-ar71xx-tiny.Linux-x86_64/dl/ubus_2018-07-26-40e0931e-1_mips_24kc.ipk.
Installing jshn (2018-07-25-c83a84af-2) to root...
Copying /net/server/src/openwrt-imagebuilder-18.06.1-ar71xx-tiny.Linux-x86_64/dl/jshn_2018-07-25-c83a84af-2_mips_24kc.ipk.
Installing netifd (2018-07-30-a0a1e52e-1) to root...
Copying /net/server/src/openwrt-imagebuilder-18.06.1-ar71xx-tiny.Linux-x86_64/dl/netifd_2018-07-30-a0a1e52e-1_mips_24kc.ipk.
Installing libjson-script (2018-07-25-c83a84af-2) to root...
Copying /net/server/src/openwrt-imagebuilder-18.06.1-ar71xx-tiny.Linux-x86_64/dl/libjson-script_2018-07-25-c83a84af-2_mips_24kc.ipk.
Installing ubox (2018-02-14-128bc35f-2) to root...
Copying /net/server/src/openwrt-imagebuilder-18.06.1-ar71xx-tiny.Linux-x86_64/dl/ubox_2018-02-14-128bc35f-2_mips_24kc.ipk.
Installing procd (2018-03-28-dfb68f85-1) to root...
Copying /net/server/src/openwrt-imagebuilder-18.06.1-ar71xx-tiny.Linux-x86_64/dl/procd_2018-03-28-dfb68f85-1_mips_24kc.ipk.
Installing jsonfilter (2018-02-04-c7e938d6-1) to root...
Copying /net/server/src/openwrt-imagebuilder-18.06.1-ar71xx-tiny.Linux-x86_64/dl/jsonfilter_2018-02-04-c7e938d6-1_mips_24kc.ipk.
Installing usign (2015-07-04-ef641914-1) to root...
Copying /net/server/src/openwrt-imagebuilder-18.06.1-ar71xx-tiny.Linux-x86_64/dl/usign_2015-07-04-ef641914-1_mips_24kc.ipk.
Installing openwrt-keyring (2018-05-18-103a32e9-1) to root...
Copying /net/server/src/openwrt-imagebuilder-18.06.1-ar71xx-tiny.Linux-x86_64/dl/openwrt-keyring_2018-05-18-103a32e9-1_mips_24kc.ipk.
Installing fstools (2018-04-16-e2436836-1) to root...
Copying /net/server/src/openwrt-imagebuilder-18.06.1-ar71xx-tiny.Linux-x86_64/dl/fstools_2018-04-16-e2436836-1_mips_24kc.ipk.
Installing fwtool (1) to root...
Copying /net/server/src/openwrt-imagebuilder-18.06.1-ar71xx-tiny.Linux-x86_64/dl/fwtool_1_mips_24kc.ipk.
Installing busybox (1.28.3-4) to root...
Copying /net/server/src/openwrt-imagebuilder-18.06.1-ar71xx-tiny.Linux-x86_64/dl/busybox_1.28.3-4_mips_24kc.ipk.
Installing dropbear (2017.75-5.1) to root...
Copying /net/server/src/openwrt-imagebuilder-18.06.1-ar71xx-tiny.Linux-x86_64/dl/dropbear_2017.75-5.1_mips_24kc.ipk.
Package fstools (2018-04-16-e2436836-1) installed in root is up to date.
Package kernel (4.9.120-1-c4141763fd59104e24eeb6bb2f68b426) installed in root is up to date.
[... 66 lines cut ...]
Configuring procd.
Configuring jsonfilter.
Configuring usign.
Configuring openwrt-keyring.
Configuring fstools.
Configuring fwtool.
Configuring base-files.
Configuring uboot-envtools.
Configuring wireless-regdb.
Configuring kmod-cfg80211.
Configuring hostapd-common.
Configuring kmod-mac80211.
Configuring kmod-ath.
Configuring kmod-ath9k-common.
Configuring kmod-ath9k.
Configuring uci.
Configuring wpad-mini.
Configuring dropbear.
Configuring mtd.
Configuring odhcpd-ipv6only.
Configuring kmod-gpio-button-hotplug.
Configuring logd.

Finalizing root filesystem...
```

brianjmurrell, 2018-09-11:
Looking at a [build log](https://as-test.stephen304.com/download/openwrt/18.06.1/ar71xx/generic/archer-c7-v2/fb52e20f73d1c28//buildlog-3bff9361af3b7c4.txt) it looks like you are putting removed packages first in the list of packages and then alpha-sorting the remaining list of packages.

Maybe try putting the removed packages at the end of the list instead?

aparcar, 2018-09-12:
@brianjmurrell thanks for further testing, I though the issue would be solved via https://github.com/aparcar/attendedsysupgrade-server/commit/c658031481d85c613c9aaef421ab7b9eba830bbb but it seems most ar71xx/tiny images fail, even tho plenty of packages are removed. You have an idea why it works on your local machine but not on the server?

brianjmurrell, 2018-09-13:
@aparcar Do you have a build run with that patch applied that I can look at the log for?

aparcar, 2018-09-13:
@brianjmurrell see [here](https://as-test.stephen304.com/download/openwrt/18.06.1/ar71xx/generic/tl-wdr4300-v1/c1c0945beb8b88c//buildlog-efc61a4ed777451.txt)
The following line looks as expected, negative packages at the end:

    PACKAGES=wpad-mini uclient-fetch uci uboot-envtools swconfig ppp-mod-pppoe opkg odhcpd-ipv6only odhcp6c netifd mtd logd libgcc kmod-usb2 kmod-usb-ledtrig-usbport kmod-usb-core kmod-gpio-button-hotplug kmod-ath9k ip6tables fstools dropbear busybox base-files -luci -iptables -firewall -dnsmasq

brianjmurrell, 2018-09-13:
Indeed.  And if you look at the build output, you didn't get the _firewall_ package installed -- or any of the others that were excluded.  So that seems to be working.

Except, that when I [try it](https://as-test.stephen304.com/download/openwrt/18.06.1/ar71xx/generic/archer-c7-v2/3c38e2420d88165//buildlog-a67d63f92a97557.txt) even though the list of packages to be excluded is at the end:

```
PACKAGES=wpad-mini uclient-fetch uci uboot-envtools swconfig ppp-mod-pppoe ppp opkg odhcpd-ipv6only odhcp6c netifd mtd luci logd libgcc kmod-usb2 kmod-usb-ledtrig-usbport kmod-usb-core kmod-gpio-button-hotplug kmod-ath9k kmod-ath10k iptables ip6tables fstools firewall dropbear dnsmasq busybox base-files ath10k-firmware-qca988x -luci-app-firewall -firewall -dnsmasq
```

they are still included:

```
Installing firewall (2018-07-26-aa8846bb-1) to root...
Copying /mnt/worker/worker1/imagebuilder/openwrt/18.06.1/ar71xx/generic/dl/firewall_2018-07-26-aa8846bb-1_mips_24kc.ipk.
Installing luci-app-firewall (git-18.247.71242-9541751-1) to root...
Copying /mnt/worker/worker1/imagebuilder/openwrt/18.06.1/ar71xx/generic/dl/luci-app-firewall_git-18.247.71242-9541751-1_all.ipk.
```

although interestingly enough _dnsmasq_ is not installed.

Ahhh.  I see.  In my local builds, I `-luci` but install `luci-base` instead.  So yeah, when I do that with https://as-test.stephen304.com/chef it seems to be working.

So this all seems to be working now.  Can probably close this as resolved with https://github.com/aparcar/attendedsysupgrade-server/commit/c658031481d85c613c9aaef421ab7b9eba830bbb.

aparcar, 2018-09-13:
Thanks for your interest and testing!

closed 2018-09-13 by aparcar