openwrt/asu #100: Improper database schema?

oneirosFade, 2018-06-11:
Execution of Vagrant up fails on 'cli.py -ia' due to:

```bash
INFO:utils.database:log initialized
INFO:utils.database:config initialized
INFO:utils.database:database connected
Traceback (most recent call last):
  File "cli.py", line 198, in <module>
    sc = ServerCli()
  File "cli.py", line 22, in __init__
    self.init_args()
  File "cli.py", line 36, in init_args
    self.init_all_imagebuilders()
  File "cli.py", line 49, in init_all_imagebuilders
    for distro, release in self.database.get_releases():
  File "/srv/utils/database.py", line 69, in get_releases
    return self.c.execute("select distro, release from releases").fetchall()
pyodbc.ProgrammingError: ('42P01', '[42P01] ERROR: relation "releases" does not exist;\nError while executing the query (7) (SQLExecDirectW)')
```

dangowrt, 2018-06-11:
Looks like create_tables() is never called from anywhere -- imho this is also not supposed to be done by the application itself but rather by the distro or the administrator, so run-time priviledges can be limited properly.
So you should create the database (and role/user-account) for the server, set this in config.yml and instantiate the db definitions from tables.sql. This can be done using `isql` (and in that way you already test that ODBC is working properly) or directly using the `psql` client for PostgreSQL.
When packaged properly, the distribution should do all that for you, see 
https://github.com/aparcar/attendedsysupgrade-server/blob/master/package/sysupgrade-image-server/files/upimaged.defaults
to get an idea of how it works when running the server itself on OpenWrt (x86_64).

aparcar, 2018-08-02:
@oneirosFade please close this issue if resolved. You could try the new Vagrant file to test the setup