openwrt/asu #122: Integration LiMe with Mikrotik

IBEBrasil, 2018-08-26:
Hi people....i am testing to integrate mesh networks using LiMe and Hotspot+Radius in RouterOS (Mikrotik).  The basic problem is that the first user in the mesh that make a login in Hotspot liberate the access for the others without login.  This is caused by the fact that Hotspot in MK register the MAC and in the mesh everyone that gets out to internet by the wan in the gateway have the MAC of this port.  Anyone knows how to resolve that in Lime or how to put the autentication to an external radius server ? even that this external radius is the routeros pluged in the wan of the lime router....thanks.  
br
marcelo
PS.: betwwen this tests we are trying to use nodogsplash, pitbull and in this week i will try freeradius in one of our routers with lime

aparcar, 2018-08-27:
Hi @IBEBrasil, thanks for your interested. I'm afraid your Topic is more related to [LibreMesh](https://github.com/libremesh) than this upgrade server...

closed 2018-08-27 by aparcar

dangowrt, 2018-08-28:
@IBEBrasil MikroTik hardware can run OpenWrt (and thus LibreMesh firmware) itself, just don't use proprietary crap like RouterOS. If you feel there really is a good reason to use RouterOS, please let us know, I'm sure there are ways around it ;)

dangowrt, 2018-08-28:
Hi Marcelo,
I hope I understood the problem you are trying to solve, it sounded like you are looking for some kind of bandwidth-fairness, ie. everybody should get a piece of the CAKE... Or is it about **airtime** fairness? AFAIK there is no ready made solution to solve mesh-wide bandwidth allocation, however, combining eSFQ (or CAKE or the like) with modern wifi drivers which do airtime fairness can already achieve acceptable results (and much better than any proprietary solution).
Please elaborate more about which exact functionality of Mikrotik's RouterOS you want to use and miss in OpenWrt and/or LibreMesh.

Cheers

Daniel

IBEBrasil, 2018-08-28:
OK Daniel, thks again and let me try to draw the scenario.
To use local network and local contents, everyone can connect and use it, but when someone try to connect to internet we create an associative proccess that only these users can connect to internet.
Said that, in mesh networks with lime in tplinks archer c5, we connect in a WAN port of one of these tplinks a RB with hotspot installed and then control bandwidth per user. Maybe controll bandwidth by IP through queues will be a sollution, but if u have a tip to use a solution like captive portal or vouchers like nodogsplash i will thanks a lot ... PS.: to run this sollution into a router or a rasp.  PS1.: Maybe  another sollution is how and where i configure a lime to authenticate in an external radius, where i put this parameters in lime ? thks

IBEBrasil, 2018-08-29:
Another thing that i saw is the doc that explain the functionality of lime and in the wan, i only have this problem, how to get the mac of the user connected in the mesh, cause the only mac that i get already is the mac of the wan. This port is configured by default as bridge, but why i can not get the mac of the user ? If i connect a radius o capitive portal in lan port i can get the mac os the user in the mash, but in wan not.  thanks again for the opportunity to this debate. br, marcelo