openwrt/asu #124: How to configure an external authentication through freeradius or pfsense into router with LiMe

IBEBrasil, 2018-08-30:
People, days ago a try to integrate hotspot mikrotik with lime, but i understand that routerOS is not to point in this group, then i want to know how to integrate LiMe to make authentication in a external freeradius server, and the problem are two : 
1 - where to configure in the router with lime the pointing to access a external freeradius.  I saw the parameter option server e option port, but i think that is another kind of configuration about vpn or something like that.
2 - the other question is that when i plug the link of internet in a wan port of the router with lime, the only mac that gets out of this port is the mac of the wan and not the mac of the user.  is it possible to change that situation. I am testing with pfsense with capitive portal and the problem is that the first that make the login liberate the access for the other that are connected in the mesh.  Thanks.

aparcar, 2018-09-02:
@IBEBrasil again, I think here is the wrong place to ask this question. Please request further help at the [LibreMesh user mailing list](https://lists.libremesh.org/mailman/listinfo/lime-users)

closed 2018-09-02 by aparcar

IBEBrasil, 2018-09-02:
Sorry. Ok thanks

Att,Marcelo SaldanhaInstituto Bem Estar Brasil
-------- Mensagem original --------De: Paul Spooren <notifications@github.com> Data: 02/09/18  06:10  (GMT-03:00) Para: aparcar/attendedsysupgrade-server <attendedsysupgrade-server@noreply.github.com> Cc: IBEBrasil <instituto@bemestarbrasil.org.br>, Mention <mention@noreply.github.com> Assunto: Re: [aparcar/attendedsysupgrade-server] How to configure an external authentication through freeradius or pfsense into router with LiMe (#124) 
@IBEBrasil again, I think here is the wrong place to ask this question. Please request further help at the LibreMesh user mailing list

—
[
{
"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/aparcar/attendedsysupgrade-server/issues/124#issuecomment-417915951",
"url": "https://github.com/aparcar/attendedsysupgrade-server/issues/124#issuecomment-417915951",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
},
{
"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
"title": "Re: [aparcar/attendedsysupgrade-server] How to configure an external authentication through freeradius or pfsense into router with LiMe (#124)",
"sections": [
{
"text": "",
"activityTitle": "**Paul Spooren**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@aparcar",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"aparcar/attendedsysupgrade-server\",\n\"issueId\": 124,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/aparcar/attendedsysupgrade-server/issues/124#issuecomment-417915951"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 374228708\n}"
}
],
"themeColor": "26292E"
}
]