openwrt/asu #129: Linksys WRT3200ACM: "Unsupported device, release, target, subtraget or board"

tlaurion, 2018-09-21:
Hi,

Attempting to search for upgrades results in a list of updates available. Requesting the firmware upgrade results in and Unsupported error. Following [past report](https://github.com/aparcar/attendedsysupgrade-server/issues/88) I provide, below, the same information that was requested to troubleshoot the problem.

ubus call system board
```
{
	"kernel": "4.14.70",
	"hostname": "Jedi",
	"system": "ARMv7 Processor rev 1 (v7l)",
	"model": "Linksys WRT3200ACM",
	"board_name": "armada-385-linksys-rango",
	"release": {
		"distribution": "OpenWrt",
		"version": "SNAPSHOT",
		"revision": "r8081-ccab68f",
		"target": "mvebu\/cortexa9",
		"description": "OpenWrt SNAPSHOT r8081-ccab68f"
	}
}
```
cat /proc/cpuinfo
```
processor	: 0
model name	: ARMv7 Processor rev 1 (v7l)
BogoMIPS	: 1866.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpd32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x4
CPU part	: 0xc09
CPU revision	: 1

processor	: 1
model name	: ARMv7 Processor rev 1 (v7l)
BogoMIPS	: 1866.00
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpd32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x4
CPU part	: 0xc09
CPU revision	: 1

Hardware	: Marvell Armada 380/385 (Device Tree)
Revision	: 0000
Serial		: 0000000000000000
```

Post from upgrade request:
```
{
    "version": "SNAPSHOT", 
    "packages": {
        "iwinfo": "2018-07-31-65b8333f-1", 
        "luci-app-adblock": "git-18.263.38473-f0512e5-1", 
        "opkg": "2017-12-07-3b417b9f-2", 
        "ubus": "2018-07-26-40e0931e-1", 
        "busybox": "1.29.3-1", 
        "kmod-btmrvl": "4.14.70-1", 
        "kmod-nf-reject6": "4.14.70-1", 
        "swconfig": "11", 
        "kmod-nf-flow": "4.14.70-1", 
        "kmod-lib-crc-ccitt": "4.14.70-1", 
        "kmod-pppoe": "4.14.70-1", 
        "kmod-pppox": "4.14.70-1", 
        "kmod-ipt-conntrack": "4.14.70-1", 
        "kmod-nf-reject": "4.14.70-1", 
        "base-files": "195-r8081-ccab68f", 
        "kmod-nf-nat": "4.14.70-1", 
        "netifd": "2018-09-18-23941d7e-1", 
        "ddns-scripts": "2.7.8-4", 
        "uboot-envtools": "2018.03-1", 
        "dnsmasq": "2.80test6-2", 
        "procd": "2018-08-06-e29966f0-1", 
        "ubusd": "2018-07-26-40e0931e-1", 
        "luci-app-ddns": "2.4.9-7", 
        "firewall": "2018-08-13-1c4d5bcd-1", 
        "kmod-nf-ipt": "4.14.70-1", 
        "libuci": "2018-08-11-4c8b4d6e-1", 
        "libip4tc": "1.6.2-3", 
        "ubi-utils": "2.0.2-1", 
        "kmod-ip6tables": "4.14.70-1", 
        "odhcp6c": "2018-07-14-67ae6a71-16", 
        "fstools": "2018-05-23-dd02dad3-2", 
        "uclient-fetch": "2018-08-03-ae1c656f-1", 
        "libuclient": "2018-08-03-ae1c656f-1", 
        "uci": "2018-08-11-4c8b4d6e-1", 
        "wpad-mini": "2018-05-21-62566bc2-4", 
        "dropbear": "2017.75-8", 
        "mwlwifi-firmware-88w8964": "2018-08-10-75374802-1", 
        "mtd": "23", 
        "odhcpd-ipv6only": "1.11-2", 
        "libjson-c": "0.12.1-2", 
        "kmod-mwifiex-sdio": "4.14.70+2017-11-01-9", 
        "libip6tc": "1.6.2-3", 
        "ppp": "2.4.7-13", 
        "libubox": "2018-07-25-c83a84af-2", 
        "kmod-gpio-button-hotplug": "4.14.70-2", 
        "logd": "2018-02-14-128bc35f-2", 
        "libjson-script": "2018-07-25-c83a84af-2", 
        "libblobmsg-json": "2018-07-25-c83a84af-2", 
        "iptables": "1.6.2-3", 
        "openwrt-keyring": "2018-05-18-103a32e9-1", 
        "luci-app-attendedsysupgrade": "git-18.263.38473-f0512e5-1", 
        "jshn": "2018-07-25-c83a84af-2", 
        "kmod-mwlwifi": "4.14.70+2018-08-10-75374802-1", 
        "kmod-ipt-core": "4.14.70-1", 
        "kmod-ppp": "4.14.70-1", 
        "libubus": "2018-07-26-40e0931e-1", 
        "kmod-nf-conntrack": "4.14.70-1", 
        "usign": "2015-07-04-ef641914-1", 
        "ip6tables": "1.6.2-3", 
        "libxtables": "1.6.2-3", 
        "kmod-nf-ipt6": "4.14.70-1", 
        "luci": "git-18.263.38473-f0512e5-1", 
        "kmod-nf-conntrack6": "4.14.70-1", 
        "ubox": "2018-02-14-128bc35f-2", 
        "kernel": "4.14.70-1-0b707ab2c75da82ad4a5527a303bf998", 
        "libnl-tiny": "0.1-5", 
        "fwtool": "1", 
        "jsonfilter": "2018-02-04-c7e938d6-1", 
        "kmod-ipt-offload": "4.14.70-1", 
        "kmod-slhc": "4.14.70-1", 
        "ppp-mod-pppoe": "2.4.7-13", 
        "luci-app-openvpn": "git-18.263.38473-f0512e5-1", 
        "kmod-ipt-nat": "4.14.70-1"
    }, 
    "upgrade_packages": "1", 
    "distro": "OpenWrt", 
    "target": "mvebu", 
    "subtarget": "cortexa9"
}
```

Post from request firmware:
```
{
    "version": "snapshot", 
    "board": "armada-385-linksys-rango", 
    "model": "Linksys WRT3200ACM", 
    "packages": {
        "base-files": "193-r7570-2a598bb", 
        "busybox": "1.28.4-3", 
        "ddns-scripts": "2.7.7-8", 
        "dnsmasq": "2.80test3-1", 
        "dropbear": "2017.75-6", 
        "firewall": "2018-07-26-aa8846bb-1", 
        "fstools": "2018-05-23-dd02dad3-1", 
        "fwtool": "1", 
        "ip6tables": "1.6.2-2", 
        "iptables": "1.6.2-2", 
        "iwinfo": "2018-05-18-e59f9253-1", 
        "jshn": "2018-06-25-c83a84af-1", 
        "jsonfilter": "2018-02-04-c7e938d6-1", 
        "kmod-btmrvl": "4.14.27-1", 
        "kmod-gpio-button-hotplug": "4.14.27-2", 
        "kmod-ip6tables": "4.14.27-1", 
        "kmod-ipt-conntrack": "4.14.27-1", 
        "kmod-ipt-core": "4.14.27-1", 
        "kmod-ipt-nat": "4.14.27-1", 
        "kmod-ipt-offload": "4.14.27-1", 
        "kmod-lib-crc-ccitt": "4.14.27-1", 
        "kmod-mwifiex-sdio": "4.14.27+2017-11-01-4", 
        "kmod-mwlwifi": "4.14.27+2018-03-05-b1b9a9e1-1", 
        "kmod-nf-conntrack": "4.14.27-1", 
        "kmod-nf-conntrack6": "4.14.27-1", 
        "kmod-nf-flow": "4.14.27-1", 
        "kmod-nf-ipt": "4.14.27-1", 
        "kmod-nf-ipt6": "4.14.27-1", 
        "kmod-nf-nat": "4.14.27-1", 
        "kmod-nf-reject": "4.14.27-1", 
        "kmod-nf-reject6": "4.14.27-1", 
        "kmod-ppp": "4.14.27-1", 
        "kmod-pppoe": "4.14.27-1", 
        "kmod-pppox": "4.14.27-1", 
        "kmod-slhc": "4.14.27-1", 
        "libblobmsg-json": "2018-06-25-c83a84af-1", 
        "libip4tc": "1.6.2-2", 
        "libip6tc": "1.6.2-2", 
        "libjson-c": "0.12.1-1", 
        "libjson-script": "2018-06-25-c83a84af-1", 
        "libnl-tiny": "0.1-5", 
        "libubox": "2018-06-25-c83a84af-1", 
        "libubus": "2018-07-26-40e0931e-1", 
        "libuci": "2018-03-24-5d2bf09e-1", 
        "libuclient": "2017-11-02-4b87d831-1", 
        "libxtables": "1.6.2-2", 
        "logd": "2018-02-14-128bc35f-2", 
        "luci": "git-18.209.45694-a0b4d2a-1", 
        "luci-app-adblock": "git-18.209.45694-a0b4d2a-1", 
        "luci-app-attendedsysupgrade": "git-18.209.45694-a0b4d2a-1", 
        "luci-app-ddns": "2.4.9-5", 
        "luci-app-openvpn": "git-18.209.45694-a0b4d2a-1", 
        "mtd": "23", 
        "mwlwifi-firmware-88w8964": "2018-06-15-8683de8e-1", 
        "netifd": "2018-07-11-5cf79759-1", 
        "odhcp6c": "2018-07-14-67ae6a71-14", 
        "odhcpd-ipv6only": "1.10-1", 
        "openwrt-keyring": "2018-05-18-103a32e9-1", 
        "opkg": "2017-12-07-3b417b9f-2", 
        "ppp": "2.4.7-13", 
        "ppp-mod-pppoe": "2.4.7-13", 
        "procd": "2018-07-05-a0372ac0-1", 
        "swconfig": "11", 
        "ubi-utils": "2.0.2-1", 
        "uboot-envtools": "2018.03-1", 
        "ubox": "2018-02-14-128bc35f-2", 
        "ubus": "2018-07-26-40e0931e-1", 
        "ubusd": "2018-07-26-40e0931e-1", 
        "uci": "2018-03-24-5d2bf09e-1", 
        "uclient-fetch": "2017-11-02-4b87d831-1", 
        "usign": "2015-07-04-ef641914-1", 
        "wpad-mini": "2018-05-21-62566bc2-4"
    }, 
    "distro": "OpenWrt", 
    "target": "mvebu", 
    "subtarget": "cortexa9"
}
```

Answer:
`{"error": "unknown device, please check model and board params"}`

tlaurion, 2018-09-24:
@chunkeey ?

chunkeey, 2018-09-24:
@tlaurion ?

I don't have any mvebu devices, but my guess is that the device's board name (armada-385-linksys-rango) does not match the sysupgrade file name (linksys-wrt3200acm-squashfs-sysupgrade.bin) . Problem is, that most mvebu devices have conflicting names... and adding them by hand will be tedious.

If you want to invest some time, I think @aparcar would appreciate having a metadata parser that could (try to) extract the metadata from all the sysupgrade* files and then use the data to generate a dynamic board rename mapping file on the fly... What do you think?

tlaurion, 2018-09-25:
Would gladly help.

tlaurion, 2018-10-05:
@aparcar @chunkeey : Where can I help?

aparcar, 2018-10-05:
@tlaurion I'm currently on vacations and therefore somewhat inactive. If you want to help with the *mapping project* please create a new issue describing the idea, a first step which would greatly help me implementing that feature would be to tell me how to read the firmware meta data. I could then automatically build all profiles and extract the "internal" name.

aparcar, 2018-10-05:
Regarding the issues here, please test the following upgrade server as the *old* one dropped snapshot support:

https://as-test.stephen304.com/

tlaurion, 2018-10-06:
@aparcar for some reason when attempting attended sysupgrade I get: 
"Upgrade server down or could not connect" using https://as-test.stephen304.com

tlaurion, 2018-10-17:
@aparcar sorry for pinging back. But pinging back. :)

aparcar, 2018-10-17:
@tlaurion sorry I'm still a bit busy, I'll try to work things out!

tlaurion, 2018-10-18:
@aparcar what's up with the test server? Is it just my device not being able to poke it on a network level?!

tlaurion, 2018-10-24:
@aparcar : I get a` error 405 not allowed` accessing test server from Attended Sysupgrade LUCI package.

Configured Attended Sysupgrade server: `https://as-test.stephen304.com/chef`

Request url:
`https://as-test.stephen304.com/chef/api/upgrade-check`

Request header:
```
Host: as-test.stephen304.com
User-Agent: Mozilla/5.0 (Windows NT 6.1; rv:60.0) Gecko/20100101 Firefox/60.0
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8
Accept-Language: en-US,en;q=0.5
Accept-Encoding: gzip, deflate, br
Access-Control-Request-Method: POST
Access-Control-Request-Headers: content-type
Origin: http://typycalhiddentorservice.onion
Connection: keep-alive
```

Header Response:
```
HTTP/1.1 405 Not Allowed
Server: nginx/1.10.3
Date: Wed, 24 Oct 2018 16:00:47 GMT
Content-Type: text/html
Content-Length: 173
```

tlaurion, 2019-07-14:
@aparcar:
Tested and works from:
https://chef.libremesh.org

tlaurion, 2019-07-14:
@aparcar:
Snapshot still results in "Unsupported device, release, target, subtraget or board"....

```
board	linksys,rango
board_name	linksys,rango
distro	OpenWrt
model	Linksys WRT3200ACM
packages	[…]
0	base-files
1	libc
2	busybox
3	dropbear
4	mtd
5	uci
6	opkg
7	netifd
8	fstools
9	uclient-fetch
10	logd
11	swconfig
12	dnsmasq
13	iptables
14	ip6tables
15	ppp
16	ppp-mod-pppoe
17	firewall
18	odhcpd-ipv6only
19	odhcp6c
20	iwinfo
21	kmod-gpio-button-hotplug
22	kmod-ipt-offload
23	kmod-btmrvl
24	kmod-mwifiex-sdio
25	luci
26	luci-app-attendedsysupgrade
27	uboot-envtools
28	kernel
29	wpad-basic
30	kmod-mwlwifi
31	mwlwifi-firmware-88w8964
32	urandom-seed
33	urngd
revision	r10497-f1875e902d
target	mvebu/cortexa9
upgrade_packages	1
version	snapshot
```

aparcar, 2020-02-25:
Image requests work on the new server at https://chef.libremesh.org/

I have to reimplement the LuCI app which will take a little while...

[image]

closed 2020-02-25 by aparcar

aparcar, 2020-02-25:
I started a [mail thread](https://lists.infradead.org/pipermail/openwrt-adm/2020-February/001342.html) to agree on some basics. If this goes somewhere I can build a LuCI app on that base.