openwrt/asu #131: Problem compiling snapshot images

julianocss, 2018-10-10:
I´m trying to compile a snapshot image for my WR840N V4 ( ramips / mt76x8 ) but i´m having the following erros:

"
Collected errors:
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for zram-swap:
 * 	kernel (= 4.14.74-1-1d3aa8bc4469e6816f0603da2e58073e) * 	kernel (= 4.14.74-1-1d3aa8bc4469e6816f0603da2e58073e) * 	kernel (= 4.14.74-1-1d3aa8bc4469e6816f0603da2e58073e) * 	kernel (= 4.14.74-1-1d3aa8bc4469e6816f0603da2e58073e) * 
 * opkg_install_cmd: Cannot install package zram-swap.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for sqm-scripts-extra:
 * 	kernel (= 4.14.74-1-1d3aa8bc4469e6816f0603da2e58073e) * 	kernel (= 4.14.74-1-1d3aa8bc4469e6816f0603da2e58073e) * 	kernel (= 4.14.74-1-1d3aa8bc4469e6816f0603da2e58073e) * 	kernel (= 4.14.74-1-1d3aa8bc4469e6816f0603da2e58073e) * 	kernel (= 4.14.74-1-1d3aa8bc4469e6816f0603da2e58073e) * 	kernel (= 4.14.74-1-1d3aa8bc4469e6816f0603da2e58073e) * 	kernel (= 4.14.74-1-1d3aa8bc4469e6816f0603da2e58073e) * 	kernel (= 4.14.74-1-1d3aa8bc4469e6816f0603da2e58073e) * 	kernel (= 4.14.74-1-1d3aa8bc4469e6816f0603da2e58073e) * 	kernel (= 4.14.74-1-1d3aa8bc4469e6816f0603da2e58073e) * 	kernel (= 4.14.74-1-1d3aa8bc4469e6816f0603da2e58073e) * 
 * opkg_install_cmd: Cannot install package sqm-scripts-extra.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for ppp-mod-pppoe:
 * 	kernel (= 4.14.74-1-1d3aa8bc4469e6816f0603da2e58073e) * 	kernel (= 4.14.74-1-1d3aa8bc4469e6816f0603da2e58073e) * 	kernel (= 4.14.74-1-1d3aa8bc4469e6816f0603da2e58073e) * 	kernel (= 4.14.74-1-1d3aa8bc4469e6816f0603da2e58073e) * 	kernel (= 4.14.74-1-1d3aa8bc4469e6816f0603da2e58073e) * 
 * opkg_install_cmd: Cannot install package ppp-mod-pppoe.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-app-upnp:
 * 	kernel (= 4.14.74-1-1d3aa8bc4469e6816f0603da2e58073e) * 	kernel (= 4.14.74-1-1d3aa8bc4469e6816f0603da2e58073e) * 	kernel (= 4.14.74-1-1d3aa8bc4469e6816f0603da2e58073e) * 
 * opkg_install_cmd: Cannot install package luci-app-upnp.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci:
 * 	kernel (= 4.14.74-1-1d3aa8bc4469e6816f0603da2e58073e) * 	kernel (= 4.14.74-1-1d3aa8bc4469e6816f0603da2e58073e) * 	kernel (= 4.14.74-1-1d3aa8bc4469e6816f0603da2e58073e) * 
 * opkg_install_cmd: Cannot install package luci.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-mt76:
 * 	kernel (= 4.14.74-1-1d3aa8bc4469e6816f0603da2e58073e) * 	kernel (= 4.14.74-1-1d3aa8bc4469e6816f0603da2e58073e) * 	kernel (= 4.14.74-1-1d3aa8bc4469e6816f0603da2e58073e) * 	kernel (= 4.14.74-1-1d3aa8bc4469e6816f0603da2e58073e) * 	kernel (= 4.14.74-1-1d3aa8bc4469e6816f0603da2e58073e) * 	kernel (= 4.14.74-1-1d3aa8bc4469e6816f0603da2e58073e) * 	kernel (= 4.14.74-1-1d3aa8bc4469e6816f0603da2e58073e) * 	kernel (= 4.14.74-1-1d3aa8bc4469e6816f0603da2e58073e) * 
 * opkg_install_cmd: Cannot install package kmod-mt76.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-leds-gpio:
 * 	kernel (= 4.14.74-1-1d3aa8bc4469e6816f0603da2e58073e) * 
 * opkg_install_cmd: Cannot install package kmod-leds-gpio.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-ipt-offload:
 * 	kernel (= 4.14.74-1-1d3aa8bc4469e6816f0603da2e58073e) * 	kernel (= 4.14.74-1-1d3aa8bc4469e6816f0603da2e58073e) * 
 * opkg_install_cmd: Cannot install package kmod-ipt-offload.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-gpio-button-hotplug:
 * 	kernel (= 4.14.74-1-1d3aa8bc4469e6816f0603da2e58073e) * 
 * opkg_install_cmd: Cannot install package kmod-gpio-button-hotplug.
make[2]: *** [package_install] Error 255
make[1]: *** [_call_manifest] Error 2
make: *** [manifest] Error 2
"

Any ideas?
Juliano

aparcar, 2018-10-18:
I think the problem was that newly downloaded and extracted snapshot archives did not overwrite existing files and just silently skip them. This should be fixed now, please try!

julianocss, 2018-10-25:
I reopened this issue because i had problems now, when i tried build latest snapshot image.

Collected errors:
 * check_data_file_clashes: Package wpad-basic wants to install file /mnt/worker/worker1/imagebuilder/openwrt/snapshots/ramips/mt76x8/build_dir/target-mipsel_24kc_musl/root-ramips/usr/sbin/hostapd
	But that file is already provided by package  * wpad-mini
 * check_data_file_clashes: Package wpad-basic wants to install file /mnt/worker/worker1/imagebuilder/openwrt/snapshots/ramips/mt76x8/build_dir/target-mipsel_24kc_musl/root-ramips/usr/sbin/wpa_supplicant
	But that file is already provided by package  * wpad-mini
 * check_data_file_clashes: Package wpad-basic wants to install file /mnt/worker/worker1/imagebuilder/openwrt/snapshots/ramips/mt76x8/build_dir/target-mipsel_24kc_musl/root-ramips/usr/sbin/wpad
	But that file is already provided by package  * wpad-mini
 * opkg_install_cmd: Cannot install package wpad-basic.
 * check_data_file_clashes: Package wpad-basic wants to install file /mnt/worker/worker1/imagebuilder/openwrt/snapshots/ramips/mt76x8/build_dir/target-mipsel_24kc_musl/root-ramips/usr/sbin/hostapd
	But that file is already provided by package  * wpad-mini
 * check_data_file_clashes: Package wpad-basic wants to install file /mnt/worker/worker1/imagebuilder/openwrt/snapshots/ramips/mt76x8/build_dir/target-mipsel_24kc_musl/root-ramips/usr/sbin/wpa_supplicant
	But that file is already provided by package  * wpad-mini
 * check_data_file_clashes: Package wpad-basic wants to install file /mnt/worker/worker1/imagebuilder/openwrt/snapshots/ramips/mt76x8/build_dir/target-mipsel_24kc_musl/root-ramips/usr/sbin/wpad
	But that file is already provided by package  * wpad-mini
 * opkg_install_cmd: Cannot install package wpad-basic.
make[2]: *** [package_install] Error 255
make[1]: *** [_call_manifest] Error 2
make: *** [manifest] Error 2

aparcar, 2018-10-30:
This commit should solve the issue, snapshots are somewhat quickly changing their names, defaults and profile names, therefore they cause plenty of additional requirements, which were initially not planned

closed 2019-01-23 by aparcar