openwrt/asu #132: Mapping internal name to rom binary name project

tlaurion, 2018-10-17:
@aparcar from [here](https://github.com/aparcar/attendedsysupgrade-server/issues/129#issuecomment-424122630)
> If you want to help with the _mapping project_ please create a new issue describing the idea, a first step which would greatly help me implementing that feature would be to tell me how to read the firmware meta data. I could then automatically build all profiles and extract the "internal" name.

tlaurion, 2018-10-17:
@aparcar 
From my understanding:

[DEVICE_DTS (board name)](https://github.com/openwrt/openwrt/blob/master/target/linux/mvebu/image/cortex-a9.mk#L41) should be linked to [TARGET_DEVICES (sysupgrade file name)](https://github.com/openwrt/openwrt/blob/master/target/linux/mvebu/image/cortex-a9.mk#L44)

tlaurion, 2018-10-17:
@aparcar just a small PoC.
under `/home/user/openwrt/target`
`find ./ -name *.mk|while read line; do grep -rHn -e Device -e DTS $line; done|less`
returns DTS lines for which the line before, if present, is the actual sysupgrade file name.

Exerpt:
```
./linux/mvebu/image/cortex-a9.mk:22:define Device/linksys-wrt1200ac
./linux/mvebu/image/cortex-a9.mk:23:  $(call Device/linksys,WRT1200AC (Caiman))
./linux/mvebu/image/cortex-a9.mk:24:  $(Device/armada-385-linksys)
./linux/mvebu/image/cortex-a9.mk:25:  DEVICE_DTS := armada-385-linksys-caiman
./linux/mvebu/image/cortex-a9.mk:30:define Device/linksys-wrt1900acv2
./linux/mvebu/image/cortex-a9.mk:31:  $(call Device/linksys,WRT1900ACv2 (Cobra))
./linux/mvebu/image/cortex-a9.mk:32:  $(Device/armada-385-linksys)
./linux/mvebu/image/cortex-a9.mk:33:  DEVICE_DTS := armada-385-linksys-cobra
./linux/mvebu/image/cortex-a9.mk:38:define Device/linksys-wrt3200acm
./linux/mvebu/image/cortex-a9.mk:39:  $(call Device/linksys,WRT3200ACM (Rango))
./linux/mvebu/image/cortex-a9.mk:40:  $(Device/armada-385-linksys)
./linux/mvebu/image/cortex-a9.mk:41:  DEVICE_DTS := armada-385-linksys-rango
./linux/mvebu/image/cortex-a9.mk:46:define Device/linksys-wrt1900acs
./linux/mvebu/image/cortex-a9.mk:47:  $(call Device/linksys,WRT1900ACS (Shelby))
./linux/mvebu/image/cortex-a9.mk:48:  $(Device/armada-385-linksys)
./linux/mvebu/image/cortex-a9.mk:49:  DEVICE_DTS := armada-385-linksys-shelby
./linux/mvebu/image/cortex-a9.mk:54:define Device/linksys-wrt32x
./linux/mvebu/image/cortex-a9.mk:55:$(call Device/linksys,WRT32X (Venom))
./linux/mvebu/image/cortex-a9.mk:56:  DEVICE_DTS := armada-385-linksys-venom
./linux/mvebu/image/cortex-a9.mk:58:  $(Device/armada-385-linksys)
./linux/mvebu/image/cortex-a9.mk:64:define Device/linksys-wrt1900ac
./linux/mvebu/image/cortex-a9.mk:65:  $(call Device/linksys,WRT1900AC (Mamba))
./linux/mvebu/image/cortex-a9.mk:66:  DEVICE_DTS := armada-xp-linksys-mamba
./linux/mvebu/image/cortex-a9.mk:68:  $(Device/NAND-128K)
./linux/mvebu/image/cortex-a9.mk:69:  $(Device/UBI-factory)
./linux/mvebu/image/cortex-a9.mk:74:define Device/openblocks-ax3-4
./linux/mvebu/image/cortex-a9.mk:75:  $(Device/UBI-factory)
./linux/mvebu/image/cortex-a9.mk:76:  DEVICE_DTS := armada-xp-openblocks-ax3-4
```

aparcar, 2018-10-23:
Hey, her is a dump for all current snapshot builders, could you continue with your script?
https://as-test.stephen304.com/download/device_dump.txt

chunkeey, 2018-10-23:
>From my understanding:
>
>DEVICE_DTS (board name) should be linked to TARGET_DEVICES (sysupgrade file name)

This could be true for mvebu, but it is not going to be the case for all other boards.
Extracting the right $boardname for all boards will be a tricky part. Because not all architectures
use the device-tree based discovery that Mathias implemented in:
https://github.com/openwrt/openwrt/commit/ac3e05c5d757334944fbf1f14583164cd1d69172#diff-e41e84796e8517b061c960b1d32c86db

yet. But apart from x86 they are mostly legacy architectures (so maybe you can ignore them?)

But:
If you want a "one solution for all device-tree base", you would need to look into the .dts files and extract the "Platform" compatible string as
per: 
https://github.com/torvalds/linux/blob/master/Documentation/devicetree/usage-model.txt#L95

and match that to the sysupgrade file. And yeah, this would involve alot of guessing work.

chunkeey, 2018-10-24:
> would it be a cleaner solution to build all images once and then extract the metadata? As the >sysupgrade command can extract the data, it should be possible in a generic way, right?

Yes :laughing: :+1:  . I initially wrote something like this too:
https://github.com/aparcar/attendedsysupgrade-server/issues/129#issuecomment-424122630

"@aparcar would appreciate having a metadata parser that could (try to) extract the metadata from all the sysupgrade* files and then use the data to generate a dynamic board rename mapping file on the fly... What do you think?"

I think this should work for all images that do append the metadata to the images for the foreseeable future.
 
Here's an example for what the metadata of the rpi3 OpenWrt 18.06.1 release looks like:

     # fwtool -i metadata openwrt-18.06.1-brcm2708-bcm2710-rpi-3-squashfs-sysupgrade.img.gz
     # cat metadata 
     { 
        "supported_devices" : 
           [
               "rpi-3-b",
               "rpi-3-b-plus",
               "raspberrypi,3-model-b", 
               "raspberrypi,3-model-b-plus"
           ],
        "version" :
         { 
           "dist": "OpenWrt", "version": "18.06.1",
           "revision": "r7258-5eb055306f", 
           "board": "brcm2708" 
         }
     }

So "openwrt-18.06.1-brcm2708-bcm2710-rpi-3-squashfs-sysupgrade.img.gz" will need to be available for all devices that match one string of the supported_devices array (e.g.: either "rpi-3-b", "rpi-3-b-plus", "raspberrypi,3-model-b" or  "raspberrypi,3-model-b-plus").

What needs to be addressed though: There could be multiple image for the same board.
For example there is also a "ext4"  image for the RPI-3 ... as well as the RPI-2 squash and
ext4 images:

     # fwtool -i metadata openwrt-18.06.1-brcm2708-bcm2709-rpi-2-ext4-sysupgrade.img.gz
     # cat metadata 
     { 
        "supported_devices" : 
           [
               "rpi-2-b",
               "rpi-3-b",
               "rpi-3-b-plus",
               "rpi-cm",
               "raspberrypi,2-model-b",
               "raspberrypi,3-model-b",
               "raspberrypi,3-model-b-plus",
               "raspberrypi,compute-module-3"
           ],
        "version" :
         { 
           "dist": "OpenWrt", "version": "18.06.1",
           "revision": "r7258-5eb055306f", 
           "board": "brcm2708" 
         }
     }

So the attendend-sysupgrade would need something like a board selection page where one could choose between the RPI-3 images (squash or ext4) and the RPI-2 Images (squash or ext4) .

As for the fwtool: It gets build during the image creation process as part of the host tools and unfortunately, it's not a separate project so you'll have to get it from:
https://github.com/openwrt/openwrt/tree/master/package/system/fwtool

That all said, there might be an alternative to parsing the sysupgrade files. But it will necessary to
convince the openwrt folks to "extend the current build code" in a way so it generates a mapping file
during the build process... Sort of like the config.seeds, target.manifests and checksum and gpg signature files are already done and become a part of the "Supplementary Files". This could have the advantage that other projects could use it. I could see that the "Table of Hardware" Maintainers might be able to use this information so new devices pages could automatically be generated... etc. (But yeah, "wishful thinking") 
built.
 
Another

(That said, it might s also possible to "extend" the image creation scripts

aparcar, 2018-10-24:
I wasn't aware of the tool and its functionality! Thank you very much this is exactly what I was looking for! I'll implement it and publish the data one I'm done

aparcar, 2018-10-30:
Some images are missing but looks good so far http://sprunge.us/3n8f0K

chunkeey, 2018-10-31:
> Some 750 images are missing but looks good so far http://sprunge.us/3n8f0K
> 
> The script I used:
> 
> ```
> for upgrade in $(find ./server/download/openwrt/18.06.1 -name "*sysupgrade*"); do
>         tail="${upgrade%*/*/*}"
>         echo -n "${tail##*/*/} "
>         ./fwtool -i metadata "$upgrade" 2>&1
>         cat metadata
> done
> ```

Can I assume that your local "/server/download/openwrt/18.06.1" is basically what' is on http://downloads.openwrt.org/releases/18.06.1/ ( same filenames and directory structure )?

It's odd, that while all the ipq40xx, lantiq, mvebu, ... are there...  none of the ipq806x (R7800, C2600, ...) , apm821xx (Meraki MR24/MX60, Netgear WNDR4700) or raspberry pis images are found. But the missing archs are listed on https://as-test.stephen304.com/supported . :man_shrugging:

aparcar, 2018-11-01:
Sorry the previous message was a bit unclear, I meant that some 750 images are about to be build and therefore the previous link was only a preview.

[Here](http://sprunge.us/jY10u7) is a list containing all sysupgrade images found, every profile of 18.01.6 was build, some faild

aparcar, 2018-11-01:
@tlaurion could you create a mapping based on the file I provide?

chunkeey, 2018-11-01:
dump-to-mapping-file.sh

```
#!/bin/sh

# This needs the "jq" utility installed

[[ $(command -v jq) ]] || exit 1

# First argument is the filename (the dump)
[[ $# -eq 1 ]] || exit 1

INPUTFILE="$1"

BADENTRIES=""
while IFS=' ' read -r file metadata && [[ -n "$file" ]] ; do
        [[ "$metadata" == *"supported_devices"* ]] || {
                BADENTRIES+="'$file' had '$metadata'"$'\n'
                continue
        }

        value="$file" 
        devices=$(echo "$metadata" | jq -r ".supported_devices[]")

        for device in $devices; do
                echo "$device" : "$value"
        done
done < "$INPUTFILE"

(1>&2 echo "$BADENTRIES")
```

$ test.sh 2 > errors

This produces something like:

https://pastebin.com/aR0T5DF5

OT: Are there still some missing images? 
And also: One image seems bad: 'ramips/rt3883/rt-n56u' had 'CRC error'

aparcar, 2019-02-01:
I reworked the script a bit and came up with this

```
BADENTRIES=""
for upgrade in $(find . -name "*sysupgrade*"); do
        tail="${upgrade%*/*/*}"
        ~/fwtool -i metadata "$upgrade" 2>&1
	metadata="$(cat metadata)"

        [[ "$metadata" == *"supported_devices"* ]] || {
                BADENTRIES+="'${tail##*/*/}' had '$metadata'"$'\n'
                continue
        }
        devices=$(echo "$metadata" | jq -r ".supported_devices[]")
        for device in $devices; do
                echo "$device" : "${tail##*/*/}"
        done
done

(1>&2 echo "$BADENTRIES")
```

The output of `bash mapping.sh | sort | uniq > mapping.txt` is found [here](https://chef.libremesh.org/download/openwrt/18.06.1/mapping.txt)

chunkeey, 2019-02-01:
Not 18.06.2?

aparcar, 2019-02-01:
@chunkeey Will do for both, but as it's a service release I don't think anything changes.

> (Also, I think the issue is no longer an issue +1 )

Why not?

chunkeey, 2019-02-01:
> @chunkeey Will do for both, but as it's a service release I don't think anything changes.

That's true. The mapping should work for all 18.06 releases.

> > (Also, I think the issue is no longer an issue +1 )
> 
> Why not?

I think the motivation and project description for the issue in https://github.com/aparcar/attendedsysupgrade-server/issues/132#issue-371112947 

> If you want to help with the mapping project please create a new issue describing the idea, a first step which would greatly help me implementing that feature would be to tell me how to read the firmware meta data. I could then automatically build all profiles and extract the "internal" name.
>
> If you want to invest some time, I think @aparcar would appreciate having a metadata parser that could (try to) extract the metadata from all the sysupgrade* files and then use the data to generate a dynamic board rename mapping file on the fly... What do you think?

has been fulfilled? Or is there something that still needs to be addressed?

aparcar, 2020-02-25:
The project now uses an dict containing the OpenWrt profile names and the supported_devices (ideally both are the same) as keys. I think the issue is solved by this.

closed 2020-02-25 by aparcar