openwrt/asu #133: "error 405 not allowed" when accessing test server from Attended Sysupgrade LUCI package

tlaurion, 2018-11-18:
@aparcar : I get a` error 405 not allowed` accessing test server from Attended Sysupgrade LUCI package.

Configured Attended Sysupgrade server: `https://as-test.stephen304.com/chef`

Request url:
`https://as-test.stephen304.com/chef/api/upgrade-check`

Request header:
```
Host: as-test.stephen304.com
User-Agent: Mozilla/5.0 (Windows NT 6.1; rv:60.0) Gecko/20100101 Firefox/60.0
Accept: text/html,application/xhtml+xml,application/xml;q=0.9,*/*;q=0.8
Accept-Language: en-US,en;q=0.5
Accept-Encoding: gzip, deflate, br
Access-Control-Request-Method: POST
Access-Control-Request-Headers: content-type
Origin: http://typycalhiddentorservice.onion
Connection: keep-alive
```

Header Response:
```
HTTP/1.1 405 Not Allowed
Server: nginx/1.10.3
Date: Wed, 24 Oct 2018 16:00:47 GMT
Content-Type: text/html
Content-Length: 173
```

tlaurion, 2018-12-09:
@aparcar : Am I doing something wrong? Is my isp blocked or something? :|

aparcar, 2018-12-11:
The correct url is ` https://as-test.stephen304.com/api/upgrade-check`, without the *chef* in between :+1:

aparcar, 2018-12-11:
However, there seems to be a bad SQL query which basically kills the server once you ask for an upgrade. I'll have to fix that first

tlaurion, 2019-01-22:
@aparcar: Any update?

aparcar, 2019-01-24:
Please try again, while [this](https://github.com/aparcar/attendedsysupgrade-server/pull/141) isn't perfect, it should to the job for now...

aparcar, 2020-03-07:
I reworked the server and app and currently everything seems to work, please reopen in case of trouble!

closed 2020-03-07 by aparcar