openwrt/asu #134: linksys-wrt3200acm - test server image creation from website - kernel dependency errors

tlaurion, 2018-11-20:
Hi @aparcar.

[Here is the error log ](https://as-test.stephen304.com/download/faillogs/faillog-6c5228971ba9.txt) that  I get when attempting to create image:
```
### BUILD COMMAND:

PACKAGES=wpad-basic uclient-fetch uci uboot-envtools swconfig ppp-mod-pppoe ppp opkg odhcpd-ipv6only odhcp6c netifd mwlwifi-firmware-88w8964 mtd luci logd libgcc kmod-mwlwifi kmod-mwifiex-sdio kmod-ipt-offload kmod-gpio-button-hotplug kmod-btmrvl iwinfo iptables ip6tables fstools firewall dropbear dnsmasq busybox base-files
DISTRO=openwrt
SUBTARGET=cortexa9
REQUEST_HASH=6c5228971ba9
PROFILE=linksys-wrt3200acm
DEFAULTS_HASH=
ID=1491
TARGET=mvebu
VERSION=snapshots
IMAGE_HASH=None
sh meta

### STDERR:

gpg: key CD84BCED626471F1: "LEDE Build System (LEDE GnuPG key for unattended build jobs) <lede-adm@lists.infradead.org>" not changed
gpg: Total number processed: 1
gpg:              unchanged: 1
gpg: key 833C6010D52BBB6B: "LEDE Release Builder (17.01 "Reboot" Signing Key) <lede-dev@lists.lede-project.org>" not changed
gpg: Total number processed: 1
gpg:              unchanged: 1
gpg: key 0F20257417E1CE16: "OpenWrt Release Builder (18.06 Signing Key) <openwrt-devel@lists.openwrt.org>" not changed
gpg: Total number processed: 1
gpg:              unchanged: 1
download checksums and signature
--2018-11-19 20:27:54--  https://downloads.openwrt.org/snapshots/targets/mvebu/cortexa9/sha256sums
Resolving downloads.openwrt.org (downloads.openwrt.org)... 148.251.78.235, 2a01:4f8:202:43ea::2
Connecting to downloads.openwrt.org (downloads.openwrt.org)|148.251.78.235|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 109889 (107K) [text/plain]
Saving to: â€sha256sumsâ€™

     0K .......... .......... .......... .......... .......... 46%  487K 0s
    50K .......... .......... .......... .......... .......... 93%  492K 0s
   100K .......                                               100% 21.6M=0.2s

2018-11-19 20:27:55 (525 KB/s) - â€sha256sumsâ€™ saved [109889/109889]

--2018-11-19 20:27:55--  https://downloads.openwrt.org/snapshots/targets/mvebu/cortexa9/sha256sums.gpg
Resolving downloads.openwrt.org (downloads.openwrt.org)... 148.251.78.235, 2a01:4f8:202:43ea::2
Connecting to downloads.openwrt.org (downloads.openwrt.org)|148.251.78.235|:443... connected.
HTTP request sent, awaiting response... 301 Moved Permanently
Location: https://downloads.openwrt.org/snapshots/targets/mvebu/cortexa9/sha256sums.asc [following]
--2018-11-19 20:27:55--  https://downloads.openwrt.org/snapshots/targets/mvebu/cortexa9/sha256sums.asc
Reusing existing connection to downloads.openwrt.org:443.
HTTP request sent, awaiting response... 200 OK
Length: 837 [text/plain]
Saving to: â€sha256sums.ascâ€™

     0K                                                       100% 16.2M=0s

2018-11-19 20:27:56 (16.2 MB/s) - â€sha256sums.ascâ€™ saved [837/837]

verifying sha256sums signature
gpg: assuming signed data in 'sha256sums'
gpg: Signature made Mon 19 Nov 2018 08:39:25 AM EST
gpg:                using RSA key F93525A88B699029
gpg: Good signature from "LEDE Build System (LEDE GnuPG key for unattended build jobs) <lede-adm@lists.infradead.org>" [ultimate]
verified sha256sums signature.
ImageBuilder is up-to-date
Downloading http://downloads.openwrt.org/snapshots/targets/mvebu/cortexa9/packages/Packages.gz
Updated list of available packages in /mnt/worker/worker2/imagebuilder/openwrt/snapshots/mvebu/cortexa9/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/root-mvebu/../../../../../../../../../../../mnt/worker/worker2/imagebuilder/openwrt/snapshots/mvebu/cortexa9/dl/openwrt_core
Downloading http://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3/base/Packages.gz
Updated list of available packages in /mnt/worker/worker2/imagebuilder/openwrt/snapshots/mvebu/cortexa9/build_dir/target-arm_cortex-a9+vfpv3_musl_eabi/root-mvebu/../../../../../../../../../../../mnt/worker/worker2/imagebuilder/openwrt/snapshots/mvebu/cortexa9/dl/openwrt_base
Downloading http://downloads.openwrt.org/snapshots/packages/arm_cortex-a9_vfpv3/luci/Packages.gz
[... 13 lines cut ...]
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for mwlwifi-firmware-88w8964:
 * 	kernel (= 4.14.81-1-3ebff4a41f399d57ae9c057531a9667b) * 	kernel (= 4.14.81-1-3ebff4a41f399d57ae9c057531a9667b) * 	kernel (= 4.14.81-1-3ebff4a41f399d57ae9c057531a9667b) * 
 * opkg_install_cmd: Cannot install package mwlwifi-firmware-88w8964.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci:
 * 	kernel (= 4.14.81-1-3ebff4a41f399d57ae9c057531a9667b) * 	kernel (= 4.14.81-1-3ebff4a41f399d57ae9c057531a9667b) * 	kernel (= 4.14.81-1-3ebff4a41f399d57ae9c057531a9667b) * 	kernel (= 4.14.81-1-3ebff4a41f399d57ae9c057531a9667b) * 	kernel (= 4.14.81-1-3ebff4a41f399d57ae9c057531a9667b) * 	kernel (= 4.14.81-1-3ebff4a41f399d57ae9c057531a9667b) * 	kernel (= 4.14.81-1-3ebff4a41f399d57ae9c057531a9667b) * 	kernel (= 4.14.81-1-3ebff4a41f399d57ae9c057531a9667b) * 
 * opkg_install_cmd: Cannot install package luci.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-mwifiex-sdio:
 * 	kernel (= 4.14.81-1-3ebff4a41f399d57ae9c057531a9667b) * 	kernel (= 4.14.81-1-3ebff4a41f399d57ae9c057531a9667b) * 
 * opkg_install_cmd: Cannot install package kmod-mwifiex-sdio.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-ipt-offload:
 * 	kernel (= 4.14.81-1-3ebff4a41f399d57ae9c057531a9667b) * 	kernel (= 4.14.81-1-3ebff4a41f399d57ae9c057531a9667b) * 
 * opkg_install_cmd: Cannot install package kmod-ipt-offload.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-gpio-button-hotplug:
 * 	kernel (= 4.14.81-1-3ebff4a41f399d57ae9c057531a9667b) * 
 * opkg_install_cmd: Cannot install package kmod-gpio-button-hotplug.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for kmod-btmrvl:
 * 	kernel (= 4.14.81-1-3ebff4a41f399d57ae9c057531a9667b) * 	kernel (= 4.14.81-1-3ebff4a41f399d57ae9c057531a9667b) * 	kernel (= 4.14.81-1-3ebff4a41f399d57ae9c057531a9667b) * 	kernel (= 4.14.81-1-3ebff4a41f399d57ae9c057531a9667b) * 	kernel (= 4.14.81-1-3ebff4a41f399d57ae9c057531a9667b) * 	kernel (= 4.14.81-1-3ebff4a41f399d57ae9c057531a9667b) * 	kernel (= 4.14.81-1-3ebff4a41f399d57ae9c057531a9667b) * 	kernel (= 4.14.81-1-3ebff4a41f399d57ae9c057531a9667b) * 	kernel (= 4.14.81-1-3ebff4a41f399d57ae9c057531a9667b) * 	kernel (= 4.14.81-1-3ebff4a41f399d57ae9c057531a9667b) * 	kernel (= 4.14.81-1-3ebff4a41f399d57ae9c057531a9667b) * 	kernel (= 4.14.81-1-3ebff4a41f399d57ae9c057531a9667b) * 	kernel (= 4.14.81-1-3ebff4a41f399d57ae9c057531a9667b) * 	kernel (= 4.14.81-1-3ebff4a41f399d57ae9c057531a9667b) * 	kernel (= 4.14.81-1-3ebff4a41f399d57ae9c057531a9667b) * 	kernel (= 4.14.81-1-3ebff4a41f399d57ae9c057531a9667b) * 	kernel (= 4.14.81-1-3ebff4a41f399d57ae9c057531a9667b) * 	kernel (= 4.14.81-1-3ebff4a41f399d57ae9c057531a9667b) * 	kernel (= 4.14.81-1-3ebff4a41f399d57ae9c057531a9667b) * 	kernel (= 4.14.81-1-3ebff4a41f399d57ae9c057531a9667b) * 	kernel (= 4.14.81-1-3ebff4a41f399d57ae9c057531a9667b) * 
 * opkg_install_cmd: Cannot install package kmod-btmrvl.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for ip6tables:
 * 	kernel (= 4.14.81-1-3ebff4a41f399d57ae9c057531a9667b) * 	kernel (= 4.14.81-1-3ebff4a41f399d57ae9c057531a9667b) * 	kernel (= 4.14.81-1-3ebff4a41f399d57ae9c057531a9667b) * 
 * opkg_install_cmd: Cannot install package ip6tables.
make[2]: *** [package_install] Error 255
make[1]: *** [_call_manifest] Error 2
make: *** [manifest] Error 2
```

aparcar, 2018-12-09:
Hey @tlaurion , thank you very much for your continuous testing. Problem was that extracting a new snapshot over an existing one, keeps various kernel versions. The ImageBuilder seems to choose the one to install alphabetically, instead of the most recent version. Via https://github.com/aparcar/meta-imagebuilder/commit/c9187e5ec547bc0b25a01a095aa957e4afc8ff2e#diff-e9a23cbc455158951716b440c3d165e0R77 all previous files are removed and it hopefully works now!

tlaurion, 2019-07-10:
@aparcar : is it working now?

aparcar, 2019-08-17:
This should be (finally) fixed via https://github.com/aparcar/openwrt-metabuilder/commit/62661b5336e3a25e7b252c7bcca6ed1f273169f0 and is deployed on https://aparcar.stephen304.com/, please give it a try!

aparcar, 2020-02-25:
Please test the new server at https://chef.libremesh.org/

[image]

closed 2020-02-25 by aparcar