openwrt/asu #135: Add INSTALL documentation

FreifunkUFO, 2018-12-14:
please provide documentation howto install and host such a server on a standard ubuntu, debian and/or openwrt installation. (including chef)

see also https://github.com/aparcar/attendedsysupgrade-server/issues/105 and https://github.com/aparcar/attendedsysupgrade-server/issues/104 , which were mentioned before.

FreifunkUFO, 2018-12-14:
we could set up a new ubuntu or openwrt instance on one of our virtualisation-hosts to prove these documentation

aparcar, 2018-12-19:
Please check [this branch](https://github.com/aparcar/attendedsysupgrade-server/blob/mixing/README.md#run-your-own-server) with some documentation. 

I'd be happy if you could try the described instructions and help out to improve it!

robimarko, 2018-12-19:
Hm, there is an issue with ansible.
It will connect via ssh and then it complains about following:
`TASK [asu : include vars] *****************************************************************************************************************************************************************************************
task path: /root/attendedsysupgrade-server/ansible/roles/asu/tasks/main.yml:1
fatal: [sysupgrade.otvorenamreza.org]: FAILED! => {
    "msg": "The task includes an option with an undefined variable. The error was: 'config_file' is undefined\n\nThe error appears to have been in '/root/attendedsysupgrade-server/ansible/roles/asu/tasks/main.yml': line 1, column 3, but may\nbe elsewhere in the file depending on the exact syntax problem.\n\nThe offending line appears to be:\n\n\n- name: include vars\n  ^ here\n"
}
	to retry, use: --limit @/root/attendedsysupgrade-server/ansible/site.retry`

robimarko, 2018-12-20:
Tried using Docker version and that fails as well as Vagrant.

FreifunkUFO, 2019-01-04:
so i was also trying to install it:
* we dont have an asu server, but at the moment debian and ubuntu.
* there was no checkout-command mentioned. so i hope its ok to have "git clone https://github.com/aparcar/attendedsysupgrade-server.git" or do i need this (mixing) brunch?
 * there is a typo, "except" where you "expect" people to have some knowledge.
* top run the "vagrant up" command i had to install "vagrant" first. it should be mentioned in the "packages to install"
* w@t:~/git/attendedsysupgrade-server$ vagrant up
No usable default provider could be found for your system.

*so now i have to install virtualbox inside of a virtual machine (hosted by virt-manager) :-/ is there a install tipp for using without vangrat or ansibel? i even dont have docker on our virtual maschine hosts.

*virtualbox installation is done, now that vagrant script is about installing debian... so that will take some time

robimarko, 2019-01-04:
It will fail as there is an issue with the ansible playbook and config file

aparcar, 2019-01-08:
@robimarko please feel free to create a PR using KVM, I just took what looked easiest...

robimarko, 2019-01-08:
@aparcar Its just used for vagrant as hypervisor, but it will fail like ansible on the same config file error.
So I tried all of the install methods and they all fail

FreifunkUFO, 2019-01-23:
@robimarko please try again, @aparcar was changing some things
see https://github.com/aparcar/attendedsysupgrade-server/issues/105
and https://github.com/aparcar/attendedsysupgrade-server/compare/mixing

FreifunkUFO, 2019-02-14:
now i tried with docker

here i got this error-message: https://gist.github.com/FreifunkUFO/ff2bf0810fbdd12da5f867b0b7bcac42

FreifunkUFO, 2019-02-14:
ein zeilenumbruch zuviel?
https://github.com/aparcar/attendedsysupgrade-server/blob/master/asu/utils/worker.py#L78

aparcar, 2019-02-14:
please try the docker-compose of the `newdb` branch

FreifunkUFO, 2019-02-14:
ok, other branch... where is it written?
less errors.. only db is coming up automatically. when i start other stuff via portainer i got https://gist.github.com/FreifunkUFO/12d38fbe6727c610a172e52c59945290

using that unicorn command i got:
https://gist.github.com/FreifunkUFO/08cdb817a724c9c3442f65e07c5cbba9

FreifunkUFO, 2019-02-19:
ok, docker-installation now working. (tried the newdb branch)

changing the "active_targets" was nice, the docker now only takes 4400MByte when being installed.

but chef isnt usable yet, problem is to get the /network-profiles/Packages file (discussed that some days ago)

aparcar, 2020-02-25:
Install documentation was added to the README

closed 2020-02-25 by aparcar