openwrt/asu #136: repair all CHEF installations

FreifunkUFO, 2019-01-04:
all (two) chef-installations (mentioned on README) are out of order. please repair it. 
the first one (dev) has 404.. and the second one has an build-queue larger than 100. (and its called the dev-system by it-self, which should be the first one)

are there any other (running) instances of chef on the internet?

FreifunkUFO, 2019-01-04:
see also https://github.com/libremesh/chef/issues/73

aparcar, 2019-01-08:
please try chef.libremesh.org
the servers storage usage went to 100%

FreifunkUFO, 2019-01-08:
thank you.

you were deleting the 2nd server https://github.com/aparcar/attendedsysupgrade-server/commit/f9d9846dd47f23d37697ba5db3b4ccf832679af3 so there was no chance to repair that dev-server?

FreifunkUFO, 2019-01-08:
on chef.libremesh.org is written beta/development: 
cite "Chef Imagebuilder BETA dev"
and "Warning: This is the development version of Chef."

so, where are the stable server, which is not beta? :-o

FreifunkUFO, 2019-01-11:
libremesh-unstable isnt working atm, but its already marked as "unstable".. (and outdated, 17.06)

closed 2019-01-11 by FreifunkUFO

aparcar, 2019-01-11:
libremesh-unstable should be fixed via https://github.com/aparcar/attendedsysupgrade-server/commit/39de9f1e45a57c532757f317071c04319093df6a