openwrt/asu #138: Add INSTALL documentation for ubuntu/debian/openwrt

FreifunkUFO, 2019-01-11:
please provide documentation howto install and host such a server on a standard ubuntu, debian and/or openwrt installation. (including chef)

see also https://github.com/aparcar/attendedsysupgrade-server/issues/135, which were mentioned before, but install method WITHOUT vagrant,ansibel, docker and virtualbox would be nice.

aparcar, 2019-01-23:
Hi @FreifunkUFO , the installation process is somewhat complex, as it does not only include cloning a repo. I added some dev steps, however they might be incomplete.

https://github.com/aparcar/attendedsysupgrade-server/tree/mixing#development

Please reopen this issue if steps are missing, also please provide PRs to improve the documentation

FreifunkUFO, 2019-01-31:
hey @aparcar, you wrote
> Please reopen this issue

but, i cant reopen issues

FreifunkUFO, 2019-01-31:
and @aparcar wrote 
> Setting up Postgresql

would be nice to have these commands to do that. (f.i. "passwd postgres" wasnt useful ) i found some, but not working for me yet.
and, i think the database setting is missing in the config.yml? but i dont know correct syntax by now.

aparcar, 2020-02-25:
I rewrote the installation manual (which is now much simpler). So for now you don't need Vagrant, Docker or any other fancy setup. I'll add a `pip` install-able package soonish.

closed 2020-02-25 by aparcar