openwrt/asu #144: raise lime-packages brunch

FreifunkUFO, 2019-01-30:
building "testing" or "unstable" libremesh via chef is using 18-06-1 (or master) openwrt. but some package-repos are still with 17.06-brunch. please change!

see f.i. https://chef.libremesh.org/download/lime/unstable/ar71xx/generic/tl-wdr3600-v1/eebcbf571383e0e//buildlog-5fcf314cbc5f146.txt

IMAGE_HASH=5fcf314cbc5f146
BIN_DIR=/tmp/tmp_lyvy_kb
J=4
REPOS=src/gz openwrt_core http://downloads.openwrt.org/snapshots/targets/{{ target }}/{{ subtarget }}/packages
src/gz openwrt_base http://downloads.openwrt.org/snapshots/packages/{{ pkg_arch }}/base
src/gz openwrt_luci http://downloads.openwrt.org/snapshots/packages/{{ pkg_arch }}/luci
src/gz openwrt_routing http://downloads.openwrt.org/snapshots/packages/{{ pkg_arch }}/routing
src/gz openwrt_packages http://downloads.openwrt.org/snapshots/packages/{{ pkg_arch }}/packages
src/gz openwrt_telephony http://downloads.openwrt.org/snapshots/packages/{{ pkg_arch }}/telephony
src imagebuilder file:packages
src/gz libremesh https://chef.libremesh.org/snapshots/packages/
src/gz libremap http://repo.libremesh.org/releases/17.06/packages/{{ pkg_arch }}/libremap
src/gz limeui http://repo.libremesh.org/releases/17.06/packages/{{ pkg_arch }}/limeui
src/gz profiles https://as-test.stephen304.com/network-profiles/

unfortunatly on the server there is no master-brunch:
http://repo.libremesh.org/releases/

aparcar, 2019-01-30:
both repos are legacy as far as I know. lime ui was merged in packages, the other one is handled by freifunk not lime anymore

aparcar, 2019-01-30:
I'll remove them shortly

aparcar, 2019-02-03:
related https://github.com/libremesh/lime-packages/issues/464

aparcar, 2021-03-14:
I think this issue is no longer relevant, if I'm wrong please reopen.

closed 2021-03-14 by aparcar