openwrt/asu #145: meeting with freifunk-leipzig for chef-installation

FreifunkUFO, 2019-01-30:
please think about a real-life-meeting for managing and testing a chef-installations (and ass-server). 

location: leipzig (changing locations, sometimes dezentrale)
time: every thursday evening.

FreifunkUFO, 2019-01-30:
see also 
* https://github.com/aparcar/attendedsysupgrade-server/issues/144
* https://github.com/aparcar/attendedsysupgrade-server/issues/142
* https://github.com/aparcar/attendedsysupgrade-server/issues/135
* https://github.com/aparcar/attendedsysupgrade-server/issues/138
* https://github.com/aparcar/attendedsysupgrade-server/issues/112

aparcar, 2019-01-30:
Please request a meeting at mail [at] apracar.org

closed 2019-01-30 by aparcar

FreifunkUFO, 2019-01-31:
hm ok, i sent an email inviting you to our jabber-chat.