openwrt/asu #152: TP-Link TL-WR841N v9: upgrade check fails with  "Unsupported device, release, target, subtraget or board"

rkunze, 2019-02-16:
I am running a TL-WR841N v9 as a simple WLAN access point with an OpenWRT 18.06.2 image generated from https://chef.libremesh.org/ with this package list:
```
odhcp6c
base-files
mtd
fstools
busybox
wpad-mini
kmod-gpio-button-hotplug
logd
swconfig
kmod-ath9k
dropbear
netifd
uci
libc
uclient-fetch
uboot-envtools
libgcc
odhcpd-ipv6only
luci-ssl
luci-app-attendedsysupgrade
```

I did remove some of the default packages (opkg, ip6tables, dnsmasq, ppp, ppp-mod-pppoe, iptables, firewall) which are not needed for my use case in order to have enough flash space for luci-ssl and luci-app-attendedsysupgrade.

Checking for upgrades via the web interface against https://chef.libremesh.org/ fails with an error message "Unsupported device, release, target, subtraget or board".

The corresponding API call to https://chef.libremesh.org/api/upgrade-check fails with HTTP error code 412. Data for the API call is:
```json
{
  "version":"18.06.2",
  "revision":"r7676-cddd7b4c77",
  "installed":{
    "libc":"1.1.19-1",
    "busybox":"1.28.4-2",
    "swconfig":"11",
    "base-files":"194.2-r7676-cddd7b4c77",
    "netifd":"2019-01-31-a2aba5c7-2.1",
    "uboot-envtools":"2018.03-1",
    "odhcp6c":"2018-07-14-67ae6a71-15",
    "fstools":"2018-12-28-af93f4b8-3",
    "uclient-fetch":"2018-11-24-3ba74ebc-1",
    "kmod-ath9k":"4.9.152+2017-11-01-9",
    "uci":"2018-08-11-4c8b4d6e-1",
    "wpad-mini":"2018-05-21-62566bc2-5",
    "luci-ssl":"git-19.046.40869-30d9bc0-1",
    "dropbear":"2017.75-7.1",
    "mtd":"23",
    "odhcpd-ipv6only":"1.15-3",
    "libgcc":"7.3.0-1",
    "kmod-gpio-button-hotplug":"4.9.152-2",
    "logd":"2018-02-14-128bc35f-2",
    "luci-app-attendedsysupgrade":"git-19.046.40869-30d9bc0-1",
    "kernel":"4.9.152-1-590acf3fecb4e0e9d01b08c7632cb375"
  },
  "upgrade_packages":"1",
  "distro":"OpenWrt",
  "target":"ar71xx",
  "subtarget":"tiny"
}
```

aparcar, 2019-02-16:
`target` and `subtarget` where merged, I'll release a new `luci-app-attendedsysupgrade` version soon

aparcar, 2019-02-16:
Plus I may add a workaround on the server side, thanks for using and reporting!

rkunze, 2019-02-16:
> thanks for using and reporting!

Thanks for providing this service - using it is _much_ easier and faster than building a customized firmware locally. And without a customized build, the TL-WR841N is pretty much useless nowadays...

aparcar, 2019-02-16:
Thanks, happy to hear that

aparcar, 2019-02-20:
Hi, I merge the latest changes to openwrt-18.06 branch, I hope they're soonish available to update. With that the upgrade process should works as intended, please keep me updated and close this issue on success!

https://github.com/openwrt/luci/pull/2544

rkunze, 2019-02-21:
Seems to work now (at least on my TL-WDR4300, where I do have enough flash space to upgrade luci-app-attendedupgrades via opkg).

But now I get "No upgrades available" from chef.libremesh.org, even if `opkg list-upgradable` lists a number of upgradable packages and I have upgrade_packages = 1 in /etc/config/attendedsysupgrade. 

Is this intended?

aparcar, 2019-02-21:
Let me check that, thanks for reporting!

rkunze, 2019-02-21:
Should I open a new ticket for this and close this one?

aparcar, 2019-02-21:
please retry

rkunze, 2019-02-21:
Mixed results :slightly_frowning_face: 
* On one box (TL-WDR4300 v1), checking for upgrades works and returns plausible results (`opkg check-upgradable` returns some more packages, but these are all dependencies of the packages returned by the upgrade check). But a subsequent "Request Firmware" runs into an HTTP 500 error on chef.libremesh.org
* On another box (TL-WR841ND), checking for upgrades now gives an HTTP 500 error on chef.libremesh.org

Do you need more info (requests, responses, ...)?

aparcar, 2019-02-21:
please send the request details

rkunze, 2019-02-21:
Broken upgrade check (originally from TL-WR841ND):
```sh
$ curl 'https://chef.libremesh.org/api/upgrade-check' -H 'Referer: https://ap-2.lan/cgi-bin/luci/admin/system/attended_sysupgrade' -H 'Origin: https://ap-2.lan' -H 'User-Agent: Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/72.0.3626.109 Safari/537.36' -H 'Content-type: application/json' --data-binary '{"version":"18.06.2","revision":"r7676-cddd7b4c77","installed":{"libc":"1.1.19-1","busybox":"1.28.4-2","swconfig":"11","base-files":"194.2-r7676-cddd7b4c77","netifd":"2019-01-31-a2aba5c7-2.1","uboot-envtools":"2018.03-1","odhcp6c":"2018-07-14-67ae6a71-15","fstools":"2018-12-28-af93f4b8-3","uclient-fetch":"2018-11-24-3ba74ebc-1","kmod-ath9k":"4.9.152+2017-11-01-9","uci":"2018-08-11-4c8b4d6e-1","wpad-mini":"2018-05-21-62566bc2-5","luci-ssl":"git-19.051.55698-76cf653-1","dropbear":"2017.75-7.1","mtd":"23","odhcpd-ipv6only":"1.15-3","kmod-gpio-button-hotplug":"4.9.152-2","logd":"2018-02-14-128bc35f-2","luci-app-attendedsysupgrade":"git-19.051.55698-76cf653-1","kernel":"4.9.152-1-590acf3fecb4e0e9d01b08c7632cb375"},"upgrade_packages":"1","distro":"OpenWrt","target":"ar71xx/tiny"}' --compressed

<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 3.2 Final//EN">
<title>500 Internal Server Error</title>
<h1>Internal Server Error</h1>
<p>The server encountered an internal error and was unable to complete your request.  Either the server is overloaded or there is an error in the application.</p>
```

Working upgrade-check (Originally from TL-WDR4300):
```sh
$ curl 'https://chef.libremesh.org/api/upgrade-check' -H 'Referer: https://ap-1.lan/cgi-bin/luci/admin/system/attended_sysupgrade' -H 'Origin: https://ap-1.lan' -H 'User-Agent: Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/72.0.3626.109 Safari/537.36' -H 'Content-type: application/json' --data-binary '{"version":"18.06.2","revision":"r7676-cddd7b4c77","installed":{"openssh-sftp-server":"7.7p1-1","libc":"1.1.19-1","opkg":"2019-01-18-7708a01a-1","triggerhappy":"0.5.0-1","kmod-usb-core":"4.9.152-1","busybox":"1.28.4-2","ip-full":"4.16.0-8","bash":"4.4.18-2","swconfig":"11","evtest":"1.33-1","sispmctl":"3.1+20120206-1","base-files":"194.2-r7676-cddd7b4c77","netifd":"2019-01-31-a2aba5c7-2.1","coreutils":"8.23-3","kmod-usb-serial-pl2303":"4.9.152-1","uboot-envtools":"2018.03-1","kmod-usb-hid":"4.9.152-1","less":"487-1","kmod-usb2":"4.9.152-1","firewall":"2018-08-13-1c4d5bcd-1","luci-app-unbound":"git-19.046.40869-30d9bc0-1","unbound-control":"1.8.1-1","tcpdump":"4.9.2-1","odhcp6c":"2018-07-14-67ae6a71-15","fstools":"2018-12-28-af93f4b8-3","uclient-fetch":"2018-11-24-3ba74ebc-1","kmod-ath9k":"4.9.152+2017-11-01-9","uci":"2018-08-11-4c8b4d6e-1","wpad-mini":"2018-05-21-62566bc2-5","luci-ssl":"git-19.046.40869-30d9bc0-1","dropbear":"2017.75-7.1","curl":"7.60.0-3","mtd":"23","odhcpd-ipv6only":"1.15-3","libgcc":"7.3.0-1","ppp":"2.4.7-12","luci-app-commands":"git-19.046.40869-30d9bc0-1","socat":"1.7.3.2-3","kmod-gpio-button-hotplug":"4.9.152-2","logd":"2018-02-14-128bc35f-2","kmod-usb-serial":"4.9.152-1","unbound":"1.8.1-1","coreutils-sleep":"8.23-3","iptables":"1.6.2-1","luci-app-attendedsysupgrade":"git-19.051.55698-76cf653-1","ip6tables":"1.6.2-1","kmod-usb-ledtrig-usbport":"4.9.152-1","coreutils-stty":"8.23-3","kernel":"4.9.152-1-def6b4b6fd5b0fd8587fe8dbf55baf5c","picocom":"1.7-2","ppp-mod-pppoe":"2.4.7-12","dnsmasq-full":"2.80-1.2"},"upgrade_packages":"1","distro":"OpenWrt","target":"ar71xx/generic"}' --compressed

{"upgrades": {"luci-ssl": ["git-19.051.55698-76cf653-1", "git-19.046.40869-30d9bc0-1"], "luci-app-commands": ["git-19.051.55698-76cf653-1", "git-19.046.40869-30d9bc0-1"], "luci-app-unbound": ["git-19.051.55698-76cf653-1", "git-19.046.40869-30d9bc0-1"]}}
```

Subsequent broken upgrade-request (Originally from TL-WDR4300):

```sh
$ curl 'https://chef.libremesh.org/api/upgrade-request' -H 'Referer: https://ap-1.lan/cgi-bin/luci/admin/system/attended_sysupgrade' -H 'Origin: https://ap-1.lan' -H 'User-Agent: Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/72.0.3626.109 Safari/537.36' -H 'Content-type: application/json' --data-binary '{"board":"tl-wdr4300","model":"TP-Link TL-WDR4300 v1","distro":"OpenWrt","target":"ar71xx/generic"}' --compressed

<!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 3.2 Final//EN">
<title>500 Internal Server Error</title>
<h1>Internal Server Error</h1>
<p>The server encountered an internal error and was unable to complete your request.  Either the server is overloaded or there is an error in the application.</p>
```

BTW: How do you determine the individual set of packages to build from that last request?

aparcar, 2019-02-22:
should be fixed via https://github.com/aparcar/attendedsysupgrade-server/commit/b1726e8f7bbcb42bfbf03cab5d860d6b2c00610a

one of your routers was simply up to date, so no `upgrades` for `json.loads()` which caused the fail.

aparcar, 2019-02-22:
> BTW: How do you determine the individual set of packages to build from that last request?

Server checks if packages are "transformed" into others, as e.g. `libgcc` is called `libgcc1` in recent snapshots. Apart from that it just throws back the packages received in the `installed` parameter.

rkunze, 2019-02-22:
> should be fixed via [b1726e8](https://github.com/aparcar/attendedsysupgrade-server/commit/b1726e8f7bbcb42bfbf03cab5d860d6b2c00610a)

OK, I'll retry this evening when I get home.

> one of your routers was simply up to date, so no `upgrades` for `json.loads()` which caused the fail.

Yep, that's one of the WR-841NDs. Those beasts have just a couple of 100kB writable space left (just enough for a fairly simple config, but not nearly enough to upgrade a package via opkg), so I had to build a fresh image first to get the new luci-mod-attendedsysupgrades in order to check if it worked. so its no wonder the image was up to date.

>> BTW: How do you determine the individual set of packages to build from that last request?
>
> Server checks if packages are "transformed" into others, as e.g. `libgcc` is called `libgcc1` in recent snapshots. Apart from that it just throws back the packages received in the `installed` parameter.

But there is no "installed" parameter in the JSON data for "upgrade-request", only in the JSON data for "upgrade-check"...

aparcar, 2019-02-22:
The device receives a list of `upgrades` and a list of `packages` to request. The former is for the users interface only, the latter will be requested by the server.

rkunze, 2019-02-22:
> OK, I'll retry this evening when I get home.

Just tried, but chef.openmesh.org seems to be down (Error 502 "Bad Gateway" on any API call).

aparcar, 2019-02-22:
please try again, server was down sorry

rkunze, 2019-02-23:
Server side seems to work now, at least for `upgrade-check`: No more server errors, and the server returns an empty object for `packages` if no upgrades are available. :+1: 

Still broken:
1) Client side of the upgrade check: If there are no upgrades available, the client now displays a "Package upgrades available" box without any packages instead of "No upgrades available". 
2) "Edit installed packages" after a successful `upgrade-check`: Runs into a JavaScript error in `/luci-static/resources/attendedsysupgrade.js` line 43 because `data.packages` is undefined.
3) "Request firmware" after a successful `upgrade-check`: Server returns an HTTP error 500 from `upgrade-request`. See https://github.com/aparcar/attendedsysupgrade-server/issues/152#issuecomment-466217513 for an example of a failed `upgrade-request`

I think 2. and 3. are related and both caused by the same error on the client side - have to investigate a bit more to be sure, though.

I'm closing this ticket now since the original bug is fixed, and I'll file new tickets and/or PRs in https://github.com/openwrt/luci/ for the client side stuff.

closed 2019-02-23 by rkunze