openwrt/asu #153: Build fails for some packages that replace busybox apps

rkunze, 2019-02-17:
When trying to build an OpenWRT image where some busybox apps are replaced by standalone utilities from different packages, sometimes the build fails. 

Example: Building for "TP-Link TL-WDR4300 v1" with the "OpenWRT default packes only" list plus the "diffutils" package will fail.

The relevant part of the build logs:
```
Collected errors:
 * check_data_file_clashes: Package busybox wants to install file /home/aparcar/worker/imagebuilder/openwrt/18.06.2/ar71xx/generic/build_dir/target-mips_24kc_musl/root-ar71xx/usr/bin/cmp
	But that file is already provided by package  * diffutils
 * opkg_install_cmd: Cannot install package busybox.
 * check_data_file_clashes: Package busybox wants to install file /home/aparcar/worker/imagebuilder/openwrt/18.06.2/ar71xx/generic/build_dir/target-mips_24kc_musl/root-ar71xx/usr/bin/cmp
	But that file is already provided by package  * diffutils
 * opkg_install_cmd: Cannot install package busybox.
```

The cause of this error seems to be the ordering of the packages for building. From the start of the build log:
```
### BUILD COMMAND:

DEFAULTS_HASH=""
REQUEST_HASH="05c83dd416f8"
REQUEST_ID="140"
PROFILE="tl-wdr4300-v1"
PACKAGES="kmod-usb-ledtrig-usbport firewall fstools odhcp6c netifd uboot-envtools kmod-ath9k swconfig iptables dropbear kmod-gpio-button-hotplug kmod-usb2 uci libgcc mtd diffutils ip6tables ppp-mod-pppoe dnsmasq busybox uclient-fetch logd -libc wpad-mini base-files ppp opkg kmod-usb-core odhcpd-ipv6only"
VERSION="18.06.2"
DISTRO="openwrt"
TARGET="ar71xx/generic"
IMAGE_HASH="None"
PACKAGES_HASH="f49592ba1aa6"
./meta image
```

Relevant here is that "busybox" comes after "diffutils" in the PACKAGES list, which will cause the build to fail. A test build with a PACKAGES list where "busybox"  is listed _before_ "diffutils" (e.g. from a package list that contains only "busybox" and "diffutils"), the build will succeed.

aparcar, 2019-02-18:
What is your suggestion to fix this?

rkunze, 2019-02-18:
I'd either try to preserve the package order as presented in the GUI (although that is not really a fix but more of a workaround, since it will still possible to break a build by specifying packages in the wrong order). 

Or make sure that PACKAGES first lists all packages from the chosen base system, and then all additional packages specified by the user (on the assumption that packages from the base system - like busybox - are more likely to be prepared to allow partial replacements).

Not really sure what the best fix is though...

aparcar, 2019-02-19:
The packages are diffed here, this is somewhat important as a router doens't know itself what packages are usually installed, so it just requests the onces installed. The server then checks the packages and *distracts* packages missing. The code is here https://github.com/aparcar/attendedsysupgrade-server/blob/master/asu/utils/worker.py#L73

I have no idea of an easy fix for that... I'll think about it but feel free to come up with something as well :)!

rkunze, 2019-02-19:
Assuming that `self.database.get_packages_hash()` returns the packages in the same order as specified in the GUI, how about this?
```python
if self.params["packages_hash"]:
    packages_image = set(self.database.get_packages_image(self.params))
    self.log.debug("packages_image %s", packages_image)
    packages_requested_inorder = self.database.get_packages_hash(
        self.params["packages_hash"])
    packages_requested = set(packages_inorder)
    self.log.debug("packages_requested %s", packages_requested)
    packages_remove = packages_image - packages_requested
    self.log.debug("packages_remove %s", packages_remove)
    self.params["packages"] =  " ".join(packages_requested_inorder 
           + list(map(lambda x: "-" + x, packages_remove)))
    self.log.debug("packages param %s", self.params["packages"])
```
i.e. put the requested packages into `self.params["packages"]` in the same order as originally requested, and simply tack the removed packages onto the end of the list...

rkunze, 2019-02-19:
As for my other idea (list system packages first, then additional packages):
```python
if self.params["packages_hash"]:
    packages_image = set(self.database.get_packages_image(self.params))
    self.log.debug("packages_image %s", packages_image)
    packages_requested = set(self.database.get_packages_hash(
        self.params["packages_hash"]))
    self.log.debug("packages_requested %s", packages_requested)
    packages_remove = packages_image - packages_requested
    self.log.debug("packages_remove %s", packages_remove)
    packages_image_keep = packages_image - packages_remove
    packages_requested_extra = packages_requests - packages_image
    packages_requested.update()
    self.params["packages"] =  " ".join(
        list(packages_image_keep) 
        + list(map(lambda x: "-" + x, packages_remove))
        + list(packages_requested_extra))
    self.log.debug("packages param %s", self.params["packages"])
```

rkunze, 2019-02-19:
Still not sure which is the best way, though. The first allows shooting yourself in the foot by using the wrong order in the GUI, and the second assumes that package order does not really matter as long as all image packages are listed before all additional packages...

aparcar, 2021-08-29:
This is really an OPKG problem, I hope we can fix this by switching over to APK...