openwrt/asu #155: banner.html

FreifunkUFO, 2019-02-19:
should that file being loaded from chef or from asu-server?
at the moment it seems to be from asu-server (of course atm with 404)

[image]

aparcar, 2019-02-19:
This is normal behavior if no banner is set, no worries

FreifunkUFO, 2019-02-20:
where can i "set" the banner?

aparcar, 2019-02-20:
rename banner.html.example to banner.html and fill it with whatever html content you like

FreifunkUFO, 2019-02-22:
do you mean https://github.com/libremesh/chef/commit/fa5c385518278ba3b07c273420b68c9ca005ba73 ?
i have a chef-directory with banner.html and it is not loaded from that chef-webserver. (but browser tries to load it from the asu-server. see screenshot above!
but inside view-py-file i cant see any banner-html-tasks.

so that shouldnt be "normal behaviour", please reopen this ticket!

p.s. 
yes i know, i can fill html into that file.. btw: its not an html-file, but only a html template file.

aparcar, 2019-02-22:
I need to add a real nginx config..m

aparcar, 2020-02-25:
No longer used in new frontend.

closed 2020-02-25 by aparcar