openwrt/asu #156: Unknown package in request

McChary, 2019-02-19:
[image]

when I try to create an image snapshots, this package is integrated into the list with luci, but when creating the image, that error comes out

FreifunkUFO, 2019-02-19:
it seems that filename has changed!

from https://downloads.openwrt.org/releases/18.06.2/targets/ar71xx/tiny/packages/libgcc_7.3.0-1_mips_24kc.ipk  
to https://downloads.openwrt.org/snapshots/targets/ar71xx/tiny/packages/libgcc1_7.4.0-1_mips_24kc.ipk

so workaround for you @McChary is to rename it (via Edit Packages) from libgcc to libgcc1

aparcar, 2019-02-19:
filename changes in a stable branch?

aparcar, 2019-02-19:
I'm not entirely sure but I could imagine it's a problem of the upstream image builders...

closed 2019-02-19 by aparcar

aparcar, 2019-02-19:
Should work now

commit 32f8753a (2019-02-19): fix #156